google/aiyprojects-raspbian #166: Enhancement: add text input to the Google Assistant

charbull, 2017-11-05:
Hello, 

It will be really helpful, to add the input text to the Google Assistant API.

1- record user voice
2- send to google cloud and get a text back
3- check if the text can be handled locally and handle it
4- if not insert the text for the google assistant 

However, it seems that the google assistant cannot take a text as input

divx118, 2017-11-05:
Sorry, if I misunderstand you, however if the 'text' can't be handled locally it will normally be handled by the assistent, because you won't call `assistant.stop_conversation()` then.
If I misunderstood, could you elaborate and maybe add a use case?

drigz, 2017-11-06:
Hey @charbull, in your use case I think it would be best to use either `stop_conversation()` or to keep the audio and send that to the Assistant API. Is there a reason those wouldn't work?

If you do need text input for your use case, you could suggest it on [googlesamples/assistant-sdk-python](https://github.com/googlesamples/assistant-sdk-python).

charbull, 2017-11-10:
Great Thank you !

charbull, 2017-11-10:
Actually, let me refine what I meant in terms of use cases:

I want to ask my Google Kit questions:
- specific questions to be forwarded to my webservice: `Where is Charbull?`
- general questions to the Google Assistant: `What is the temperature outside?`

So here are the steps:
1- I want to record `The voice of Where is Charbull ?` 
2- Interact with the google speech cloud to get the `text of Where is Charbull ?`
3a- If my webservice answers with a response then I render it on the pi.audio
3b- If my webservice answers with 0, then it is something my webservice can't answer for example `What is the temperature outside?`.
 In this case I want to forward it to the Google Assistant because it already knows how to answer this `What is the temperature outside`. 
Since I have the text already provided by the Google Speech, I would like to pass it in String format to the Google Assistant.
So far the Google Assistant takes an audio as an input and I didn't see a text input (`String`)

Is there a way to do this?

divx118, 2017-11-10:
@charbull IMO you could use https://github.com/google/aiyprojects-raspbian/blob/voicekit/src/assistant_library_with_local_commands_demo.py as an example and go from there. No extra text input needed. 
I am working with my son on https://github.com/divx118/aiyprojects-raspbian/blob/work-voicekit/src/assistent_with_button_local_commands_mpd.py It just filters the commands we execute locally and the rest is forwarded to the assistent. 
PS: It is a WIP, probably will rewrite it into using some sort of database so it will be easier to add stuff and don't look cluttered however we are still brainstorming about that.

mpember, 2017-11-11:
@charbull 
You appear to be a tad confused about how the code works.  If you are only wanting to get the text of the user's request, there is no need to use the Cloud API.  You can do it with the Assistant API (which is free).

closed 2018-01-04 by drigz

andrewtvuong, 2018-12-04:
hey @charbull, did you ever get your answer. I have been wondering about this for quite a while. Essentially to send input in form of text by terminal to get voice pi to output what it would have done if I pushed a button and asked the same question by voice.

dmitriykovalev, 2018-12-06:
@andrewtvuong I guess you can do it by calling [send_text_query](https://developers.google.com/assistant/sdk/reference/library/python/#google.assistant.library.Assistant.send_text_query) function from Assistant Library API.

dmitriykovalev, 2018-12-06:
@andrewtvuong There is a `text_query` in [AssistConfig](https://developers.google.com/assistant/sdk/reference/rpc/google.assistant.embedded.v1alpha2#assistconfig) used in gRPC API. By the way, assistant Library API should work on Pi Zero without hotword detection.

andrewtvuong, 2018-12-07:
@dmitriykovalev awesome, thanks for sharing! I've been trying to hack this for a while but put it down earlier this year.