google/aiyprojects-raspbian #168: installed on Ubuntu Mate for pi,Failed to find the voiceHAT soundcard

makelove, 2017-11-09:
I installed this  on Ubuntu Mate for pi, follow HACKING.md
When I run checkpoints/check_audio.py :
(env) pi@pi-desktop:~/voice-recognizer-raspi$ ./checkpoints/check_audio.py
Failed to find the voiceHAT soundcard. Refer to HACKING.md for how to
setup the voiceHAT driver: https://git.io/v99yK
Press Enter to close...

-
(env) pi@pi-desktop:~/voice-recognizer-raspi$ aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
pulse
    PulseAudio Sound Server
softvol
micboost
default
    Playback/recording through the PulseAudio sound server

-
(env) pi@pi-desktop:~/voice-recognizer-raspi$ lsmod | grep snd
snd_soc_bcm2835_i2s     6354  0
snd_soc_core          125949  1 snd_soc_bcm2835_i2s
snd_pcm_dmaengine       3391  1 snd_soc_core
snd_bcm2835            20447  0
snd_pcm_oss            37649  0
snd_mixer_oss          14119  1 snd_pcm_oss
snd_pcm                75762  4 snd_bcm2835,snd_pcm_oss,snd_soc_core,snd_pcm_dmaengine
snd_seq_dummy           1269  0
snd_seq_oss            30670  0
snd_seq_midi            4503  0
snd_rawmidi            18516  1 snd_seq_midi
snd_seq_midi_event      5546  2 snd_seq_oss,snd_seq_midi
snd_seq                53119  6 snd_seq_midi_event,snd_seq_oss,snd_seq_dummy,snd_seq_midi
snd_seq_device          3621  4 snd_seq,snd_rawmidi,snd_seq_oss,snd_seq_midi
snd_timer              19288  2 snd_pcm,snd_seq
snd                    51908  10 snd_bcm2835,snd_pcm_oss,snd_soc_core,snd_timer,snd_pcm,snd_seq,snd_rawmidi,snd_seq_oss,snd_seq_device,snd_mixer_oss

-
Raspberry pi 3b  + Ubuntu Mate
What is the problem ? How to fix it ?

makelove, 2017-11-09:
another problem when I was installing scripts/install-deps.sh
`
Package python3-pysocks is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  python3-socks

E: Package 'python3-pysocks' has no installation candidate
`

I fixed this by replace  python3-pysocks with python3-socks .

Just telling you guys.

drigz, 2017-11-09:
Thanks for the report!

You need kernel 4.9 for the driver. Can you update with `sudo rpi-update`?
If that doesn't get you to 4.9 you may need to ask on the Ubuntu forums for
support.

If you have 4.9, you may need to run `sudo scripts/configure_driver.sh`
from the AIY repository, then restart. If this doesn't work, please post
your `/boot/config.txt` along with the new `lsmod` output.

makelove, 2017-11-10:
rpi-update is succeful,
speaker is OK,can speak sound,BUT microphone is NOT working,can not hear me speak
```bash
(env) pi@pi-desktop:~/voice-recognizer-raspi$ ./checkpoints/check_audio.py
Playing a test sound...
Did you hear the test sound? (y/n) y
When you're ready, press enter and say 'Testing, 1 2 3'...
Recording...
Playing back recorded audio...
Did you hear your own voice? (y/n) n
There may be a problem with your microphone. Check that it's connected
properly.
Press Enter to close...
```
I check the system-sound-countorl-pad, Not found microphone input.

drigz, 2017-11-10:
This is probably an issue with the ALSA configuration. Did you run scripts/install-alsa-config.sh?

Can you post the output of `arecord --list-devices`?

makelove, 2017-11-10:
```bash
pi@aiy:~$ arecord --list-devices
**** List of CAPTURE Hardware Devices ****
card 0: sndrpigooglevoi [snd_rpi_googlevoicehat_soundcar], device 0: Google voiceHAT SoundCard HiFi voicehat-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```
sure,I ran scripts/install-alsa-config.sh.

drigz, 2017-11-10:
OK, can you post the output from:

`arecord -v -c 1 -f s16 -r 16000 -d 3 test.wav`

and also the test.wav that it creates? It will record for three seconds, please try talking and tapping on the microphone.

makelove, 2017-11-10:
```bash
pi@aiy:~/Downloads$ arecord -v -c 1 -f s16 -r 16000 -d 10 test.wav
Recording WAVE 'test.wav' : Signed 16 bit Little Endian, Rate 16000 Hz, Mono
ALSA <-> PulseAudio PCM I/O Plugin
Its setup is:
  stream       : CAPTURE
  access       : RW_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 1
  rate         : 16000
  exact rate   : 16000 (16000/1)
  msbits       : 16
  buffer_size  : 8000
  period_size  : 2000
  period_time  : 125000
  tstamp_mode  : NONE
  tstamp_type  : GETTIMEOFDAY
  period_step  : 1
  avail_min    : 2000
  period_event : 0
  start_threshold  : 1
  stop_threshold   : 8000
  silence_threshold: 0
  silence_size : 0
  boundary     : 2097152000
```
result : No sound come out from speaker

drigz, 2017-11-10:
It looks like the ALSA config that the script installs is not being used. Can you post the output of:

`strace -e open arecord -c 1 -f s16 -r 16000 -d 1 test.wav`

makelove, 2017-11-10:
```bash
pi@aiy:~/Downloads$ strace -e open arecord -c 1 -f s16 -r 16000 -d 1 test.wav
open("/etc/ld.so.preload", O_RDONLY|O_CLOEXEC) = 3
open("/usr/lib/arm-linux-gnueabihf/libarmmem.so", O_RDONLY|O_CLOEXEC) = 3
open("/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
open("/lib/arm-linux-gnueabihf/librt.so.1", O_RDONLY|O_CLOEXEC) = 3
open("/usr/lib/arm-linux-gnueabihf/libasound.so.2", O_RDONLY|O_CLOEXEC) = 3
open("/lib/arm-linux-gnueabihf/libpthread.so.0", O_RDONLY|O_CLOEXEC) = 3
open("/lib/arm-linux-gnueabihf/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
open("/lib/arm-linux-gnueabihf/libm.so.6", O_RDONLY|O_CLOEXEC) = 3
open("/lib/arm-linux-gnueabihf/libdl.so.2", O_RDONLY|O_CLOEXEC) = 3
open("/usr/lib/locale/locale-archive", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
open("/usr/share/alsa/alsa.conf", O_RDONLY) = 3
open("/usr/share/alsa/alsa.conf.d/", O_RDONLY|O_NONBLOCK|O_LARGEFILE|O_DIRECTORY|O_CLOEXEC) = 3
open("/usr/share/alsa/alsa.conf.d//50-pulseaudio.conf", O_RDONLY) = 3
open("/usr/share/alsa/alsa.conf.d//pulse.conf", O_RDONLY) = 3
open("/etc/asound.conf", O_RDONLY)      = 3
open("/usr/lib/arm-linux-gnueabihf/alsa-lib/libasound_module_conf_pulse.so", O_RDONLY|O_CLOEXEC) = 3
open("/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
open("/usr/lib/arm-linux-gnueabihf/libpulse.so.0", O_RDONLY|O_CLOEXEC) = 3
open("/usr/lib/arm-linux-gnueabihf/pulseaudio/tls/v7l/neon/vfp/libjson-c.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/lib/arm-linux-gnueabihf/pulseaudio/tls/v7l/neon/libjson-c.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/lib/arm-linux-gnueabihf/pulseaudio/tls/v7l/vfp/libjson-c.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/lib/arm-linux-gnueabihf/pulseaudio/tls/v7l/libjson-c.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/lib/arm-linux-gnueabihf/pulseaudio/tls/neon/vfp/libjson-c.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/lib/arm-linux-gnueabihf/pulseaudio/tls/neon/libjson-c.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/lib/arm-linux-gnueabihf/pulseaudio/tls/vfp/libjson-c.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/lib/arm-linux-gnueabihf/pulseaudio/tls/libjson-c.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/lib/arm-linux-gnueabihf/pulseaudio/v7l/neon/vfp/libjson-c.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/lib/arm-linux-gnueabihf/pulseaudio/v7l/neon/libjson-c.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/lib/arm-linux-gnueabihf/pulseaudio/v7l/vfp/libjson-c.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/lib/arm-linux-gnueabihf/pulseaudio/v7l/libjson-c.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/lib/arm-linux-gnueabihf/pulseaudio/neon/vfp/libjson-c.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/lib/arm-linux-gnueabihf/pulseaudio/neon/libjson-c.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/lib/arm-linux-gnueabihf/pulseaudio/vfp/libjson-c.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/lib/arm-linux-gnueabihf/pulseaudio/libjson-c.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/lib/arm-linux-gnueabihf/libjson-c.so.2", O_RDONLY|O_CLOEXEC) = 3
open("/usr/lib/arm-linux-gnueabihf/pulseaudio/libpulsecommon-8.0.so", O_RDONLY|O_CLOEXEC) = 3
open("/usr/lib/arm-linux-gnueabihf/pulseaudio/libdbus-1.so.3", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/lib/arm-linux-gnueabihf/libdbus-1.so.3", O_RDONLY|O_CLOEXEC) = 3
open("/usr/lib/arm-linux-gnueabihf/pulseaudio/libxcb.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/lib/arm-linux-gnueabihf/libxcb.so.1", O_RDONLY|O_CLOEXEC) = 3
open("/usr/lib/arm-linux-gnueabihf/pulseaudio/libsystemd.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/lib/arm-linux-gnueabihf/libsystemd.so.0", O_RDONLY|O_CLOEXEC) = 3
open("/usr/lib/arm-linux-gnueabihf/pulseaudio/libwrap.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/lib/arm-linux-gnueabihf/libwrap.so.0", O_RDONLY|O_CLOEXEC) = 3
open("/usr/lib/arm-linux-gnueabihf/pulseaudio/libsndfile.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/lib/arm-linux-gnueabihf/libsndfile.so.1", O_RDONLY|O_CLOEXEC) = 3
open("/usr/lib/arm-linux-gnueabihf/pulseaudio/libasyncns.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/lib/arm-linux-gnueabihf/libasyncns.so.0", O_RDONLY|O_CLOEXEC) = 3
open("/usr/lib/arm-linux-gnueabihf/pulseaudio/libXau.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/lib/arm-linux-gnueabihf/libXau.so.6", O_RDONLY|O_CLOEXEC) = 3
open("/usr/lib/arm-linux-gnueabihf/pulseaudio/libXdmcp.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/lib/arm-linux-gnueabihf/libXdmcp.so.6", O_RDONLY|O_CLOEXEC) = 3
open("/usr/lib/arm-linux-gnueabihf/pulseaudio/libselinux.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/lib/arm-linux-gnueabihf/libselinux.so.1", O_RDONLY|O_CLOEXEC) = 3
open("/usr/lib/arm-linux-gnueabihf/pulseaudio/liblzma.so.5", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
[... cut ...]
open("/usr/share/locale/zh_CN/LC_MESSAGES/alsa-utils.mo", O_RDONLY) = -1 ENOENT (No such file or directory)
open("/usr/share/locale/zh/LC_MESSAGES/alsa-utils.mo", O_RDONLY) = -1 ENOENT (No such file or directory)
open("/usr/share/locale-langpack/zh_CN/LC_MESSAGES/alsa-utils.mo", O_RDONLY) = -1 ENOENT (No such file or directory)
open("/usr/share/locale-langpack/zh/LC_MESSAGES/alsa-utils.mo", O_RDONLY) = -1 ENOENT (No such file or directory)
Recording WAVE 'test.wav' : Signed 16 bit Little Endian, Rate 16000 Hz, Mono
open("test.wav", O_WRONLY|O_CREAT, 0644) = 10
+++ exited with 0 +++
```

Thank you ！

drigz, 2017-11-10:
Can you also post the output of `arecord -L`?

makelove, 2017-11-10:
```bash
(env) pi@aiy:~/voice-recognizer-raspi$ arecord -L
null
    Discard all samples (playback) or generate zero samples (capture)
pulse
    PulseAudio Sound Server
softvol
micboost
default
    Playback/recording through the PulseAudio sound server
sysdefault:CARD=sndrpigooglevoi
    snd_rpi_googlevoicehat_soundcar,
    Default Audio Device
dmix:CARD=sndrpigooglevoi,DEV=0
    snd_rpi_googlevoicehat_soundcar,
    Direct sample mixing device
dsnoop:CARD=sndrpigooglevoi,DEV=0
    snd_rpi_googlevoicehat_soundcar,
    Direct sample snooping device
hw:CARD=sndrpigooglevoi,DEV=0
    snd_rpi_googlevoicehat_soundcar,
    Direct hardware device without any conversions
plughw:CARD=sndrpigooglevoi,DEV=0
    snd_rpi_googlevoicehat_soundcar,
    Hardware device with all software conversions
```

```bash
(env) pi@aiy:~/voice-recognizer-raspi$ cat  /etc/asound.conf
options snd_rpi_googlemihat_soundcard index=0

pcm.softvol {
    type softvol
    slave.pcm dmix
    control {
        name Master
        card 0
    }
}

pcm.micboost {
    type route
    slave.pcm dsnoop
    ttable {
        0.0 30.0
        1.1 30.0
    }
}

pcm.!default {
    type asym
    playback.pcm "plug:softvol"
    capture.pcm "plug:micboost"
}

ctl.!default {
    type hw
    card 0
}
```

drigz, 2017-11-10:
OK, try this:

```
arecord -v -D plug:micboost -c 1 -f s16 -r 16000 -d 3 -t wav test.wav
aplay test.wav
```

and also this:

`export | grep ALSA`

makelove, 2017-11-10:
actually ,it can record sounds
When I open [sound perfermence ] --> Hardware -->config file 
change to : Analog Stereo Input
it can record sounds,I can hear that
But
if change to : Analog Stereo Input, the speaker can not speak
[image]

makelove, 2017-11-10:
I change back to  [Analog Stereo Output]
```
arecord -v -D plug:micboost -c 1 -f s16 -r 16000 -d 3 -t wav test.wav
aplay test.wav
```
it  work!

```
export | grep ALSA
```
output nothing

drigz, 2017-11-10:
OK, I don't know what the Ubuntu Sound Preferences do, but they appear to be breaking our default ALSA configuration.

As a workaround, you can try changing `default` to `plug:micboost` in [_recorder.py](https://github.com/google/aiyprojects-raspbian/blob/3332a45591e64095c7a6d2d349cd13114d8962ef/src/aiy/_drivers/_recorder.py#L42), but I don't have any ideas about how to fix the default config.

makelove, 2017-11-10:
```
(env) pi@aiy:~/voice-recognizer-raspi$ ./checkpoints/check_audio.py
Playing a test sound...
Did you hear the test sound? (y/n) y
When you're ready, press enter and say 'Testing, 1 2 3'...
Recording...
Playing back recorded audio...
Did you hear your own voice? (y/n) y
The audio seems to be working.
Press Enter to close...
```
it works !
Thank you!

can you help me look into Proxy SSL error ?
https://github.com/google/aiyprojects-raspbian/issues/169

dikkedimi, 2017-12-05:
I'm having similar problems. Although I've managed to get audio input and output working, running AIY on my headless Ubuntu MATE now causes this issue:
`[pulseaudio] server-lookup.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11`

drigz, 2017-12-05:
@dikkedimi Can you provide details about how you got audio input and output working? Does the check_audio.py script work?

dikkedimi, 2017-12-05:
I changed the input in _recorder.py from 'default' to 'plug:micboost' It goes through the audio test fine then. but when running the assistant_library_demo.py it returns this a while after saying OK Google!:
```
Say "OK, Google" then speak, or press Ctrl+C to quit...
[1777:1798:ERROR:mic_failure_detector.cc(120)] Channel 1 has failed. Powers -inf and -inf

```
* Audio test only works when I set my MATE installation to login to GUI/Desktop at boot (used raspi-config, probably not optimal)
* I checked and there is no .asoundrc in my home folder

also ran 
```
pactl list cards
Card #0
	Name: alsa_card.platform-soc_sound
	Driver: module-alsa-card.c
	Owner Module: 6
	Properties:
		alsa.card = "0"
		alsa.card_name = "snd_rpi_googlevoicehat_soundcar"
		alsa.long_card_name = "snd_rpi_googlevoicehat_soundcard"
		alsa.driver_name = "snd_soc_googlevoicehat_soundcard"
		device.bus_path = "platform-soc:sound"
		sysfs.path = "/devices/platform/soc/soc:sound/sound/card0"
		device.string = "0"
		device.description = "snd_rpi_googlevoicehat_soundcar"
		module-udev-detect.discovered = "1"
		device.icon_name = "audio-card"
	Profiles:
		input:analog-stereo: Analog Stereo Input (sinks: 0, sources: 1, priority: 60, available: yes)
		output:analog-stereo: Analog Stereo Output (sinks: 1, sources: 0, priority: 6000, available: yes)
		off: Off (sinks: 0, sources: 0, priority: 0, available: yes)
	Active Profile: output:analog-stereo
	Ports:
		analog-input: Analog Input (priority: 10000, latency offset: 0 usec)
			Part of profile(s): input:analog-stereo
		analog-output: Analog Output (priority: 9900, latency offset: 0 usec)
			Part of profile(s): output:analog-stereo
```

driver seems to be installed and loading (at least voor the GUI desktop). I think some ubuntu pulse audio modules are claiming exclusive ownership, but I'm too much of a noob to fix that by myself.

drigz, 2017-12-05:
assistant_library_demo.py won't use the input settings in _recorder.py. You might have more luck with assistant_grpc_demo.py, although that doesn't support "OK Google" - you have to use the button.

dikkedimi, 2017-12-05:
Thanks for the feedback,
I'll use your feedback to look into this, but as I'm not a huge linux expert (my raspbian is okay, but my ubuntu-force is not so strong), I might opt to first try baking ROS into the raspbian image i've had previous succes with and then installing google assistent on that. Running low on SD cards though.

My main reason for using Ubuntu MATE was that it offers easy installation of ROS (Robot Operating System), which is something I'm looking to integrate in my project :) I feel raspbian probably offers better performance anyway.

closed 2018-11-17 by dmitriykovalev