google/aiyprojects-raspbian #169: How to Connect Google service from China ? ERROR !

makelove, 2017-11-09:
I am a programer from Beijing China,just buy AIY kit a few days ago
Everythings is OK,but Network is big problem,
I had install Lantern https://getlantern.org/ to skip over GFW
but jump out another problems !

-
How I setup :
add this at the end of ~/.bashrc file
#proxy
export http_proxy='http://192.168.0.159:50493'
export https_proxy='https://192.168.0.159:50493'
export no_proxy=localhost,127.0.0.0/8

-
RESULT:
check_audio.desktop OK
check_wifi.desktop OK
check_cloud.desktop tell me :
Rendezvous of RPC that terminated with (StatusCode.PERMISSION_DENIED, Project piaiy-185305 (#903298171188) has billing disabled. Please enable it 
**--I know what happen**

-
~/AIY-voice-kit-python/src
assistant_grpc_demo.py is ONLY working ,fine
sometime it also ERROR
```bash
(env) pi@AIY:~/AIY-voice-kit-python/src $ ./assistant_grpc_demo.py
/home/pi/AIY-voice-kit-python/src/aiy/_drivers/_led.py:51: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(channel, GPIO.OUT)
Press the button and speak
[2017-11-08 15:53:23,220] INFO:recorder:started recording
Listening...
Traceback (most recent call last):
  File "/home/pi/AIY-voice-kit-python/env/lib/python3.4/site-packages/requests/packages/urllib3/connectionpool.py", line 594, in urlopen
    self._prepare_proxy(conn)
  File "/home/pi/AIY-voice-kit-python/env/lib/python3.4/site-packages/requests/packages/urllib3/connectionpool.py", line 810, in _prepare_proxy
    conn.connect()
  File "/home/pi/AIY-voice-kit-python/env/lib/python3.4/site-packages/requests/packages/urllib3/connection.py", line 326, in connect
    ssl_context=context)
  File "/home/pi/AIY-voice-kit-python/env/lib/python3.4/site-packages/requests/packages/urllib3/util/ssl_.py", line 325, in ssl_wrap_socket
    return context.wrap_socket(sock, server_hostname=server_hostname)
  File "/usr/lib/python3.4/ssl.py", line 364, in wrap_socket
    _context=self)
  File "/usr/lib/python3.4/ssl.py", line 577, in __init__
    self.do_handshake()
  File "/usr/lib/python3.4/ssl.py", line 804, in do_handshake
    self._sslobj.do_handshake()
ssl.SSLEOFError: EOF occurred in violation of protocol (_ssl.c:600)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/AIY-voice-kit-python/env/lib/python3.4/site-packages/requests/adapters.py", line 438, in send
    timeout=timeout
  File "/home/pi/AIY-voice-kit-python/env/lib/python3.4/site-packages/requests/packages/urllib3/connectionpool.py", line 630, in urlopen
    raise SSLError(e)
requests.packages.urllib3.exceptions.SSLError: EOF occurred in violation of protocol (_ssl.c:600)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/AIY-voice-kit-python/env/lib/python3.4/site-packages/google/auth/transport/requests.py", line 112, in __call__
    **kwargs)
  File "/home/pi/AIY-voice-kit-python/env/lib/python3.4/site-packages/requests/sessions.py", line 518, in request
    resp = self.send(prep, **send_kwargs)
  File "/home/pi/AIY-voice-kit-python/env/lib/python3.4/site-packages/requests/sessions.py", line 639, in send
    r = adapter.send(request, **kwargs)
  File "/home/pi/AIY-voice-kit-python/env/lib/python3.4/site-packages/requests/adapters.py", line 512, in send
    raise SSLError(e, request=request)
requests.exceptions.SSLError: EOF occurred in violation of protocol (_ssl.c:600)
[... 69 lines cut ...]
Traceback (most recent call last):
  File "./cloudspeech_demo.py", line 53, in <module>
    main()
  File "./cloudspeech_demo.py", line 37, in main
    text = recognizer.recognize()
  File "/home/pi/AIY-voice-kit-python/src/aiy/cloudspeech.py", line 47, in recognize
    return self._request.do_request().transcript
  File "/home/pi/AIY-voice-kit-python/src/aiy/_apis/_speech.py", line 270, in do_request
    raise Error('Exception in speech request') from exc
aiy._apis._speech.Error: Exception in speech request
Exception ignored in: <bound method Button.__del__ of <aiy._drivers._button.Button object at 0x747cead0>>
Traceback (most recent call last):
  File "/home/pi/AIY-voice-kit-python/src/aiy/_drivers/_button.py", line 57, in __del__
AttributeError: 'NoneType' object has no attribute 'cleanup'
```

-
First time ,it's working fine with assistant_grpc_demo.py,now it's ERROR
you can check my final video :https://www.youtube.com/watch?v=Y9nQisOEvf8
Please help me How to setup Network proxy problems, otherwise it will be trash of useless .
Thank you !

-
then I try to use **Ubuntu Mate** with AIY kits,but fail too.
https://github.com/google/aiyprojects-raspbian/issues/168

makelove, 2017-11-10:
I had finish moving aiyprojects  to Ubuntu Mate,
I had try all demos,still have same problems ! ONLY assistant_grpc_demo.py is working
even I setup Network Proxy in system-control
[image]

-
Why I move to Ubuntu Mate,because I think **setup Network Proxy in system-control** can avoid these problems or bugs

drigz, 2017-11-13:
> google.auth.exceptions.RefreshError: ('invalid_grant: Invalid JWT: Token must be a short-lived token (60 minutes) and in a reasonable timeframe. Check your iat and exp values and use a clock with [...]

This error normally occurs when the time is incorrect on the RPi, which you can fix with `sudo /usr/sbin/ntpdate -u pool.ntp.org`.

Regarding the other issues, the product is not launched and supported in China.

closed 2017-11-13 by drigz