google/aiyprojects-raspbian #171: What does a podcast request return?

mpember, 2017-11-11:
Does the Assistant API ever return more than just text?

e.g. If I ask "play the latest episode of <podcast>", the response from the Assistant suggests the device is about to play the podcast.  However, the spoken response is all I get.

Is it too much to assume that the API should be either playing the podcast or at the least returning the URL of the podcast and letting me pass that to a local audio player (MPD or VLC)?

manoj7410, 2019-11-14:
@mpember Is this issue still reproducible ?

mpember, 2019-11-14:
@manoj7410 

The current response is simply that of "Sorry, I can't play .... yet".

I gave up long ago on the prospect of this project ever offering additional functionality.  The only steps have been that functionality that previously behaved oddly (such as the 'good morning' response telling you it will read headlines and then stopping the response) have now resulted in polite messages that the functionality is either not available or not compatible.

I opted to simply build my own podcast/radio functionality.  In my case, I have a service that runs in the background and schedules a task to extract the details of podcasts I want to grab.  It stores the metadata and the URL of the podcast.  I then have a voice command that either lists recent podcasts or passes the URL of a requested podcast episode to VLC.  Unlike a regular podcast client, it never actually downloads an episode.  It simply streams the episodes when I want to hear them.

The relevant code can be found here:
https://github.com/mpember/aiyprojects-raspbian/blob/aiyprojects/src/modules/music.py