google/aiyprojects-raspbian #173: Question: How is the 60 minutes of usage measured in the cloudspeech.

divx118, 2017-11-12:
I think more people have this question. The 500 requests per day is clear to me. But how about the 60 minute limit per month. Is this the total server time used or ... ?
I have an interest in making a home project with cloudspeech and build a prototype for daily usage. Would I then go over the 60 minute limit if I made about 50 simple requests a day for a month? I just have no idea how the 60 minutes compares to the number of requests.

divx118, 2017-11-12:
Ok, to answer my own question quote from https://cloud.google.com/speech/pricing :

> The Cloud Speech API is priced monthly based on the amount of audio successfully processed by the service, measured in increments rounded up to 15 seconds. Pricing tiers are based on the total amount of audio processed by the service per month.

So that should mean if I make some short 3 seconds audio requests. 3600/3=1200 --> That would be about 40 3 second requests a day. Hmm...

closed 2017-11-13 by divx118

drigz, 2017-11-13:
Note that the increments are rounded up to 15 seconds, allowing 240 short
queries a month or 8 a day.

divx118, 2017-11-13:
@drigz Thanks for the info I already was afraid of that. You will then go easily over the limit, certainly during testing. For commercial use that won't be a problem, but for a simple hobby project it is for me of no use.