google/aiyprojects-raspbian #176: Using AIY Image Unit does not LED does not light up

vinniet, 2017-11-13:
Hello .. I have strange issue.  I bought the AIY kit from Microcenter and when I put it together using the packaged image, HAT does not light and button does not either.  I have bought three kits and its all the same.  I thought it was power was the issue since I have three kits but that does not seem to be the issue.

I did run the audio test where it plays audio and records it and that works.

Its so odd .. any help I would appreciate.

ensonic, 2017-11-13:
The light in the button can go in two ways, please try turning it 180°. Also please double check the colors of the wires on the button:
https://aiyprojects.withgoogle.com/voice/#assembly-guide-4-put-it-all-together

vinniet, 2017-11-13:
Hello ... Yes I tried that .. besides the led button it totally different.  You can not remove the bulb.  But I did switch wires.  Hard to believe all three kits had that issue.  Lastly the LED on the Voice HAT did not light, that should not care about the button LED.

Thanks!

drigz, 2017-11-14:
Are you running one of the demos, such as `assistant_library_demo.py`? In the new version of the image, the LED doesn't light until you run one of the scripts.

If you are running one of the scripts, this is a hardware issue (GitHub is for software issues) - you can try posting in the [RPi forums](https://www.raspberrypi.org/forums/viewforum.php?f=114) to ask for help with the assembly, or contacting the retailer if you think the kits are faulty.

vinniet, 2017-11-14:
Hello ... No I did not run any demos other then the audio test which worked.  But since then I did reboot.  I find it really hard to believe this is hardware since this is  my third kit and none of them work.

vinniet, 2017-11-15:
Hello .. The instructions say the light will go on after booting.   See below from the book:

Verify it works
Once booted, the small red LED in the center of the Voice HAT and the LED inside the arcade button should both indicate the device is running by emitting a slow pulse. If you don’t see the LED pulse, check the troubleshooting guide in the appendix.

Are you saying that the instruction are not correct?  

I am trying to set one kit up to see how complicated for my nephew.  

I did try the demo but got the following message:
./assistant_library_with_button_demo.py
You need client secrets to use the Assistant API.
Follow these instructions:
    https://developers.google.com/api-client-library/python/auth/installed-app#creatingcred
and put the file at /home/pi/assistant.json

Thanks for all your help ..

drigz, 2017-11-15:
Ah, thanks for the extra information! Unfortunately the instructions are wrong.

If you continue with the instructions about [setting up your device](https://aiyprojects.withgoogle.com/voice#users-guide), the LED should light up when you get the demo running.

The instructions are wrong on the website, too - I'm looking in to how to fix this.

vinniet, 2017-11-15:
Thanks .. makes sence why I could not get three of them working.  I am just
surprised I have not found anyone else reporting this.  Main concern is if
I can not do it school child will not be able either.

RichardBronosky, 2017-11-27:
I too experienced this "Once booted, the small red LED in the center of the Voice HAT and the LED inside the arcade button should both indicate the device is running by emitting a slow pulse." It caused my daughter and I hours of confusion. Where is the source for the instructions? I'd like to contribute a correction.

vinniet, 2017-11-27:
Its amazing with all of the kits that were sold that only two have noticed
this.

Anil-Motwani, 2017-11-29:
Three now. Spent server hours to figure this out because reading your query and observations.
It indeed is an error on the documentation!! Misleading too!!

drigz, 2017-11-29:
Thanks for sharing your feedback - we're sorry about all the confusion and wasted time here. An update to the website is on the way.

Where were you all following the instructions from? eg website, MagPi #57, or the Essentials book that comes with the kit?

vinniet, 2017-11-29:
The website, printed essential book and the PDF version.  I never got copy of the MagPi.

drigz, 2017-11-30:
OK, thanks - do you think if the website and PDF were correct, it would be enough to minimize the confusion? There's a note in the book saying that the latest instructions on the website, so I'm hoping most users would check the website when they were unable to follow the LED instructions in the printed book.

drigz, 2017-11-30:
FYI, the [instructions on the website](https://aiyprojects.withgoogle.com/voice/#assembly-guide-6-1--check-audio) are now amended.

DroneBotWorkshop, 2017-12-12:
Actually the instructions on the website have only partially been amended. The reference to the lights has been removed from the "Verify it works" section but the "Troubleshooting" section still says "If the lamp in the button doesn't light up, check that the wire colors are the same as the picture in step 7.".

I also ran into this, tested the switch with a multimeter and tested its LED with a 5-volt power supply and 330 ohm dropping resistor to verify that they are working.  And I still haven't spotted the LED on the Voice HAT board although maybe it's just my old eyes - those surface mount LED's are pretty tiny. 

Thought I was going crazy until I saw that others are experiencing teh same problem!

drigz, 2017-12-12:
Thanks for sharing, @DroneBotWorkshop. I thought it would be safe to avoid splitting the troubleshooting section in two but I guess it's still confusing.

I'll look into moving that troubleshooting item further down.

DroneBotWorkshop, 2017-12-12:
Thanks @drigz !  I think a lot of the confusion may be with the MagPi manual that is being shipped with these kits. It's a beautiful manual but it does contain a number of inaccuracies, probably not the author's fault but just because the hardware and software has been changed since it was written.

You might just want to add a note to the website indicating that the information in the MagPi manual regarding the light is not accurate. 

And for the record I finally did spot the LED on the Voice HAT board (although I would have described it as being next to the switch connector and not "at the center of the board" as the printed manual does).

RichardBronosky, 2017-12-19:
This is much better. I'm sure it will save many other people a lot of frustration. I'm giving several of these out for Christmas. I know I appreciate it.

drigz, 2017-12-19:
We've cleaned up the instructions:
https://aiyprojects.withgoogle.com/voice/#assembly-guide-7-appendix
https://aiyprojects.withgoogle.com/voice/#users-guide-3-appendix

There's one more fix on the way for the target of the link to the troubleshooting in the User's Guide, and then I'll close this issue.

Thanks so much for your feedback everyone! We also have changed our review process, so fixes like this are quicker in future.

5airkas, 2017-12-21:
I'm completely new to this (this forum and Raspberry Pi/Google AIY) and can echo previous user comments. I pulled a switcheroo (multiple times) until the LED shined brightly, much brighter than previously wired. I also encountered what seems to be a generally faulty button. When I run the button demo, I press multiple times before a gremlin activates the press function.

drigz, 2018-01-08:
Closing since the website is updated now.

@5airkas, you might need to press for longer, since the debouncing logic can sometimes ignore very short presses.

closed 2018-01-08 by drigz