google/aiyprojects-raspbian #177: Illegal instruction on running any of the demos

gomako, 2017-11-15:
I am using a Pi Zero W with one of the aiy kits, and whenever I try one of the demos that support the Pi Zero W (`assistant_grpc_demo.py` or `cloud_speech_demo.py`) I get `Illegal instruction` and the program exits.

There are no lights on the HAT or LED button (my button does not have a removable lamp).

The check audio program ran fine.

t1m0thyj, 2017-11-15:
The newest version of the Google Assistant SDK won't run on the Pi Zero W because it has an ARMv6 CPU. It only runs on Pis with an ARMv7 processor, like the Pi 2 and 3. You should still be able to set up "OK Google" detection on a Pi Zero using this project: https://github.com/warchildmd/google-assistant-hotword-raspi

drigz, 2017-11-15:
@gomako, are you using the AIY image or another distro?

gomako, 2017-11-15:
@t1m0thyj Ah ok, that would explain it.

@drigz I am using the AIY image, is there an AIY image that supports ARMv6?

drigz, 2017-11-15:
I would have expected the demos other than google_assistant_library.py to work, but I will try to reproduce the problem.

gomako, 2017-11-15:
It would appear that others have had this issue [reported here](https://www.raspberrypi.org/forums/viewtopic.php?f=114&t=197470&sid=677ba6dd29a70ba6c405c8cdd86012b3), and have got round it using an old image. I will try this tonight and report back.

gomako, 2017-11-16:
To update this, the old image dated 2017-05-03 lights the button and everything seems to work on the hardware side. The credentials flow goes as expected and the button press is registered, I get the speech transcript back in the terminal but the audio from the kit is always "Give me permission to help you"

giano, 2018-02-05:
D*rn! The Pi zero w has a very convenient form factor! Moving to Alexa then.

MartinMajewski, 2018-02-28:
What? Really? They should make a big disclaimer on the SDK page then! This is totally stupid... The best use case for my zeros gone! Damn you, Google! 👎

jtgans, 2018-02-28:
We are aware of this issue and are working to resolve it, please be patient with us. The root cause is due to a binary used in the assistant SDK, and we are working with teams internally to resolve this.

PeterMalkin, 2018-03-01:
You should grab the latest image from https://dl.google.com/dl/aiyprojects/aiyprojects-latest.img.xz, the assistant_grpc_demo.py and cloud_speech_demo.py should run just fine. 

The other assistant demos will not - the issue is that the processor on Pi 0 is so slow it cannot run hotword detect.

closed 2018-03-01 by PeterMalkin

llealgt, 2018-07-24:
I'm having this issue using a PI Zero WH using the old kit , since the new version of the kit includes a PI Zero WH i suppose there's got to be a way to run the demos using PI Zero WH, what's the difference between the new and old kit that makes this possible? and how can we use the old version kit to run all demos?

dmitriykovalev, 2018-07-24:
@llealgt Which SD-card image do you use to run demos? Have you tried the latest one from https://github.com/google/aiyprojects-raspbian/releases?

llealgt, 2018-07-24:
I think i'm using the last image, but i'll try to flash my SD-card with the latest one,after reading on the internet i got a question, does the hotword detection will work using last SD-card image version if i use a PI Zero WH , or just the other demos? And In case it does not work in Zero WH voice kit v2(which includes a Zero WH) will also only run non-hotword demos?

ahosna, 2021-02-09:
Is it possible this is still a thing ? Have zero w and google-could-texttospeech is blowing up with illegal instruction