google/aiyprojects-raspbian #178: Cloud you add tts for japanese?

garicchi, 2017-11-15:


ensonic, 2017-11-15:
The local tts is pico2wave and only supports:
en-US
en-GB
de-DE
es-ES
fr-FR
it-IT

This is the place where it is called from:
https://github.com/google/aiyprojects-raspbian/blob/voicekit/src/aiy/_drivers/_tts.py

If you know a replacement that can do japanese, you should be able to adapt the code.

garicchi, 2017-11-15:
Google AIY Kit is very nice! but text-to-speech can not speak Japanese text.
AIY kit used peco2wave for tts but it does not correspond japanese text.
raspbian can use japanese tts if install open-jtalk package which is open source and can get apt repository.

if you add following code to say function in _tts.py , AIY can correspond japanese tts.

```py
def say(player, words, lang='en-US'):
    """Say the given words with TTS.

    Args:
      player: To play the text-to-speech audio.
      words: string to say aloud.
      lang: language for the text-to-speech engine.
    """
    try:
        (fd, tts_wav) = tempfile.mkstemp(suffix='.wav', dir=TMP_DIR)
    except IOError:
        logger.exception('Using fallback directory for TTS output')
        (fd, tts_wav) = tempfile.mkstemp(suffix='.wav')
    os.close(fd)
    if lang=='ja-JP':
        try:
            tts_text = os.path.join(TMP_DIR,'tts.txt')
            with open(tts_text,'w') as f:
                f.write(words)
            subprocess.call([
                'open_jtalk',
                '-m','/usr/share/hts-voice/nitech-jp-atr503-m001/nitech_jp_atr503_m001.htsvoice',
                '-x','/var/lib/mecab/dic/open-jtalk/naist-jdic/',
                '-ow',tts_wav,
                tts_text
            ])
            player.play_wav(tts_wav)
        except FileNotFoundError as e:
            logger.error('open_jtalk not found. please install following command -> sudo apt-get install open-jtalk open-jtalk-mecab-naist-jdic hts-voice-nitech-jp-atr503-m001')
        finally:
            os.unlink(tts_wav)
            os.unlink(tts_text)
    else:
        words = '<volume level="60"><pitch level="130">%s</pitch></volume>' % words
        try:
            subprocess.call(['pico2wave', '--lang', lang, '-w', tts_wav, words])
            player.play_wav(tts_wav)
        finally:
            os.unlink(tts_wav)

```

Can I submit pull request?

ensonic, 2017-11-15:
This repo is mostly meant as examples and we would like to keep it somewhat simple. In the above snippet I'd probably refactor it to have the player.play_wav(tts_wav) + os.unlink(tts_wav) not duplicated. Not sure what others think.

garicchi, 2017-11-16:
how is this code?

```py
def synthesize(words,lang,tts_wav,tts_text):
    tts_commands = {
        'default':[
            'pico2wave',
            '--lang',lang,
            '-w', tts_wav,
            '<volume level="60"><pitch level="130">%s</pitch></volume>' % words
        ],
        'ja-JP':[
                'open_jtalk',
                '-m','/usr/share/hts-voice/nitech-jp-atr503-m001/nitech_jp_atr503_m001.htsvoice',
                '-x','/var/lib/mecab/dic/open-jtalk/naist-jdic/',
                '-ow',tts_wav,
                tts_text
            ]
    }
    command = None
    if not lang in tts_commands:
        command = tts_commands['default']
    else:
        command = tts_commands['ja-JP']
        
    subprocess.call(command)
        

def say(player, words, lang='en-US'):
    """Say the given words with TTS.

    Args:
      player: To play the text-to-speech audio.
      words: string to say aloud.
      lang: language for the text-to-speech engine.
    """
    try:
        (fd_wav, tts_wav) = tempfile.mkstemp(suffix='.wav', dir=TMP_DIR)
        (fd_text, tts_text) = tempfile.mkstemp(suffix='.txt', dir=TMP_DIR)
    except IOError:
        logger.exception('Using fallback directory for TTS output')
        (fd_wav, tts_wav) = tempfile.mkstemp(suffix='.wav', dir=TMP_DIR)
        (fd_text, tts_text) = tempfile.mkstemp(suffix='.txt', dir=TMP_DIR)
    os.close(fd_wav)
    os.close(fd_text)

    with open(tts_text,'w') as f:
        f.write(words)
    try:
        synthesize(words,lang,tts_wav,tts_text)
        player.play_wav(tts_wav)
        
    finally:
        os.unlink(tts_wav)
        os.unlink(tts_text)

```

mscdo, 2018-04-03:
How can i test or use the tts? i have to create a file that executes de tts.py?