google/aiyprojects-raspbian #180: Cannot erase shopping list

MarsNL, 2017-11-18:
Hi, I added items to the shopping list but
cannot remove them as the AIY needs to have
an connection with the Google Home app.
Is there any way to clear it?

drigz, 2017-11-20:
Not as far as I'm aware, see #6.

You could try [the support forums](https://github.com/google/aiyprojects-raspbian/blob/voicekit/README.md#support) to see if there are known workarounds.

sheridat, 2017-11-20:
Hi, go to [](https://express.google.com/  ) and sign in (if your not using chrome and aren't signed in).
On the left hand side menu you will see "list" -click on it
You can then delete whichever items you want.

drigz, 2017-11-23:
Good suggestion, @sheridat.

Closing as a duplicate of #6.

dmitriykovalev, 2018-01-09:
Duplicate of #6

closed 2018-01-09 by dmitriykovalev