google/aiyprojects-raspbian #181: How to set up Italian Language for Google Assistant AIYProject ?

devilmind, 2017-11-22:
hello,
Google Assistant API has recently released the Italian language.
How to set up Italian Language for Google Assistant AIYProject ?
Can you share the steps to do that please?

Thks

Regards

Antonio

javporris, 2017-11-22:
Hi,

I need the same instructions for spanish language. Is it possible?

Thanks for your work.

javporris, 2017-11-22:
Hola,

Necesito las mismas instrucciones para el idioma español. ¿Es posible?

Gracias por tu trabajo.

javporris, 2017-11-22:
Somebody has try to include this in the demo file?

aiy.i18n.set_language_code('it-IT')
or
aiy.i18n.set_language_code('es-ES')

I'm waiting the kit, then I can't try it.

sheridat, 2017-11-22:
Hi,

This whole business about setting aiy.i18n.set_language_code('language code') is very misunderstood.

Setting this only has an effect on the TTS program that is used to say words when you program a local command. 

IT DOES NOT have any effect whatsoever on the Google Assistant - responses from the Google Assistant will remain as an American female voice whatever you set this value to.

To test the effect of setting the aiy.I18 etc variable just do the following in the Start dev terminal 

pico2wave --wave sample.wav -l fr-FR "Here is a sample, using the PICO text to speech engine." && aplay sample.wav

or

pico2wave --wave sample.wav -l it-IT "Here is a sample, using the PICO text to speech engine." && aplay sample.wav 

but remember it is only relevant to local commands that you program

devilmind81, 2017-11-22:
Are you telling that there isn' t a way to switch the existing Google assistente api to another available language ? Only TTS One for local commands?

sheridat, 2017-11-22:
I'm advising you that your anticipation regarding the effect of setting aiy.18 may be unfounded.

If you run any of the files in the src folder that start with assistant_library and say for example 
1. What is the time;
2. What is today's date;
3. Will it rain today;
4. Tell me a joke
etc etc etc 
 the Google Assistant on this device will reply in english irrespective of the setting of the aiy.I18 etc variable

Why don't you give it a go?

devilmind81, 2017-11-22:
Ok understood, you are probably replying to javaporris user and I want thank you for the clarification. I wiuld fiundbthe way to translate all in italian language.

sheridat, 2017-11-22:
The cloudspeech facility seems to understand multiple languages - i don't know anything about it - you have to go through the signing up for billing route to use it.

devilmind, 2017-11-22:
ok I will do some tests and let you know....meanwhile I hope that someone has already done this and can help us here in order to get it working with a different language.

javporris, 2017-11-22:
Sorry for the confusion, I just wanted to share what I had read in the documentation. Thanks for the clarification.

I will remain attentive to the progress of this issue and when I have the kit I will perform tests.

devilmind, 2017-11-22:
Ok keep updated in the progresses too, thank you

drigz, 2017-11-23:
The Google Assistant API/SDK only supports English. You can stay up-to-date on new changes on [this list](https://groups.google.com/forum/#!forum/google-assistant-sdk-updates).

If you want to use other languages with AIY, you'll have to [change to the Cloud Speech API](https://aiyprojects.withgoogle.com/voice/#makers-guide-3-1--change-to-the-cloud-speech-api) and [set the language](https://github.com/google/aiyprojects-raspbian/issues/181#issuecomment-346394638) as described by @javporris.

closed 2017-11-23 by drigz

Enzoinventor, 2017-12-22:
@devilmind https://github.com/googlesamples/assistant-sdk-python/issues/79

rodrigob, 2018-09-04:
Based on [this post](https://groups.google.com/forum/#!topic/google-assistant-sdk-updates/zMVgRYQpgpw), it seems that new languages are now supported ?

robferrero, 2018-09-27:
I've redownloaded the latest SD image from here :
https://dl.google.com/dl/aiyprojects/aiyprojects-latest.img.xz

the file is called 
aiyprojects-2018-08-03.img.xz

and I've succesfully tested italian speaking language with Assistant _gRPC_python.py on KIT-Voice v1
adding only  aiy.i18n.set_language_code('it-IT')

in the code of src/examples/voice/assistant_grpc_demo.py 

    status_ui.status('starting')
    aiy.i18n.set_language_code('it-IT')
    assistant = aiy.assistant.grpc.get_assistant()

dmitriykovalev, 2018-10-01:
@robferrero Thanks for checking this!

nyxover, 2018-12-17:
Merci ça marche aussi en français.
aiy.i18n.set_language_code('fr-FR')

a condition de bien respecter l'indentation

dimgod59, 2019-01-15:
Excuse me but i'm a beginner ... How do you change this piece of code? please
Can you give me the commands or explain to me how to do

dmitriykovalev, 2019-01-16:
@nyxover This is the old way of doing this. Please take a look at `language_code` parameter at https://aiyprojects.readthedocs.io/en/latest/aiy.assistant.html#module-aiy.assistant.library from the latest SD card image https://github.com/google/aiyprojects-raspbian/releases.

docdiesel, 2019-01-25:
Furthermore, the latest version takes the locale settings as default. So set the language with raspi-config and after next reboot the script will use that as standard.