google/aiyprojects-raspbian #182: Check Cloud closes the window

javporris, 2017-11-22:
Hi,

I have installed everything as indicated here:
https://aiyprojects.withgoogle.com/voice

My problem comes when launching "check cloud" on the desktop. The terminal opens and closes automatically without being able to see if the check is correct. "Check voice" and "Check wifi" work well.

Then I run "src / assistant_library_demo.py", apparently it does it well but does not respond to "ok google". The same thing happens with "src / assistant_grpc_demo.py", I press the button and it does not do anything and if I leave it pressed the screen turns off.
I think the error may come from the "Check cloud", but as it closes automatically I really do not know if there is an error or not. I followed all the steps to generate the .json file and put it in "/ home / pi".

Where can the error be?

Greetings and many thanks.

sheridat, 2017-11-22:
two things:
1. If Check cloud closes automatically that means you have set it up correctly - it only prints something out if there is a problem.
2. You are clicking on Start dev terminal and then typing in src/assistant_library_demo.py

you should see this
Build info: Tue Sep 12 00:36:20 UTC 2017 @ 3955cac
Dev terminal is ready! Hint: enter src/assistant_library_demo.py
(env) pi@raspberrypi:~/AIY-voice-kit-python $ src/assistant_library_demo.py
Say "OK, Google" then speak, or press Ctrl+C to quit...

javporris, 2017-11-22:
Yes I see that message, but I say Ok Google and nothing happend...

sheridat, 2017-11-22:
What happens if you run src/assistant_library_with_button_demo.py and press the button and then say something does the light in the button light up?

javporris, 2017-11-22:
After I execute the with button version works!

closed 2017-11-22 by javporris