google/aiyprojects-raspbian #186: Convert AIY Voice Kit into a battery powered megaphone with offline voice command

HaleyMitt, 2017-11-27:
I have and AIY Voice Kit setup with Google Assistant API. I'm trying to accomplish three things I need help with. 

1. Turn voice input (into the microphone) to sound output (out of the speaker). Essentially I want it to work like a megaphone, to make voice louder. 
2. Use voice command to turn on and off the device as well as adjust the volume (these need to be local to the device so they can be accessed without an internet connection). 
3. Connect to a portable power source such as a lithium ion battery. 

If you can provide help with any of these three I'd greatly appreciate it. Cheers!

divx118, 2017-11-27:
@HaleyMitt I don't think this project will be right for you since it uses a server connection for voice recognition. So you would always need an internet connection.
There are however some other ways to get voice recognition on your rpi without the need for internet. IMO the Voicekit hardware would be good for those projects just do a google with "raspberry pi voice recognition offline" that would give you some results to get started.
1) some alsa mixer settings in `.asound.conf` should accomplish that, just google for some general linux tutorials or questions on that topic.
3) The easiest way is to get a big power bank with which you also can charge a phone. For example http://www.techradar.com/news/top-portable-chargers

sheridat, 2017-11-27:
Number 2. Can I graciously point out that turning on the device using a voice command when the processor is not running is illogical. There is a mechanical/electrical way of achieving it. 
There are two pads on the Raspberry PI marked Run - if you short them it will start the pi. Normally people solder two header onto those pads and connect a button. I've done that with my voice kit (as an aside you can also add a reboot switch by connecting a button to a gpio and a ground - google it)

@divx118 - wouldn't @HaleyMitt be able to shut it down with a local command such as the one in assistant_library_with_local_commands_demo.py ???? even though it isn't connected to the internet - I've always assumed that you could use local commands without a connection - but happy to be instructed otherwise.

drigz, 2017-11-27:
I'm closing this issue as the [forums](https://www.raspberrypi.org/forums/viewforum.php?f=114&sid=cfe5e516ad165805e39a22afa8b02a3f) are the right place for this question. GitHub issues are for bugs in the software.

I will add - I think you'll run into issues with feedback. ie a noise occurs, the mic hears it, the speakers make it louder, the mic hears the louder version and and the speakers make it even louder. This is the cause of the unpleasant hum you hear when people hold a microphone too near a speaker. Some guidance from [this website](http://www.campaigngears.com/can_you_hear_me_now):

> The thing about megaphones is that the speaker is so close to the microphone. That means two things: the gain on the microphone is set very low so the user has to yell in order to be properly projected and the speaker is very directional (in order to avoid feedback) so if you crowd is widespread, you need to move the megaphone in different directions and repeat yourself a few times.

closed 2017-11-27 by drigz

divx118, 2017-11-27:
@sheridat No she wouldn't, for the initial speech recognition there is a connection to the server needed which gives you the text you intercept and then handle it locally.

sheridat, 2017-11-27:
Cheers

shivasiddharth, 2017-11-28:
@HaleyMitt ,
For 1.  refer to this. https://www.raspberrypi-spy.co.uk/2017/03/amplified-voice-changer-using-a-raspberry-pi-zero/

fscii, 2018-06-07:
hate to chime in late but alexa, google etc. have "wake words" in other words it can hear a command and recognize it locally.  It would be nice to add this feature but with say a handful of words such as
<wake word for specific device><basic command><one parameter>
e.g. Pi volume 10

I am thinking instead of premade command you would type into a config file the wake word and it can process it via mycroft google, alexa and return whatevers needed to activate this as the wake word.  Then, ditto for the few commands to operate it (on/off/dimmer) and the variable (50%).

This way there's no premade set of audio, but the data is essentially created by you ahead of time, and the results saved to an sd or thumb drive and installed into the device in question.  Clearly it should also have a way to go into standby mode with just the most minimal power level to the mic - which the voltage from the mic can be used to ONLY listen for the wake word and then go into full power.  Just like the voice assistants do now.