google/aiyprojects-raspbian #188: assistant stop working after losing internet connection temporarily

daverck, 2017-11-27:
Hi,

I have this issue with my raspberry.

When I ask something and the pi doesn't have internet, google tell me "there was a glitch", "please try again in a few seconds", ect.
That's OK, but the when my internet connection come back up, google assistant keep saying it.
I have a poor internet connection so this append quite often.

Is there any solution for this ?

Here are the service logs :
```
root@serck-pi:~# journalctl -u voice-recognizer.service -n 60 -f
-- Logs begin at ven 2017-11-24 00:56:00 CET. --
nov 26 15:24:24 serck-pi python3[16429]: File "/home/pi/scripts/personal_assistant/voice-recognizer-raspi/env/lib/python3.4/site-packages/requests/packages/urllib3/connectionpool.py", line 345, in _make_request
nov 26 15:24:24 serck-pi python3[16429]: self._validate_conn(conn)
nov 26 15:24:24 serck-pi python3[16429]: File "/home/pi/scripts/personal_assistant/voice-recognizer-raspi/env/lib/python3.4/site-packages/requests/packages/urllib3/connectionpool.py", line 844, in _validate_conn
nov 26 15:24:24 serck-pi python3[16429]: conn.connect()
nov 26 15:24:24 serck-pi python3[16429]: File "/home/pi/scripts/personal_assistant/voice-recognizer-raspi/env/lib/python3.4/site-packages/requests/packages/urllib3/connection.py", line 284, in connect
nov 26 15:24:24 serck-pi python3[16429]: conn = self._new_conn()
nov 26 15:24:24 serck-pi python3[16429]: File "/home/pi/scripts/personal_assistant/voice-recognizer-raspi/env/lib/python3.4/site-packages/requests/packages/urllib3/connection.py", line 150, in _new_conn
nov 26 15:24:24 serck-pi python3[16429]: self, "Failed to establish a new connection: %s" % e)
nov 26 15:24:24 serck-pi python3[16429]: requests.packages.urllib3.exceptions.NewConnectionError: <requests.packages.urllib3.connection.VerifiedHTTPSConnection object at 0x73483870>: Failed to establish a new connection: [Errno 113] No route to host
nov 26 15:24:24 serck-pi python3[16429]: During handling of the above exception, another exception occurred:
nov 26 15:24:24 serck-pi python3[16429]: Traceback (most recent call last):
nov 26 15:24:24 serck-pi python3[16429]: File "/home/pi/scripts/personal_assistant/voice-recognizer-raspi/env/lib/python3.4/site-packages/requests/adapters.py", line 438, in send
nov 26 15:24:24 serck-pi python3[16429]: timeout=timeout
nov 26 15:24:24 serck-pi python3[16429]: File "/home/pi/scripts/personal_assistant/voice-recognizer-raspi/env/lib/python3.4/site-packages/requests/packages/urllib3/connectionpool.py", line 649, in urlopen
nov 26 15:24:24 serck-pi python3[16429]: _stacktrace=sys.exc_info()[2])
nov 26 15:24:24 serck-pi python3[16429]: File "/home/pi/scripts/personal_assistant/voice-recognizer-raspi/env/lib/python3.4/site-packages/requests/packages/urllib3/util/retry.py", line 388, in increment
nov 26 15:24:24 serck-pi python3[16429]: raise MaxRetryError(_pool, url, error or ResponseError(cause))
nov 26 15:24:24 serck-pi python3[16429]: requests.packages.urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='accounts.google.com', port=443): Max retries exceeded with url: /o/oauth2/token (Caused by NewConnectionError('<requests.packages.urllib3.connection.VerifiedHTTPSConnection object at 0x73483870>: Failed to establish a new connection: [Errno 113] No route to host',))
nov 26 15:24:24 serck-pi python3[16429]: During handling of the above exception, another exception occurred:
nov 26 15:24:24 serck-pi python3[16429]: Traceback (most recent call last):
nov 26 15:24:24 serck-pi python3[16429]: File "/usr/local/lib/python3.4/dist-packages/google/auth/transport/requests.py", line 118, in __call__
nov 26 15:24:24 serck-pi python3[16429]: **kwargs)
nov 26 15:24:24 serck-pi python3[16429]: File "/home/pi/scripts/personal_assistant/voice-recognizer-raspi/env/lib/python3.4/site-packages/requests/sessions.py", line 518, in request
nov 26 15:24:24 serck-pi python3[16429]: resp = self.send(prep, **send_kwargs)
nov 26 15:24:24 serck-pi python3[16429]: File "/home/pi/scripts/personal_assistant/voice-recognizer-raspi/env/lib/python3.4/site-packages/requests/sessions.py", line 639, in send
nov 26 15:24:24 serck-pi python3[16429]: r = adapter.send(request, **kwargs)
nov 26 15:24:24 serck-pi python3[16429]: File "/home/pi/scripts/personal_assistant/voice-recognizer-raspi/env/lib/python3.4/site-packages/requests/adapters.py", line 502, in send
nov 26 15:24:24 serck-pi python3[16429]: raise ConnectionError(e, request=request)
nov 26 15:24:24 serck-pi python3[16429]: requests.exceptions.ConnectionError: HTTPSConnectionPool(host='accounts.google.com', port=443): Max retries exceeded with url: /o/oauth2/token (Caused by NewConnectionError('<requests.packages.urllib3.connection.VerifiedHTTPSConnection object at 0x73483870>: Failed to establish a new connection: [Errno 113] No route to host',))
nov 26 15:24:24 serck-pi python3[16429]: The above exception was the direct cause of the following exception:
nov 26 15:24:24 serck-pi python3[16429]: Traceback (most recent call last):
nov 26 15:24:24 serck-pi python3[16429]: File "/usr/lib/python3.4/threading.py", line 920, in _bootstrap_inner
nov 26 15:24:24 serck-pi python3[16429]: self.run()
nov 26 15:24:24 serck-pi python3[16429]: File "/usr/lib/python3.4/threading.py", line 1186, in run
nov 26 15:24:24 serck-pi python3[16429]: self.function(*self.args, **self.kwargs)
nov 26 15:24:24 serck-pi python3[16429]: File "/home/pi/scripts/personal_assistant/voice-recognizer-raspi/env/lib/python3.4/site-packages/google/assistant/library/auth_helpers.py", line 65, in _handle_refresh_timer
nov 26 15:24:24 serck-pi python3[16429]: self._credentials.refresh(http_request)
nov 26 15:24:24 serck-pi python3[16429]: File "/usr/local/lib/python3.4/dist-packages/google/oauth2/credentials.py", line 117, in refresh
nov 26 15:24:24 serck-pi python3[16429]: self._client_secret))
nov 26 15:24:24 serck-pi python3[16429]: File "/usr/local/lib/python3.4/dist-packages/google/oauth2/_client.py", line 191, in refresh_grant
nov 26 15:24:24 serck-pi python3[16429]: response_data = _token_endpoint_request(request, token_uri, body)
nov 26 15:24:24 serck-pi python3[16429]: File "/usr/local/lib/python3.4/dist-packages/google/oauth2/_client.py", line 105, in _token_endpoint_request
nov 26 15:24:24 serck-pi python3[16429]: method='POST', url=token_uri, headers=headers, body=body)
nov 26 15:24:24 serck-pi python3[16429]: File "/usr/local/lib/python3.4/dist-packages/google/auth/transport/requests.py", line 122, in __call__
nov 26 15:24:24 serck-pi python3[16429]: six.raise_from(new_exc, caught_exc)
nov 26 15:24:24 serck-pi python3[16429]: File "<string>", line 3, in raise_from
nov 26 15:24:24 serck-pi python3[16429]: google.auth.exceptions.TransportError: HTTPSConnectionPool(host='accounts.google.com', port=443): Max retries exceeded with url: /o/oauth2/token (Caused by NewConnectionError('<requests.packages.urllib3.connection.VerifiedHTTPSConnection object at 0x73483870>: Failed to establish a new connection: [Errno 113] No route to host',))
nov 27 04:33:54 serck-pi python3[16429]: ALSA lib pcm.c:7843:(snd_pcm_recover) overrun occurred
nov 27 07:27:09 serck-pi python3[16429]: [2017-11-27 07:27:09,622] INFO:root:ON_CONVERSATION_TURN_STARTED
nov 27 07:27:09 serck-pi python3[16429]: [2017-11-27 07:27:09,668] INFO:main:listening...
nov 27 07:27:09 serck-pi python3[16429]: [16429:24982:ERROR:speech_processing_task.cc(646)] Detected communication error: Downstream closed prematurely.
nov 27 07:27:09 serck-pi python3[16429]: [16429:24982:ERROR:assistant_output_handler.cc(589)] Failed with a general speech recognition error
nov 27 07:27:09 serck-pi python3[16429]: [2017-11-27 07:27:09,935] INFO:root:ON_ASSISTANT_ERROR:
nov 27 07:27:09 serck-pi python3[16429]: {'is_fatal': False}
nov 27 07:27:09 serck-pi python3[16429]: [2017-11-27 07:27:09,941] INFO:root:ON_RESPONDING_STARTED:
nov 27 07:27:09 serck-pi python3[16429]: {'is_error_response': True}
nov 27 07:27:13 serck-pi python3[16429]: [2017-11-27 07:27:13,806] INFO:root:ON_RESPONDING_FINISHED
nov 27 07:27:13 serck-pi python3[16429]: [2017-11-27 07:27:13,809] INFO:root:ON_CONVERSATION_TURN_FINISHED:
nov 27 07:27:13 serck-pi python3[16429]: {'with_follow_on_turn': False}
nov 27 07:27:13 serck-pi python3[16429]: [2017-11-27 07:27:13,813] INFO:main:ready...
```

drigz, 2017-11-29:
Thanks for reporting this! Could you try and reproduce this by disabling and enabling the Raspberry Pi's WiFi, instead of waiting for your connection to drop out?

If that behaves the same way, you can [file an issue for the Assistant SDK](https://github.com/googlesamples/assistant-sdk-python/issues) including instructions on how to reproduce.

From the logs, it looks like the error is in an authentication thread created by the Assistant SDK, so that's almost certainly the place to fix it.

daverck, 2017-12-05:
I think internet has to be down for a relatively long time.
I cannot reproduce this right now bc I have a roommate and I don't want to annoy him.

Although I had the same error a few day back and that was also after I lost internet connection :

```
déc 01 17:51:06 serck-pi python3[5258]: Exception in thread Thread-46:
déc 01 17:51:06 serck-pi python3[5258]: Traceback (most recent call last):
déc 01 17:51:06 serck-pi python3[5258]: File "/home/pi/scripts/personal_assistant/voice-recognizer-raspi/env/lib/python3.4/site-packages/requests/packages/urllib3/connection.py", line 141, in _new_conn
déc 01 17:51:06 serck-pi python3[5258]: (self.host, self.port), self.timeout, **extra_kw)
déc 01 17:51:06 serck-pi python3[5258]: File "/home/pi/scripts/personal_assistant/voice-recognizer-raspi/env/lib/python3.4/site-packages/requests/packages/urllib3/util/connection.py", line 83, in create_conn        ection
déc 01 17:51:06 serck-pi python3[5258]: raise err
déc 01 17:51:06 serck-pi python3[5258]: File "/home/pi/scripts/personal_assistant/voice-recognizer-raspi/env/lib/python3.4/site-packages/requests/packages/urllib3/util/connection.py", line 73, in create_conn        ection
déc 01 17:51:06 serck-pi python3[5258]: sock.connect(sa)
déc 01 17:51:06 serck-pi python3[5258]: OSError: [Errno 113] No route to host
déc 01 17:51:06 serck-pi python3[5258]: During handling of the above exception, another exception occurred:
déc 01 17:51:06 serck-pi python3[5258]: Traceback (most recent call last):
déc 01 17:51:06 serck-pi python3[5258]: File "/home/pi/scripts/personal_assistant/voice-recognizer-raspi/env/lib/python3.4/site-packages/requests/packages/urllib3/connectionpool.py", line 600, in urlopen
déc 01 17:51:06 serck-pi python3[5258]: chunked=chunked)
déc 01 17:51:06 serck-pi python3[5258]: File "/home/pi/scripts/personal_assistant/voice-recognizer-raspi/env/lib/python3.4/site-packages/requests/packages/urllib3/connectionpool.py", line 345, in _make_reque        st
déc 01 17:51:06 serck-pi python3[5258]: self._validate_conn(conn)
déc 01 17:51:06 serck-pi python3[5258]: File "/home/pi/scripts/personal_assistant/voice-recognizer-raspi/env/lib/python3.4/site-packages/requests/packages/urllib3/connectionpool.py", line 844, in _validate_c        onn
déc 01 17:51:06 serck-pi python3[5258]: conn.connect()
déc 01 17:51:06 serck-pi python3[5258]: File "/home/pi/scripts/personal_assistant/voice-recognizer-raspi/env/lib/python3.4/site-packages/requests/packages/urllib3/connection.py", line 284, in connect
déc 01 17:51:06 serck-pi python3[5258]: conn = self._new_conn()
déc 01 17:51:06 serck-pi python3[5258]: File "/home/pi/scripts/personal_assistant/voice-recognizer-raspi/env/lib/python3.4/site-packages/requests/packages/urllib3/connection.py", line 150, in _new_conn
déc 01 17:51:06 serck-pi python3[5258]: self, "Failed to establish a new connection: %s" % e)
déc 01 17:51:06 serck-pi python3[5258]: requests.packages.urllib3.exceptions.NewConnectionError: <requests.packages.urllib3.connection.VerifiedHTTPSConnection object at 0x74973770>: Failed to establish a new         connection: [Errno 113] No route to host
déc 01 17:51:06 serck-pi python3[5258]: During handling of the above exception, another exception occurred:
déc 01 17:51:06 serck-pi python3[5258]: Traceback (most recent call last):
déc 01 17:51:06 serck-pi python3[5258]: File "/home/pi/scripts/personal_assistant/voice-recognizer-raspi/env/lib/python3.4/site-packages/requests/adapters.py", line 438, in send
déc 01 17:51:06 serck-pi python3[5258]: timeout=timeout
déc 01 17:51:06 serck-pi python3[5258]: File "/home/pi/scripts/personal_assistant/voice-recognizer-raspi/env/lib/python3.4/site-packages/requests/packages/urllib3/connectionpool.py", line 649, in urlopen
déc 01 17:51:06 serck-pi python3[5258]: _stacktrace=sys.exc_info()[2])
déc 01 17:51:06 serck-pi python3[5258]: File "/home/pi/scripts/personal_assistant/voice-recognizer-raspi/env/lib/python3.4/site-packages/requests/packages/urllib3/util/retry.py", line 388, in increment
déc 01 17:51:06 serck-pi python3[5258]: raise MaxRetryError(_pool, url, error or ResponseError(cause))
déc 01 17:51:06 serck-pi python3[5258]: requests.packages.urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='accounts.google.com', port=443): Max retries exceeded with url: /o/oauth2/token (Caused b        y NewConnectionError('<requests.packages.urllib3.connection.VerifiedHTTPSConnection object at 0x74973770>: Failed to establish a new connection: [Errno 113] No route to host',))
déc 01 17:51:06 serck-pi python3[5258]: During handling of the above exception, another exception occurred:
déc 01 17:51:06 serck-pi python3[5258]: Traceback (most recent call last):
déc 01 17:51:06 serck-pi python3[5258]: File "/usr/local/lib/python3.4/dist-packages/google/auth/transport/requests.py", line 118, in __call__
déc 01 17:51:06 serck-pi python3[5258]: **kwargs)
déc 01 17:51:06 serck-pi python3[5258]: File "/home/pi/scripts/personal_assistant/voice-recognizer-raspi/env/lib/python3.4/site-packages/requests/sessions.py", line 518, in request
déc 01 17:51:06 serck-pi python3[5258]: resp = self.send(prep, **send_kwargs)
déc 01 17:51:06 serck-pi python3[5258]: File "/home/pi/scripts/personal_assistant/voice-recognizer-raspi/env/lib/python3.4/site-packages/requests/sessions.py", line 639, in send
déc 01 17:51:06 serck-pi python3[5258]: r = adapter.send(request, **kwargs)
[... cut ...]
déc 01 17:51:07 serck-pi python3[5258]: six.raise_from(new_exc, caught_exc)
déc 01 17:51:07 serck-pi python3[5258]: File "<string>", line 3, in raise_from
déc 01 17:51:07 serck-pi python3[5258]: google.auth.exceptions.TransportError: HTTPSConnectionPool(host='accounts.google.com', port=443): Max retries exceeded with url: /o/oauth2/token (Caused by NewConnecti        onError('<requests.packages.urllib3.connection.VerifiedHTTPSConnection object at 0x74973770>: Failed to establish a new connection: [Errno 113] No route to host',))
```

chrisrwilkes, 2017-12-09:
Hi, I've also experienced this. I've got a dodgy wifi area where the voice kit sits and if the connection drops then the assistant hangs.

LondonSi72, 2017-12-09:
My local Internet provider (BT) have provided a useless service recently with the internet dropping regularly. If I try to use the Assistant even after the Internet service is back up, the Assistant fails with a 'hmm, something went wrong, please try again in a few seconds'. I have to reboot to correct it. I'm thinking of running a service to check the internet access and rebooting if it detects it's dropped...

divx118, 2017-12-10:
@LondonSi72  If you just restart the assistant instead of a reboot does that work?

edisoncxr, 2018-01-22:
+1 follow. same issue... any update on this?
Jan 22 21:53:17 raspberrypi bash[3193]: method='POST', url=token_uri, headers=headers, body=body)
Jan 22 21:53:17 raspberrypi bash[3193]: File "/home/pi/AIY-voice-kit-python/env/lib/python3.4/site-packages/google/auth/transport/requests.py", line 115, in __call__
Jan 22 21:53:17 raspberrypi bash[3193]: raise exceptions.TransportError(exc)
Jan 22 21:53:17 raspberrypi bash[3193]: google.auth.exceptions.TransportError: HTTPSConnectionPool(host='accounts.google.com', port=443): Max retries exceeded with url: /o/oauth2/token (Caused b...
Jan 22 22:06:21 raspberrypi bash[3193]: [3193:3209:ERROR:speech_recognition_activity.cc(452)] Detected communication error: Request had invalid authentication credentials. Expected OAuth 2 acces...
Jan 22 22:06:21 raspberrypi bash[3193]: [3193:3209:ERROR:assistant_output_handler.cc(618)] Failed with a general speech recognition error
Jan 22 23:42:13 raspberrypi bash[3193]: [3193:3209:ERROR:speech_recognition_activity.cc(452)] Detected communication error: Request had invalid authentication credentials. Expected OAuth 2 acces...
Jan 22 23:42:13 raspberrypi bash[3193]: [3193:3209:ERROR:assistant_output_handler.cc(618)] Failed with a general speech recognition error
Jan 22 23:42:21 raspberrypi bash[3193]: [3193:3209:ERROR:speech_recognition_activity.cc(452)] Detected communication error: Request had invalid authentication credentials. Expected OAuth 2 acces...
Jan 22 23:42:21 raspberrypi bash[3193]: [3193:3209:ERROR:assistant_output_handler.cc(618)] Failed with a general speech recognition error

LondonSi72, 2018-02-01:
@divx118 Sorry - missed your query.  Yes restarting the service also works.

divx118, 2018-02-01:
@LondonSi72 Well I know it will be a workaround, but a simple shell script that pings www.google.com every 15 or so seconds and if it fails just restart the service.

LondonSi72, 2018-02-01:
@divx118 - yes. That's kind of what I was looking at doing as a workaround - I was just hoping for something programmatic :)

divx118, 2018-02-01:
@LondonSi72 Didn't test it, but something like below should do it. 

```sh
#!/bin/sh
OFFLINE=false
while true; do 
    if ping www.google.com -c 1|grep -q time; then
        echo online
        # Only restart service when there was an offline
        if OFFLINE;
            systemctl restart voice-recognizer.service
            OFFLINE=false
        fi
    else
        # Set restart service flag
        OFFLINE=true
        echo offline
    fi
    sleep 15
done
```

daverck, 2018-02-03:
I corrected it a bit (was missing **then** and **$**)

```
#!/bin/sh
#
### BEGIN INIT INFO
# Provides:          voice-recognizer-restart 
# Required-Start:    $syslog $remote_fs
# Required-Stop:     $syslog $remote_fs
# Should-Start:      $local_fs
# Should-Stop:       $local_fs
# Default-Start:     2 3 4 5
# Default-Stop:      0 1 6
# Short-Description: restart the voice-recognizer service if internet connection come back up
# Description:       restart the voice-recognizer service if internet connection come back up
#                    broadcast address on the local LAN 
### END INIT INFO

OFFLINE=false

while true;
do 
    if ping www.google.com -c 1|grep -q time;
    then
        echo online
        # Only restart service when there was an offline
	if $OFFLINE;
        then
            systemctl restart voice-recognizer.service
            OFFLINE=false
        fi
    else
        # Set restart service flag
        OFFLINE=true
        echo offline
    fi
    sleep 15
done

```

daverck, 2018-02-03:
It would be better if the assistant handled losing internet connection correctly but that will do for now.

divx118, 2018-02-04:
@daverck oops sorry, corrected in the above code. I agree with you that the api should handle this. Thanks for keeping the issue open.

LondonSi72, 2018-02-06:
As I'm relatively new to Linux and creating init.d or rc.local or systemd started tasks is still a bit of a mystery, for me the simplest way of implementing your (really useful) script was to create it as an executable bash script and add it to crontab to start at reboot.

I removed the INIT related content, updated the systemctl command to run with sudo, saved the script as **restart-voice-recognizer.sh**,  made it executable (**chmod +x**)  and updated crontab as follows:

`crontab -e`

I chose nano as my editor of choice and added:

`@reboot /home/pi/restart-voice-recognizer.sh`

Here's the script I used (for some reason it's losing the formatting when I add this here - better to cut and paste @daverck's code above and make the changes I did):

#!/bin/sh

OFFLINE=false

while true;
do 
    if ping www.google.com -c 1|grep -q time;
    then
        echo online
        # Only restart service when there was an offline
	if $OFFLINE;
        then
            sudo systemctl restart voice-recognizer.service
            OFFLINE=false
        fi
    else
        # Set restart service flag
        OFFLINE=true
        echo offline
    fi
    sleep 15
done

divx118, 2018-02-06:
@LondonSi72 
You can post code as follows on github:
```
```sh
#!/bin/sh
OFFLINE=false
while true; do 
    if ping www.google.com -c 1|grep -q time; then
        echo online
        # Only restart service when there was an offline
        if $OFFLINE; then
            systemctl restart voice-recognizer.service
            OFFLINE=false
        fi
    else
        # Set restart service flag
        OFFLINE=true
        echo offline
    fi
    sleep 15
done
``` start with 3 back ticks and end with 3 back ticks after the first 3 you can add the 
language used so you get highlighting. 
```
Result of the above
```sh
#!/bin/sh
OFFLINE=false
while true; do 
    if ping www.google.com -c 1|grep -q time; then
        echo online
        # Only restart service when there was an offline
        if $OFFLINE; then
            systemctl restart voice-recognizer.service
            OFFLINE=false
        fi
    else
        # Set restart service flag
        OFFLINE=true
        echo offline
    fi
    sleep 15
done
```

LondonSi72, 2018-02-06:
@divx118 Cool :) Thanks