google/aiyprojects-raspbian #189: Cannot stop timer (only in voicekit branch)

janvda, 2017-11-27:
I am using the voicekit branch.

I can start a timer with the command "OK Google start timer for 1 minute", and it is also properly ringing after one minute but the ringing doesn't stop when I speak the command "OK Google stop the timer".  This is very annoying as it makes the timer feature almost useless.

This issue is not appearing in the master branch.  So it would be good to migrate this fix from the master branch to the voicekit branch.

kr
Jan

sheridat, 2017-11-27:
Hi, please see the last post - of this post https://www.raspberrypi.org/forums/viewtopic.php?f=114&t=197800 
I am using the voicekit branch and have no problem in stopping timers and alarms

janvda, 2017-11-28:
@sheridat thanks for the response.  The post is referencing to _assistant_library_with_button_demo.py_  while I am using _assistant_library_demo.py._  So i don't think it is working for that python program.  I will give _assistant_library_with_button_demo.py_ a try later today when at home :-)  .

sheridat, 2017-11-28:
'@janvda saying "ok google stop" works equally well with the _assistant_library_demo.py._
If you don't have the _assistant_library_with_button_demo.py_ file just 
1. Click on Dev start terminal;
2. Type _git checkout voicekit_
3. Type _git pull origin voicekit_
That will bring your AIY voicekit up to date with this repo

janvda, 2017-11-28:
@sheridat I already have a very recent version of voicekit, but to be sure I have executed the commands again to pull latest version.  It made no difference:  I still have the same issue with both _assistant_library_demo.py_ and _assistant_library_with_button_demo.py_.  So i have to say the command _ok google stop_ several times before it is really stopping.

sheridat, 2017-11-28:
Well that's me done with this one.

janvda, 2017-11-28:
I think I got it fixed by pauzing a bit longer between "ok google" and "stop".  Most likely the "stop" wasn't caught because I spoke it too quickly.  So what worked for me in most cases is saying "ok google", then wait until the light is full on (indicating that it is listening) and then saying "stop".

sheridat, 2017-11-28:
excellent

closed 2017-11-29 by drigz