google/aiyprojects-raspbian #193: Check Cloud desktop shortcut doesn't work?

sheridat, 2017-11-30:
Hi, please see the fourth entry in this post from the AIY forum https://www.raspberrypi.org/forums/viewtopic.php?f=114&t=196612&p=1241009#p1241009 by pendragn32

If believe he is right.

If I click on the desktop "Check Cloud" a window opens and closes in milliseconds and nothing appears to be amiss.
If however I execute the python "script" Check Cloud" calls in a standard terminal  I get this error *
pi@raspberrypi:~ $ **/home/pi/AIY-voice-kit-python/checkpoints/check_cloud.py**
Traceback (most recent call last):
  File "/home/pi/AIY-voice-kit-python/checkpoints/check_cloud.py", line 26, in <module>
    import aiy._apis._speech  # noqa
  File "/home/pi/AIY-voice-kit-python/src/aiy/_apis/_speech.py", line 24, in <module>
    import google.auth
ImportError: No module named 'google'

If I click Start dev terminal and execute the script it works
Build info: Tue Sep 12 00:36:20 UTC 2017 @ 3955cac
Dev terminal is ready! Hint: enter src/assistant_library_demo.py
(env) pi@raspberrypi:~**/AIY-voice-kit-python $ /home/pi/AIY-voice-kit-python/checkpoints/check_cloud.py**
Testing the Google Cloud Speech API...
Everything's set up to use the Google Cloud.
Press Enter to close...

shivasiddharth, 2017-12-01:
if you directly run a bash script, and if it has some error, it will close automatically fast right, that is what is happening. The python script has to be run within the environment. Double clicking it does not make it run within the env and as a result of the error, it is closing fast! I dont see any issue here !

sheridat, 2017-12-01:
I do see an issue here.

Have you used the voicekit image to build your installation?
Have you followed the online instructions for building the voicekit?
Do you have the Check Cloud "thingy" on your desktop?

This is what the online instructionshttps://aiyprojects.withgoogle.com/voice/#makers-guide-3-1--change-to-the-cloud-speech-api  say 
**_CHECK THAT IT WORKS CORRECTLY
On your desktop, double-click the Check Cloud icon. Follow along with the script. If everything is working correctly, you’ll see this:

The cloud connection seems to be working

If you see an error message, follow the details and try the Check Cloud script again._**

If I run the cloudspeech demo it works fine - therefore I would expect to double-click the Check Cloud icon and see the words **The cloud connection seems to be working** I do not see that - I don't see any message just a terminal window opening and closing very quickly.

If your contention is that it can never work - well what is the point of having the  Check Cloud icon on the desktop?

drigz, 2017-12-01:
The script should work when run from the icon - I'll have a look at this.

Thanks for the report!

sheridat, 2017-12-01:
Cheers - the other two Desktop checks Check audio and Check WiFi work fine

shivasiddharth, 2017-12-01:
@sheridat , Ok, i used a wrong phrase. Since you had reported the execution of python script outside the environment from a normal terminal i said it was not an issue, not about the desktop shortcut. 
Regarding your queries about have i used voicekit, yes. Hmmm .. thats quite a bit of hammering.

drigz, 2017-12-01:
This could be fixed either by changing check_cloud.py to use subprocess (like it used to) or changing the shortcut to invoke it inside the virtualenv.

drigz, 2018-01-04:
Fixed by a4298c65 on the aiyprojects branch.

closed 2018-01-04 by drigz