google/aiyprojects-raspbian #194: Voicekit - Correct paths in shortcuts #191 - voice-recognizer-raspi in path perhaps incorrect.

sheridat, 2017-11-30:
Hi, 
I did my daily update of voicekit (**git checkout voicekit** then **git pull origin voicekit**) tonight and I pulled three new versions of the desktop shortcuts in folder /home/pi/AIY-voice-kit-python/shortcuts

So I had a look at issue 191 and I am puzzled. I am wondering whether there is a rabbit off.  The path names in "Exec" now have voice-recognizer-raspi rather than AIY-voice-kit-python in the name structure.

Check Audio for example now looks like this

[Desktop Entry]
Encoding=UTF-8
Type=Application
Name=Check audio
Comment=Check that the voiceHAT audio input and output are both working.
Exec=/home/pi/**voice-recognizer-raspi**/checkpoints/check_audio.py
Terminal=true

and an hour ago I was looking at it for another issue and it looked like this
[Desktop Entry]
Encoding=UTF-8
Type=Application
Name=Check audio
Comment=Check that the voiceHAT audio input and output are both working.
Exec=/home/pi/**AIY-voice-kit-python**/checkpoints/check_audio.py
Terminal=true

The original makes a lot more sense to me - but as ever happy to be told it is correct.

divx118, 2017-11-30:
According to the [HACKING.md](https://github.com/google/aiyprojects-raspbian/blob/voicekit/HACKING.md) `/home/pi/voice-recognizer-raspi/` is correct. However when looking at (the documentation)[https://aiyprojects.withgoogle.com/voice/#makers-guide-1-software-extensions] there is still a mention of `$HOME/AIY-voice-kit-python/src/` IMO the documentation needs some updating.

sheridat, 2017-11-30:
Ahhh.
I see - these files are for people who are not using the "image" but have cloned the repo. 
I will close - the google folk will read it anyway and presumably update the documentation.
Thank Maurice for straightening my thinking out.

closed 2017-11-30 by sheridat