google/aiyprojects-raspbian #196: requirements: google-auth-oauthlib version needs to be updated

captainwhippet, 2017-12-01:
As I posted on the AIY projects forum (https://www.raspberrypi.org/forums/viewtopic.php?f=114&t=198933&p=1241439#p1241439). 

The version of `google-auth-oauthlib` version needs to be updated to `0.2.0`, otherwise an `ImportError: cannot import name 'opentype'` error occurs when running `check_cloud.py`.

sheridat, 2017-12-21:
This also seems to be linked to https://github.com/google/aiyprojects-raspbian/issues/202. 
The instructions for device registration https://developers.google.com/assistant/sdk/reference/device-registration/register-device-manual ask you to update to the latest version as follows

Make sure the authorization tool is up to date:

python -m pip install --upgrade google-auth-oauthlib[tool]

LondonSi72, 2017-12-21:
I got an email with some details of the SDK update, and if all the new features filter down to the AIY image it will be a big improvement - the device-I'd looks like it will allow integration with the Google Assistant on phone etc:

https://groups.google.com/forum/m/#!msg/google-assistant-sdk-updates/zMVgRYQpgpw/8UfASE9SBgAJ

captainwhippet, 2017-12-22:
Looks like the version has been updated in this commit: https://github.com/google/aiyprojects-raspbian/commit/a4298c65b8d5e583f78570da7f38ab2f32e06c2f#diff-b4ef698db8ca845e5845c4618278f29a

Maybe this issues should be marked fixed?

drigz, 2017-12-22:
Yep, you're right @captainwhippet. There's another issue to track the rest of the upgrade.

closed 2017-12-22 by drigz