google/aiyprojects-raspbian #197: Filled SD card with logs

av4625, 2017-12-03:
ALSA lib pcm.c:7843:(snd_pcm_recover) overrun occurred
This log fills syslog and daemon.log in /var/log.

When i start the virtual env and start main.py it spams these continually.

I am running master branch and on Jessie.
I haven't changed anything recently, it just seemed to stop working

drigz, 2017-12-04:
Are you using the Voice Kit or other audio hardware?

av4625, 2017-12-04:
Im using the voicekit, I got the kit with the magpi magazine.
It all used to work, and then it "it just stopped"
The only thing I did recently is enable the remote gpio demon in raspi-config, im not sure if that caused any issues.
Whats the best way to get it back to default/working state without needing to install a new image?

drigz, 2017-12-04:
Honestly, I've not heard about anyone having this problem before, so I can't say.

If you check with `top`, is something using lots of CPU? That can cause overrun issues.

You could also try the forums to see if anyone there had this issue. If you figure anything out, post it here for anyone who runs in to the same problem.

av4625, 2017-12-04:
A python3 process was using a decent amount of cpu

sheridat, 2017-12-04:
The python3 process using a lot of CPU is the voicekit software.
Suggest you save your credential files and then re-image.

av4625, 2017-12-04:
Ill try a hard reset later as I have a mysql and webserver on that pi too

av4625, 2017-12-11:
I did a fresh install with the new version.
Worked at first, now the sd is filled again.
When i run from a virtual env it just constantly spams:
ALSA lib pcm.c:7843:(snd_pcm_recover) overrun occurred

drigz, 2017-12-12:
@av4625 What happens if you run this instead of running main.py? This is the process that main.py starts to do the recording:

`arecord -q -t raw -D default -c 1 -f s16 -r 16000 > /dev/null`

av4625, 2017-12-12:
Ill try later, is it the same for the version without main.py and just the demos?

Would the remote gpio daemon be doing something with one of the gpio pins that the speaker or mic is using? Feels like an issue like that

sheridat, 2017-12-12:
Yep it is Remote GPIO

I just enabled Remote GPIO. Clicked on Start dev terminal and then typed src/myassistant.py
bombarded by the ALSA lib pcm.c:7843:(snd_pcm_recover) overrun occurred message and the assistant isn't running.

I am on Jessie and I built using the google supplied image

av4625, 2017-12-12:
http://abyz.me.uk/rpi/pigpio/faq.html#Sound_isnt_working

Yea I think its this issue,

closed 2018-11-17 by dmitriykovalev