google/aiyprojects-raspbian #198: Sample format non available

firoz91, 2017-12-03:
Hi,

I am trying to setup the google voice hat and unable to hear any audio from the speaker. Please see below the error that I am getting. Appreciate any help.

Playing a test sound...
aplay: set_params:1233: Sample format non available
Available formats:
- S32_LE
Traceback (most recent call last):
  File "/home/pi/AIY-voice-kit-python/checkpoints/check_audio.py", line 138, in <module>
    main()
  File "/home/pi/AIY-voice-kit-python/checkpoints/check_audio.py", line 133, in main
    do_checks()
  File "/home/pi/AIY-voice-kit-python/checkpoints/check_audio.py", line 117, in do_checks
    if not check_speaker_works():
  File "/home/pi/AIY-voice-kit-python/checkpoints/check_audio.py", line 78, in check_speaker_works
    aiy.audio.play_wave(TEST_SOUND_PATH)
  File "/home/pi/AIY-voice-kit-python/src/aiy/audio.py", line 102, in play_wave
    player.play_wav(wave_file)
  File "/home/pi/AIY-voice-kit-python/src/aiy/_drivers/_player.py", line 71, in play_wav
    self.play_bytes(frames, wav.getframerate(), wav.getsampwidth())
  File "/home/pi/AIY-voice-kit-python/src/aiy/_drivers/_player.py", line 52, in play_bytes
    aplay.stdin.write(audio_bytes)
BrokenPipeError: [Errno 32] Broken pipe
Press Enter to close...

Thanks,
Firoz

drigz, 2017-12-04:
Maybe #183, does removing ~/.asoundrc help?

firoz91, 2017-12-05:
@drigz 

Thanks for the help.

Removing ~/.asoundrc makes the error disappear, but there is no sound coming out from speakers at all.

I followed the below steps you pointed out in the post, but no sound from speaker yet. any suggestions?

scripts/configure-driver.sh
reboot
scripts/install-alsa-config.sh
python3 checkpoints/check_audio.py
reboot

drigz, 2017-12-05:
If no sound is coming from the speakers, you should double-check you've connected the speakers correctly. If that doesn't help, you may have a hardware problem - do you know anyone with a Voice HAT you could test with?

firoz91, 2017-12-06:
Thanks @drigz.

I just double-triple-checked the speaker connection, no audio yet.

Unfortunately I do not know anyone with a Voice HAT that I can test with. 

I am able to hear some audio coming out of the raspberry pi headphone jack(using "omxplayer example.mp3" command) which means raspberry is at least working fine. Any other suggestion you can think of other than buying new Voice HAT hardware/Speaker?

drigz, 2017-12-06:
Have you tried with the [AIY image from the website](https://dl.google.com/dl/aiyprojects/voice/aiyprojects-latest.img.xz) (~1GB download)? If you've checked the connections and it doesn't work with the official image, then you should return your kit.

firoz91, 2017-12-06:
Actually I have been trying all these steps with the AIY image. Is there any other image that I can try with? If not, I would have to return the kit.

sheridat, 2017-12-06:
Just before you return it - are you logging into it using VNC?
If you are - on the desktop top line there is a loudspeaker symbol. If you hover your cursor over it does it show a numerical value
Can you also confirm that your /boot/config.txt has the following
#dtparam=audio=on
dtoverlay=i2s-mmap
dtoverlay=googlevoicehat-soundcard

firoz91, 2017-12-07:
@sheridat 

sorry, I just got a chance to look at the hardware. 

There is a loud speaker symbol at the top right and it shows 100. Also /boot/config.txt has same parameters as you mentioned. Is it time to return? :D

drigz, 2017-12-07:
Actually, let's try one more thing. Can you run this command and post the output?

`/usr/bin/aplay -v -d 1 -c 1 -t raw -f S32_LE /dev/zero`

sheridat, 2017-12-07:
And can you also right click on the speaker symbol and confirm that 
1.  It shows just two lines
snd_rpi_googlevoicehat_soundcar
External Device Settings
2, There is a green Tick symbol in front of snd_rpi_googlevoicehat_soundcar

Were just confirming that sound is being sent to the voicehat.

firoz91, 2017-12-07:
I am away from the hardware. Will try in evening and let you guys know. Thanks.

firoz91, 2017-12-08:
@drigz, here is the output that I got.

pi@raspberrypi:~ $ /usr/bin/aplay -v -d 1 -c 1 -t raw -f S32_LE /dev/zero
Playing raw data '/dev/zero' : Signed 32 bit Little Endian, Rate 8000 Hz, Mono
Plug PCM: Rate conversion PCM (48000, sformat=S32_LE)
Converter: libspeex (builtin)
Protocol version: 10002
Its setup is:
  stream       : PLAYBACK
  access       : RW_INTERLEAVED
  format       : S32_LE
  subformat    : STD
  channels     : 1
  rate         : 8000
  exact rate   : 8000 (8000/1)
  msbits       : 32
  buffer_size  : 2730
  period_size  : 170
  period_time  : 21333
  tstamp_mode  : NONE
  period_step  : 1
  avail_min    : 170
  period_event : 0
  start_threshold  : 2730
  stop_threshold   : 2730
  silence_threshold: 0
  silence_size : 0
  boundary     : 178913280
Slave: Route conversion PCM (sformat=S32_LE)
  Transformation table:
    0 <- 0
    1 <- 0
Its setup is:
  stream       : PLAYBACK
  access       : MMAP_INTERLEAVED
  format       : S32_LE
  subformat    : STD
  channels     : 1
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 32
  buffer_size  : 16384
  period_size  : 1024
  period_time  : 21333
  tstamp_mode  : NONE
  period_step  : 1
  avail_min    : 1024
  period_event : 0
  start_threshold  : 16384
  stop_threshold   : 16384
  silence_threshold: 0
  silence_size : 0
  boundary     : 1073741824
Slave: Soft volume PCM
Control: Master
min_dB: -51
max_dB: 0
resolution: 256
Its setup is:
  stream       : PLAYBACK
  access       : MMAP_INTERLEAVED
  format       : S32_LE
  subformat    : STD
  channels     : 2
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 32
  buffer_size  : 16384
  period_size  : 1024
  period_time  : 21333
  tstamp_mode  : NONE
  period_step  : 1
  avail_min    : 1024
  period_event : 0
  start_threshold  : 16384
  stop_threshold   : 16384
  silence_threshold: 0
  silence_size : 0
  boundary     : 1073741824
Slave: Direct Stream Mixing PCM
Its setup is:
  stream       : PLAYBACK
  access       : MMAP_INTERLEAVED
  format       : S32_LE
  subformat    : STD
  channels     : 2
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 32
  buffer_size  : 16384
  period_size  : 1024
  period_time  : 21333
  tstamp_mode  : NONE
  period_step  : 1
  avail_min    : 1024
  period_event : 0
  start_threshold  : 16384
  stop_threshold   : 16384
  silence_threshold: 0
  silence_size : 0
  boundary     : 1073741824
Hardware PCM card 0 'snd_rpi_googlevoicehat_soundcar' device 0 subdevice 0
Its setup is:
  stream       : PLAYBACK
  access       : MMAP_INTERLEAVED
  format       : S32_LE
  subformat    : STD
  channels     : 2
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 32
  buffer_size  : 16384
  period_size  : 1024
  period_time  : 21333
  tstamp_mode  : ENABLE
  period_step  : 1
  avail_min    : 1024
  period_event : 0
  start_threshold  : 1
  stop_threshold   : 1073741824
  silence_threshold: 0
  silence_size : 1073741824
  boundary     : 1073741824
  appl_ptr     : 0

@sheridat ..Yes, only two of them shown and there is a green Tick symbol in front of snd_rpi_googlevoicehat_soundcar.

drigz, 2017-12-08:
Afraid I can't think of any explanation other than a bad Voice HAT, so you'll have to return it.

closed 2017-12-08 by drigz

firoz91, 2017-12-13:
@drigz @sheridat After exchanging the hardware, I am able to hear the sound from the speaker and able to record my voice as well. This confirms that the hardware I had earlier was defective. 

Thank you both for the assistance.