google/aiyprojects-raspbian #199: Assistant selectively using a British Voice

sheridat, 2017-12-12:
Hi, I thought yourselves at Google in charge of the AIY repo would be interested to know that the AIY platform is now selectively using a British voice
If I ask what is the time - the response is an American voice
If I ask will it rain today - the response is a British voice
If I ask do I need an Umbrella today - the response is a British voice
If I ask what is the temperature - the response is a British voice
All of the British accent responses are followed by this error - [1557:1569:ERROR:assistant_output_handler.cc(543)] Ignoring kUnimplemented action error!`

I haven't done anything to update the assistant software so presumable this is a server side change.

Can you tell me the correct repo against which I should raise an issue???? 
(

drigz, 2017-12-12:
[assistant-sdk-python](https://github.com/googlesamples/assistant-sdk-python/issues) would be the right place for this. Thanks for reporting!

closed 2017-12-12 by drigz

sheridat, 2017-12-20:
I raised the issue
https://github.com/googlesamples/assistant-sdk-python/issues/140#issuecomment-353140432

 They asked my to try google-assistant-library 0.1.0 (we are currently using google-assistant-library-0.0.3)

I updated my env to  google-assistant-library 0.1.0 and then ran  src/assistant_library_with_local_commands_demo.py

It broke - 
(env) pi@raspberrypi:~/AIY-voice-kit-python $ src/assistant_library_with_local_commands_demo.py
Traceback (most recent call last):
File "src/assistant_library_with_local_commands_demo.py", line 99, in 
main()
File "src/assistant_library_with_local_commands_demo.py", line 93, in main
with Assistant(credentials) as assistant:
TypeError: init() missing 1 required positional argument: 'device_model_id'

**Does something need to happen with the AIY software to make it compatible with google-assistant-library 0.1.0?**