google/aiyprojects-raspbian #200: Cloudspeech: Conversational (optional hotwords)

rkkautsar, 2017-12-19:
In my project, I need to create a conversational bot, something like this:

- Bot, put an alarm for me
- OK, when will it be?
- 7.00 AM will be fine

I want to do it with the hotword "bot" (without button trigger), but [currently](https://github.com/google/aiyprojects-raspbian/blob/voicekit/src/aiy/cloudspeech.py#L50-L65) the transcript must include hotwords in order to be recognized. Is there any way to call the recognizer to immediately listens when called while still being able to listen to the hotword for initializing the conversation?

drigz, 2018-01-04:
We don't have a way to do this ("follow-on queries" in the language of the Assistant SDK) with Cloud Speech or with Assistant gRPC at the moment. With the Assistant Library, it will only work for built-in follow-on queries (like "OK Google, hello" ... "Hi, what can I do for you?" ... "What's the weather?").

With Cloud Speech hotwording, you could try enabling and disabling the hotword depending on the previous query. Currently you'd have to access the private member (eg `recognizer._hotwords = []`) but if it's useful I'd accept a pull request to add an `immediate=True` parameter to `recognizer.recognize()`.

Let us know how you get on!

rkkautsar, 2018-01-04:
Cool, I'll try submitting a PR for this

PR #234 (2018-01-04): add immediate option for cloudspeech recognize
Closes #200
closed 2018-01-08 by drigz