google/aiyprojects-raspbian #201: Hotword overlap

mpember, 2017-12-20:
With the mobile phone assistant now supporting 'Hey Google' as a hotword, it is even easier to accidentally trigger the wrong device.

Is there any potential for the hotword for the AIY devices to be customised?

It seems extreme to have to investigate implementing a whole different AI project just for the sake of having a different hotword.

sheridat, 2017-12-20:
+1 we have 3 google phones, 2 google nexus tablets and an AIY box all responding to Ok Google or Hey Google.

drigz, 2017-12-20:
We don't have plans to go beyond what the Google Assistant SDK offers for hotwording. Upstream issue: https://github.com/googlesamples/assistant-sdk-python/issues/81

mpember, 2017-12-20:
I guess that means I'll have to investigate switching to another project for my AIY kit.  It is a bit disappointing that Google appear to have co-opted the popularity of the Raspberry Pi devices to lock users into their walled garden.