google/aiyprojects-raspbian #202: Needs to be updated for new Google Assistant SDK version

thomwiggers, 2017-12-21:
You now need to provide a `device_id` and register a device model. I also think the button demo broke for me.

sheridat, 2017-12-21:
+1 - See updated issue https://github.com/google/aiyprojects-raspbian/issues/199 regarding  google-assistant-library 0.1.0 and the need for a device_model_id
Can you confirm that there will be a change made to this repo to support version 0.1.0 of the assistant?

drigz, 2017-12-21:
I'm discussing with the Assistant SDK team regarding device type. Unfortunately I'm out-of-office for the moment, so I won't be able to test anything until January.

@thomwiggers, could you give some more details on how the button demo broke?

thomwiggers, 2017-12-21:
In the demo, the button doesn't trigger the assistant to listen anymore. I haven't had much time to test it but my suspicions were that either `Assistant.start_conversation()` doesn't work, or that perhaps there is some issue with the threading.

drigz, 2017-12-21:
OK, thanks for the info. Filed #204 to track - if no one else figures it out before then I'll try to reproduce in January.

thomwiggers, 2017-12-21:
I should clarify that I meant the 'hotword with button' demo https://github.com/google/aiyprojects-raspbian/blob/voicekit/src/assistant_library_with_button_demo.py

@sheridat I registered it as a 'LIGHT' as per the Google Assistant Library SDK examples. It doesn't seem to matter much at this point. Following those examples also enabled me to register the AIY example application and set up a `device_model`

t1m0thyj, 2017-12-26:
With the new version of the Assistant SDK (0.1.0), I am able to see my Pi listed in the Google Assistant app as "Voice Kit". In the Music settings of the Assistant app on my phone, I set Google Play Music as the default provider, which should work since I have a paid subscription. However, if I ask Assistant on my Pi to play music, it still says "Sorry, I'm not sure how to help". Does music not work yet even though the Pi is now connected to the Assistant app?

thomwiggers, 2017-12-26:
Support for music wasn't in the changelog, so I guess not.

commit 4e08348c (2018-02-27): Update to the Assistant API v1alpha2

assistant_grpc_demo.py now supports other languages with
`aiy.i18n.set_language_code`.

Fixes #202, #237.

Change-Id: Ic51bee63aeb3117f12b6bacfc5452b055b003b7d
closed 2018-02-27 by mbrooksx