google/aiyprojects-raspbian #204: Triggering doesn't work in the button demo

drigz, 2017-12-21:
Reported at https://github.com/google/aiyprojects-raspbian/issues/202#issuecomment-353341576 by @thomwiggers.

In the hotword-and-button demo, the button doesn't trigger the assistant to listen anymore. I haven't had much time to test it but my suspicions were that either `Assistant.start_conversation()` doesn't work, or that perhaps there is some issue with the threading.

divx118, 2017-12-21:
Is there a hotword and button demo? IMO there is only hotword **or** button demo. I combined the 2 demos for my own use which works. Or did I miss a pull request...

thomwiggers, 2017-12-21:
See https://github.com/google/aiyprojects-raspbian/blob/voicekit/src/assistant_library_with_button_demo.py

divx118, 2017-12-21:
@thomwiggers Yeah sorry, my mistake.

drigz, 2018-01-26:
I'm not seeing this any more. If anyone else sees it, please comment and I'll reopen it.

closed 2018-01-26 by drigz