google/aiyprojects-raspbian #205: running cloudspeech_demo.py errors out

mikedougherty3, 2017-12-21:
Hello, I am trying to get this up and running as a christmas present for my daughter, but I am running into an issue. I was able to run cloudspeech_demo.py (or my own script) and then all of  a sudden started getting the following error when I push the button. So I  re-burned the image on the SD card, and set everything back up, but I am still getting this error. I have looked around but can't find anything useful. Also, assistant_library_demo.py works great, if that's any help. Thanks!

`pi@raspberrypi:~/AIY-voice-kit-python $ src/cloudspeech_demo.py 
Press the button and speak
Listening...
Traceback (most recent call last):
  File "/home/pi/AIY-voice-kit-python/src/aiy/_apis/_speech.py", line 263, in do_request
    return self._handle_response_stream(response_stream)
  File "/home/pi/AIY-voice-kit-python/src/aiy/_apis/_speech.py", line 208, in _handle_response_stream
    for resp in response_stream:
  File "/home/pi/AIY-voice-kit-python/env/lib/python3.4/site-packages/grpc/_channel.py", line 348, in __next__
    return self._next()
  File "/home/pi/AIY-voice-kit-python/env/lib/python3.4/site-packages/grpc/_channel.py", line 342, in _next
    raise self
grpc._channel._Rendezvous: <_Rendezvous of RPC that terminated with (StatusCode.UNIMPLEMENTED, GRPC target method can't be resolved.)>

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "src/cloudspeech_demo.py", line 53, in <module>
    main()
  File "src/cloudspeech_demo.py", line 37, in main
    text = recognizer.recognize()
  File "/home/pi/AIY-voice-kit-python/src/aiy/cloudspeech.py", line 47, in recognize
    return self._request.do_request().transcript
  File "/home/pi/AIY-voice-kit-python/src/aiy/_apis/_speech.py", line 268, in do_request
    raise Error('Exception in speech request') from exc
aiy._apis._speech.Error: Exception in speech request
Exception ignored in: <bound method Button.__del__ of <aiy._drivers._button.Button object at 0x74891a50>>
Traceback (most recent call last):
  File "/home/pi/AIY-voice-kit-python/src/aiy/_drivers/_button.py", line 57, in __del__
TypeError: 'NoneType' object is not callable
`

sheridat, 2017-12-22:
+1 my previously working cloudspeech.demo is no longer working. Traceback is similar to the above
(env) pi@raspberrypi:~/AIY-voice-kit-python $ src/cloudspeech_demo.py
Press the button and speak
Listening...
Traceback (most recent call last):
  File "/home/pi/AIY-voice-kit-python/src/aiy/_apis/_speech.py", line 266, in do_request
    return self._handle_response_stream(response_stream)
  File "/home/pi/AIY-voice-kit-python/src/aiy/_apis/_speech.py", line 211, in _handle_response_stream
    for resp in response_stream:
  File "/home/pi/AIY-voice-kit-python/env/lib/python3.4/site-packages/grpc/_channel.py", line 345, in __next__
    return self._next()
  File "/home/pi/AIY-voice-kit-python/env/lib/python3.4/site-packages/grpc/_channel.py", line 339, in _next
    raise self
grpc._channel._Rendezvous: <_Rendezvous of RPC that terminated with (StatusCode.UNIMPLEMENTED, GRPC target method can't be resolved.)>

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "src/cloudspeech_demo.py", line 53, in <module>
    main()
  File "src/cloudspeech_demo.py", line 37, in main
    text = recognizer.recognize()
  File "/home/pi/AIY-voice-kit-python/src/aiy/cloudspeech.py", line 49, in recognize
    text = self._request.do_request().transcript
  File "/home/pi/AIY-voice-kit-python/src/aiy/_apis/_speech.py", line 271, in do_request
    raise Error('Exception in speech request') from exc
aiy._apis._speech.Error: Exception in speech request
(env) pi@raspberrypi:~/AIY-voice-kit-python $

drigz, 2017-12-22:
Likely because of this: https://groups.google.com/forum/#!topic/cloud-speech-discuss/p8ZTlAVXyo0

I'll look into updating our code.

drigz, 2017-12-22:
You should be able to `git pull` for a fix. If you have changes in a local branch, you can `git pull --rebase origin voicekit`.

Once we've released an updated image we'll mark this issue as fixed.

sheridat, 2017-12-22:
@drigz Thank you for fixing it! Working fine.

sheridat, 2017-12-22:
Just a thought - won't the people who have remained on the master branch have the same issue?
BTW - I will put a post on the forum advising voicekit users that they need to pull the fix.

mikedougherty3, 2017-12-23:
@drigz This definitely worked! I appreciate your responsiveness in getting it done! Now my daughter can listen to Bad Lip Reading Star Wars over and over just by asking for it.

drigz, 2018-01-26:
Closing: the v1beta1 servers were brought back up, and the upgrade to the stable v1 API was released in the 2018-01-03 image.

closed 2018-01-26 by drigz