google/aiyprojects-raspbian #208: Is there new image for raspberry pi zero w?

idealist33, 2017-12-23:
I have an old image name of "aiyprojects-2017-05-03.img" and it works fine when using google assistant.
However as it is an an old image, there is no google cloud speech example.

I cloned github of here and followed HACKING.md but it was not perfect for raspberry pi zero w. 
Many python errors come out...this github might not be targeted for rpi zero w.

Is there any new image which contains same examples of latest github for raspberry pi zero w ?

PeterMalkin, 2017-12-23:
Yes, you can get it here:
https://dl.google.com/dl/aiyprojects/vision/aiyprojects-2017-12-18.img.xz

idealist33, 2017-12-24:
Thanks, but is this image for raspberry pi 3 b only? it does not work perfectly with my pi zero w.

PeterMalkin, 2017-12-24:
I have tested it on Raspberri pi 0 W

sheridat, 2017-12-24:
@idealist33  Your not being specific in saying what is not working for you. Anything to do with activating the assistant by saying "OK Google" or "Hey Google" does not work on the W

idealist33, 2017-12-25:
@sheridat Thanks a lot. Hotword and some python libraries havn't been loaded when executing python scripts. I will try the lated aiy image.

jjfretired, 2017-12-25:
Download and installed aiyprojects 2017-12-18.img.xz.  Using the instructions for HACKING.md to install the dependencies It failed to clone https://github.com/google/aiyprojects-raspbian.git voice-recognizer-raspi.  After logging into github it failed with Repository not found"

drigz, 2018-01-04:
@idealist33 Hotwording doesn't work with the Pi 0 at the moment, as there are performance problems with the less powerful processor. Check out the [compatibility table](https://aiyprojects.withgoogle.com/voice/#users-guide-3-using-your-device).

closed 2018-01-04 by drigz