google/aiyprojects-raspbian #209: All AIY scripts fail with python errors after Raspbian 'Stretch' upgrade

srcshelton, 2017-12-26:
After updating the latest AIYProjects image from Jessie to Stretch, running any of this repo's `make check`, `scripts/install_deps.sh`, or example code fails with:

```
Could not find platform independent libraries <prefix>
Could not find platform dependent libraries <exec_prefix>
Consider setting $PYTHONHOME to <prefix>[:<exec_prefix>]
Fatal Python error: Py_Initialize: Unable to get the locale encoding
ImportError: No module named 'encodings'
```

... whilst explicitly setting PYTHONHOME shortens this a little:

```
$ PYTHONHOME=/home/pi/AIY-voice-kit-python/env python 
Fatal Python error: Py_Initialize: Unable to get the locale encoding
ImportError: No module named 'encodings'
Aborted
```

Stretch has python-3.5.3 (... as well as python-2.7.13) by default, so is the `virtualenv` environment actually relying on some system packages which have since been upgraded?

(N.B. From the `Start dev terminal` desktop shortcut, `pip` itself can't run due to the above error...)

ScriptAutomate, 2017-12-27:
I was running into an issue on Stretch and I think it may be related.

I had to modify the `scripts/install-deps.sh` script so that the virtualenv it was creating was **not** using the `--system-site-packages` flag, otherwise I was unable to move forward at all. There seemed to be an issue with certain system Python packages it was referencing that was resolved when I just had the virtualenv be completely isolated from it.

This was on an initial first-time setup on a Raspbian Stretch base, fully updated.

RichardBronosky, 2017-12-27:
I usually don't upgrade OSes (for this very reason [mystery artifacts not present with clean installs]). I replace them. That said, I was following https://github.com/google/aiyprojects-raspbian/blob/voicekit/HACKING.md on top of an install of https://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2017-12-01/ and found only a mild mismatch of paths. I describe that here: https://gist.github.com/RichardBronosky/4d99935bef50b88ee086973fb81a906f

I found this question as I was researching before creating an issue on it.

# TL;DR
[HACKING.md](https://github.com/google/aiyprojects-raspbian/blob/voicekit/HACKING.md) says to clone into `~/voice-recognizer-raspi`, not `~/AIY-projects-python`, so you need to search in `scripts/install-deps.sh` for the former and change it to the later.

drigz, 2018-01-04:
@srcshelton virtualenvs can be sensitive to changes in the filesystem layout, so I'm not surprised the upgrade broke it. You could try:

```
cd ~/AIY-voice-kit-python
rm -r env
./scripts/install-deps.sh
```

or if that doesn't work, follow the instructions in https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/HACKING.md from scratch. Let us know how you get on!

srcshelton, 2018-01-05:
That's looking much better, thanks.
Just a thought - if Python virtualenv directories are somewhat fragile, should they be excluded from the repo and users instructed to run `install-deps.sh` for their OS as a mandatory setup step?
If not, perhaps a note on the rebuilding of the `env` directory could be added to the documentation, to assist other users who may not be familiar with virtualenv configurations?

drigz, 2018-01-05:
Hopefully anyone who runs in to this problem will find the solution in this issue. The OS upgrade case is relatively uncommon, as most users won't remember what they read in the AIY docs when they do the upgrade a few months later.

Is there anywhere we could have put a troubleshooting recommendation that would have been easier for you to find than this issue would be?

srcshelton, 2018-01-05:
Well, yes - I was thinking in the top-level README.md to be honest.

A quick:

> If you encounter python libraries warnings, try erasing your `env` directory and re-running `scripts/install-deps.sh` - see Issue #209

... should suffice?