google/aiyprojects-raspbian #210: voiceHAT not working

vipersgratitude, 2017-12-26:
I bought the MagPi with the AIY voice kit months ago and completely forgot about it. I tried to set it up today but I'm having zero luck.

I installed the latest image and all I get is "Failed to find the voiceHAT soundcard"
I followed the instructions in Hacking.md, but the problem persists.
arecord --list-devices returns "no soundcards found..."

Any ideas?

PeterMalkin, 2017-12-26:
Can you check to see if your config.txt has the driver overlay enabled for
the soundcard?

vipersgratitude, 2017-12-26:
Ahhh...what lines need to be enabled?

sheridat, 2017-12-26:
try adding these 2

dtoverlay=i2s-mmap
dtoverlay=googlevoicehat-soundcard

drigz, 2018-01-04:
@vipersgratitude did you make any progress? I'd recommend trying with the [official image](https://aiyprojects.withgoogle.com/voice/#assembly-guide-1-get-the-voice-kit-sd-image) as there's less room for error.

Fleker, 2018-01-05:
I also saw this issue. Are you using the most recent image (12/18)

I checked my `/boot/config.txt` and found a few things commented out . After I enabled 

```
dtoverlay=i2s-mmap
dtoverlay=googlevoicehat-soundcard
```

and rebooted, it appeared to work.

```
pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sndrpigooglevoi [snd_rpi_googlevoicehat_soundcar], device 0: Google voiceHAT SoundCard HiFi voicehat-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
pi@raspberrypi:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: sndrpigooglevoi [snd_rpi_googlevoicehat_soundcar], device 0: Google voiceHAT SoundCard HiFi voicehat-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

vipersgratitude, 2018-01-07:
Sorry for the late reply, guys. Yes, I got it to work. It was just a matter of editing the config file.

The image I used was freshly downloaded, so I assumed everything was set up to work in tandem the magazine's instructions. That's why, when I ran in to the error, my brain skipped the obvious and went straight to wondering if it was a hardware problem...or something.

drigz, 2018-01-08:
Good to hear it. We try to make sure the magazine instructions continue to work, but 2017-12-18 image was missing something. We've fixed that in the 2018-01-03 image: the Check Audio script will make this change automatically.

closed 2018-01-08 by drigz