google/aiyprojects-raspbian #211: install-deps.sh has hardcoded project path in conflict with instructions in HACKING.md

RichardBronosky, 2017-12-27:
Compare:
https://github.com/google/aiyprojects-raspbian/blame/voicekit/HACKING.md#L16
To:
https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/scripts/install-deps.sh#L35-L36

HACKING.md says to clone into `~/voice-recognizer-raspi`, but install-deps.sh has a hardcoded path of `~/AIY-projects-python`

Wouldn't it make more sense to construct that path based on the `scripts_dir` variable [defined on line 18](https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/scripts/install-deps.sh#L18)?

t1m0thyj, 2017-12-27:
There are two active branches of this GitHub repo right now, [i]aiyprojects[/i] and [i]voicekit[/i]. One uses a hardcoded path of `~/AIY-projects-python`, and the other uses `~/voice-recognizer-raspi`. Your links are referencing two different branches.

If you're using the [i]aiyprojects[/i] branch, the version of HACKING.md you want to use is this one which has already been updated with the correct path: https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/HACKING.md

RichardBronosky, 2017-12-27:
@t1m0thyj I just added a commit that would work in either branch. You are correct though, I didn't realize that there were different branches at play here. I got twisted because HACKING.md in the voicekit branch clones this project and ignores branches. Therefore, it gets aiyprojects by default. I guess that is the bigger issue.

So, how should we proceed? I see that the creation of that `pth` originated in the aiyprojects branch. https://github.com/google/aiyprojects-raspbian/commit/a4298c65b8d5e583f78570da7f38ab2f32e06c2f#diff-68a15966113a64d31b3e52ddc4c93227
I still think it should be relative as I have made it. But also, shouldn't HACKING.md in the voicekit branch clone the voicekit branch?

    git clone -b voicekit https://github.com/google/aiyprojects-raspbian.git AIY-projects-python

And for that matter, should we also honor "[Explicit is better than implicit](https://www.python.org/dev/peps/pep-0020/)" and change HACKING.md in the aiyprojects branch also?

    git clone -b aiyprojects https://github.com/google/aiyprojects-raspbian.git AIY-projects-python

RichardBronosky, 2017-12-27:
Also, I have thus far ignored not knowing why `$CLONED_PATH/env/lib/python3.5/site-packages/aiy.pth` is created for the aiyprojects branch but not the voicekit branch. I'd appreciate any insite you may have to offer on that.

t1m0thyj, 2017-12-27:
> But also, shouldn't HACKING.md in the voicekit branch clone the voicekit branch?

@RichardBronosky Good thinking, I totally agree.

> Also, I have thus far ignored not knowing why `$CLONED_PATH/env/lib/python3.5/site-packages/aiy.pth` is created for the aiyprojects branch but not the voicekit branch. I'd appreciate any insite you may have to offer on that.

I'm not sure, my guess is that the libraries for the AIY vision kit require that, since AFAIK the *aiyprojects* branch was created as a common base for both the AIY voice and vision kits.

ScriptAutomate, 2017-12-27:
This is interesting -- I don't see anything in the documentation that talks about why one would use one branch over the other. I have the AIY Kit, and have followed the [HACKING.md doc that is part of the aiyprojects branch](https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/HACKING.md) -- this is because `aiyprojects` is the default branch, and the HACKING.md files are technically using it. As a result, I had to figure out where to find the `check_audio.py` file on my own, which is actually properly referenced in the [voicekit branch HACKING.md](https://github.com/google/aiyprojects-raspbian/blob/voicekit/HACKING.md).

Should I have been using the `voicekit` branch? Before I get too far into my initial configurations? The `voicekit` branch says it is behind `aiyprojects` by a few commits, but also ahead by a few commits. Do they have different purposes/goals? I see that the following is mentioned in the README.md: 

> ...The new code is in the `voicekit` branch, and is included in images starting with aiyprojects-2017-09-11.img.

I want to be sure I'm using the proper branch so that issues I comment on or open are based on the properly recommended branch.

RichardBronosky, 2017-12-31:
@ScriptAutomate I'm glad it's not just me that is struggling with this.

Is there a location/community for discussion? It doesn't feel like github issues are the place to have conversations. (not complaining, accusing you of hijacking, etc. just observing) There is https://www.reddit.com/r/aiyprojects but that is hardly official. I see there is nothing on https://gitter.im

t1m0thyj, 2017-12-31:
@RichardBronosky Do you think it would be good for the community involved with this project to have a Discord server or perhaps even an online forum? If enough people are interested, I'd be glad to get involved.

t1m0thyj, 2017-12-31:
@RichardBronosky There is an AIY Projects subforum on the Raspberry Pi forum, which is the closest to an official community for discussion that I'm aware of: https://www.raspberrypi.org/forums/viewforum.php?f=114

RichardBronosky, 2018-01-01:
@t1m0thyj Thanks for [your guidance](https://www.raspberrypi.org/forums/viewtopic.php?p=1222969#p1222969) on using a the systemd service (that [you got re-added](https://github.com/google/aiyprojects-raspbian/pull/149#ref-pullrequest-266533860) to the repo) to start google Assistant on boot. In trying to implement that, I noticed another branch-induced path mismatch (the root of this issue I created) in `systemd/voice-recognizer.service`

I think I need to either change my branch, or change the path I'm checked out to. Would you mind giving me your opinion on which branch we should be using? https://www.reddit.com/r/aiyprojects/comments/7mhqld/which_branch_should_we_be_using/

sheridat, 2018-01-01:
@RichardBronosky This post indicates that the aiyprojects branch is the long term strategy https://github.com/google/aiyprojects-raspbian/issues/215
The aiyprojects branch is currently a little behind the voicekit branch https://github.com/google/aiyprojects-raspbian/issues/203

RichardBronosky, 2018-01-02:
@sheridat That is extremely helpful. I'll clone/checkout the aiyprojects branch to the instructed location (one of the nice things about the instruction discrepancy) start working off of that. I hope they get this straightened out soon.

RichardBronosky, 2018-01-02:
I hope they get this straightened out soon. I don't know if my observations are consistent with the greater community, but I am seeing a lot of very young Makers building these things. That might just be because I am very active in [my local Maker Space](http://decaturmakers.org/) which is very much committed to community outreach and a family oriented environment (no alcohol, #grumble). I expect the stepper drivers or relay boards I get off of aliexpress to be hard to use and have terrible or zero documentation. I expect to have to suffer for my art. But, when I see that beautiful glossy instruction book in the MagPi kit and I see Middle School children assembling them, I cringe to think of them getting discouraged and deciding STEM is not for them.

In my case, my gift to my 9 & 10 year old daughters was a pair of RPi3s, Voice Hat kits, replacement arcade buttons in their favorite colors, stepper motors and drivers, NeoPixel Rings (to go around the arcade buttons). And of course, access to my entire stockpile of spare components.

drigz, 2018-01-04:
@RichardBronosky and others, thanks for the feedback and sorry about the confusion! I've made #232 to improve the README. Let us know if you encounter any more problems (especially with the instructions on the website - like you, we want to make sure these are easy for newbies to follow).

dmitriykovalev, 2018-08-16:
Both `HACKING.md` and `README.md` have been recently updated.

closed 2018-08-16 by dmitriykovalev