google/aiyprojects-raspbian #213: How can I improve the TTS functionality ?

DEKHTIARJonathan, 2017-12-27:
I currently use the TTS function as shown in the samples projects:

`aiy.audio.say('My top priority text here!')`

However, this generates a very bad voice... I guess there might be a way to obtain the same voice as I get when the Google Assistant answers any question from wikipedia from instance.

Anyone know how to use the Assistant API to perform TTS (Text to Speech) with a decent quality ?

Thanks a lot,

Jonathan

ensonic, 2017-12-28:
There is no 'easy' way to make the assistant say arbitrary things. The local voice is definitely not as good as the actual assistant voice.
First please describe in which way the local tts sounds bad (clipped audio, intonation, clarity, ...) - ideally upload a sample.

sheridat, 2017-12-28:
@DEKHTIARJonathan  I find that a nice improvement to the TTS voice is to change from American to British  pronunciation (but I am british). 
You can change globally within any of the demos by declaring aiy.i18n.set_language_code('en-GB') or for each utterance by amending aiy.audio.say as per the following example ("I ran the script", lang='en-GB')

You can also alter the pitch and volume either globally or per utterance - for example aiy.audio.say("I ran the script", lang='en-GB', volume=65, pitch=105) 

Try it - you may find it makes the TTS voice acceptable 

If you happen to be french, german, spanish or italian you should change the language code accordingly - it will translate the output string into your language

DEKHTIARJonathan, 2017-12-28:
I tried to use the command `aiy.i18n.set_language_code('en-GB') ` which is an improvement in my point of view. However, it still doesn't sound natural at all like the original Google Assistant Voice.

**Sample Audio:** [Audio recording 2017-12-28 14-27-21.zip](https://github.com/google/aiyprojects-raspbian/files/1591272/Audio.recording.2017-12-28.14-27-21.zip)

I guess this is quite similar to the result you have. Maybe there is way to use another TTS API that gives a better results and play it back.

PS: @sheridat The command `aiy.audio.say("I ran the script", pitch=105)` does not work. It says that the argument pitch is not known.

sheridat, 2017-12-28:
@DEKHTIARJonathan  Ahhh perhaps you need to update your software - the ability to change the pitch was added recently as part of this PR https://github.com/google/aiyprojects-raspbian/pull/185

DEKHTIARJonathan, 2017-12-28:
@sheridat : Oh! You were right, now it's working.

Still, does any of you have any idea or experience with a TTS library/API and how to play to it back using the AIY library?

It would be a great bonus if it could handle other languages (French/German/Spanish/etc.)

Thanks a lot,

Jonathan

drigz, 2018-01-04:
The current TTS engine (pico2wave) has the best quality and flexibility we could find in a free, offline solution. If you want quality to match the Google Assistant's responses, you could use [Dialogflow](https://dialogflow.com/), although it's not as simple to perform on-device actions.

closed 2018-01-04 by drigz