google/aiyprojects-raspbian #215: AIYPROJECTS image contains VOICEKIT aiy code and examples in src ????

sheridat, 2017-12-29:
Hi, 
Any change of explaining why the Vision Kit image contains all of the Voicekit examples in the src folder such as assistant_library_demo.py and cloudspeech_demo.py as well as all the other software such as the voicehat drivers aiy apis and so on???

Surely all of that stuff is superfluous to the Vision Kit???

PeterMalkin, 2017-12-29:
No,

This has been done on purpose. I'm sorry, I am away from my desk today -
it's a holiday at work.

But the vision kit image will work fine with both voice and vision kits. In
fact, we strive to have one sdcard image to support all of the kits on the
market now, and the future ones whenever they come.

As far as the source code, we aim to provide one python library that covers
all of the kits functionality with one coherent python api. Therefore it's
not a coincidence that you see voice kit related code in the sources.

Hope this helps explain our thinking

sheridat, 2017-12-29:
If I understand your strategy doesn't it mean that voicekit branch is headed for depreciation? Otherwise your maintaining two branches for one product

PeterMalkin, 2017-12-29:
Yes, your understanding is correct.

That's why I marked the "aiyprojects" as the default branch. We will try to
phase out the support for the voicehat bracnh. But not before we have
soaked the new code and are certain that no one is broken. So there's not
an immediate plan to obsolete it or anything like that.

Hopefully eventually everyone will get to the new codebase and it will
happen organically. At this point in time however the code for voice kit is
the same in both branches. So "aiyprojects" only has expansions for vision
kit support.

Hope this makes sense.

sheridat, 2017-12-29:
I do understand what your doing but how can you hope to prove your new image is viable for both the voice kit and vision kit when the online download guidance for the two products points to divergent download images? 
The Voice Kit guidance https://aiyprojects.withgoogle.com/voice/#assembly-guide-1-get-the-voice-kit-sd-image points to the following download https://dl.google.com/dl/aiyprojects/voice/aiyprojects-2017-09-11.img.xz

The Vision Kit guidance https://aiyprojects.withgoogle.com/vision/#assembly-guide-1-get-the-vision-kit-sd-image points to the following download https://dl.google.com/dl/aiyprojects/vision/aiyprojects-2017-12-18.img.xz

Don't you need to change the Voice Kit guidance to point to the same download as the Vision Kit? 

Otherwise your relying on people who clone the aiyprojects branch and use it for the Voice Kit to prove that it is pukka for the Voice Kit product

jhewitt123, 2018-01-03:
So far a disaster for me. Apparently fried board or something cause google mislabelled the 'Pi' endof the cable, also switched out a new camera cable, - which way does that one go ', what image do I need and where cause nothing runs now

jhewitt123, 2018-01-03:
Okay, re did all the connections and still didn;t run, but wait for it, when I removed the magnetic lens the camera started to  work and I could get a blue light when I frowned and a white/yellow light when I smiled.

PeterMalkin, 2018-01-03:
jhewitt123,

happy it works for you! Does it mean you can close the https://github.com/google/aiyprojects-raspbian/issues/229?

jhewitt123, 2018-01-03:
sure, thanks for the help, butcan you let me knowhowtorun thedemos, ie from
the command line, from within python, or how, but yes

PeterMalkin, 2018-01-04:
Sure.

You can just follow the Makers guide to run the demos manually
https://aiyprojects.withgoogle.com/vision/#makers-guide

You can use the serial connection as described there, or you can attach keyboard, mouse and monitor to your raspberry pi.

Please be sure to stop the joy detection autorun if you intend to run demos by hand. If one process occupies the hardware, it will be inaccessible to other processes. This is how you do it: 
https://aiyprojects.withgoogle.com/vision/#makers-guide-3-vision-kit-options

closed 2018-01-04 by sheridat