google/aiyprojects-raspbian #216: grpc Demo not workin

endamaco, 2017-12-30:
Hi,
trying the grpc demo when pushing the arcade button I get this error:

`Press the button and speak
[2017-12-30 15:25:07,757] INFO:recorder:started recording
Listening...
Traceback (most recent call last):
  File "src/assistant_grpc_demo.py", line 54, in <module>
    main()
  File "src/assistant_grpc_demo.py", line 42, in main
    text, audio = assistant.recognize()
  File "/home/pi/voice-recognizer-raspi/src/aiy/assistant/grpc.py", line 49, in recognize
    response = self._request.do_request()
  File "/home/pi/voice-recognizer-raspi/src/aiy/_apis/_speech.py", line 258, in do_request
    service = self._make_service(self._channel_factory.make_channel())
  File "/home/pi/voice-recognizer-raspi/src/aiy/_apis/_speech.py", line 370, in _make_service
    return embedded_assistant_pb2.EmbeddedAssistantStub(channel)
AttributeError: module 'google.assistant.embedded.v1alpha1.embedded_assistant_pb2' has no attribute 'EmbeddedAssistantStub'
Segmentation fault
`

drigz, 2018-01-03:
What version of the Raspberry Pi do you have? Which image file did you flash? (eg aiyprojects-2017-09-11.img.xz)

endamaco, 2018-01-03:
It's the raspberry 3 and I'm using the source code from this GitHub.. downloaded 1 week ago

drigz, 2018-01-03:
We don't support google-assistant-library==0.1.0 yet - see #207.

If you install the dependencies with env/bin/pip install -r requirements.txt you should get the correct versions.

drigz, 2018-01-04:
One correction: `env/bin/pip install -r requirements.txt` won't install `google-assistant-library` - you can use `./scripts/install-deps.sh` to install all dependencies.

I'm going to close this - please reopen if it still doesn't work if you follow the instructions in https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/HACKING.md.

closed 2018-01-04 by drigz