google/aiyprojects-raspbian #217: Vision Kit: bonnet boot failed -113

chadwallacehart, 2017-12-30:
I purchased the Vision Kit yesterday with a new Pi Zero W, but I cannot get any of the vision code to work. It seems the Myriad chip is not loading:

Running the samples and inspecting the Joy Detection Demo service gives me a `FileNotFoundError: [Errno 2] No such file or directory: '/dev/vision_spicomm'`

journalctl shows the following on my first boot after loading the image:
```
pi@raspberrypi:~ $ journalctl | grep visionbonnet
Dec 18 10:36:29 raspberrypi kernel: googlevisionbonnet spi0.0: Initializing
Dec 18 10:36:29 raspberrypi kernel: googlevisionbonnet spi0.0: Resetting myriad on probe
Dec 18 10:36:29 raspberrypi kernel: googlevisionbonnet spi0.0: Resetting myriad
Dec 18 10:36:37 raspberrypi kernel: googlevisionbonnet spi0.0: Writing myriad firmware
Dec 18 10:36:41 raspberrypi kernel: googlevisionbonnet spi0.0: Myriad booting
Dec 18 10:36:46 raspberrypi kernel: googlevisionbonnet spi0.0: Slave not ready after 5000 ms
Dec 18 10:36:46 raspberrypi kernel: googlevisionbonnet spi0.0: Myriad did not boot in a timely fashion
Dec 18 10:36:46 raspberrypi kernel: googlevisionbonnet spi0.0: Inital bonnet boot failed: -113
Dec 18 10:36:46 raspberrypi kernel: googlevisionbonnet: probe of spi0.0 failed with error -113
```

I am able to get the camera to work if I connect it directly to the Pi Zero W and `~/AIY-projects-python/src/example/buzzer_demo.py` works fine too.

I also tried enabling the SPI through `raspi-config` and `/boot/config.txt` with no success.

What does `error -113` mean? Are there any diagnostics I can do?

sheridat, 2017-12-30:
Hi, Whil's you wait for a response from google - Just a thought - in the /boot/config.txt file there is a bunch of commented out lines

# dtoverlay=i2s-mmap
# dtoverlay=googlevoicehat-soundcard
# dtoverlay=googlevisionbonnet-myriad
# dtoverlay=aiy-io
# dtoverlay=aiy-leds
start_x=1
gpu_mem=128
# dtoverlay=dwc2

what I can tell you is that the first 2 are used by the voicehat software. I uncommented them and can now use my voicehat.
# dtoverlay=i2s-mmap
# dtoverlay=googlevoicehat-soundcard

**but** the following look like they are for the Vision Kit. Try "un commenting" them

dmitriykovalev, 2017-12-30:
@chadwallacehart could you please check dmesg output for any kind of errors?

sheridat, 2017-12-31:
@chadwallacehart 
Probably best to uncomment the above one at a time

chadwallacehart, 2017-12-31:
Without making any changes, dmesg gives the the same errors as journalctl (grepped to filter - I did not see anything else interesting in the complete output): 
```
pi@raspberrypi:~ $ dmesg | grep 'google\|aiy'
[   23.974274] aiy-io-i2c 1-0051: AIYIO Status:OK-V0.1
[   24.597639] googlevisionbonnet spi0.0: Initializing
[   24.834490] googlevisionbonnet spi0.0: Resetting myriad on probe
[   24.834511] googlevisionbonnet spi0.0: Resetting myriad
[   28.833055] googlevisionbonnet spi0.0: Writing myriad firmware
[   34.668882] googlevisionbonnet spi0.0: Myriad booting
[   39.692090] googlevisionbonnet spi0.0: Slave not ready after 5000 ms
[   39.692107] googlevisionbonnet spi0.0: Myriad did not boot in a timely fashion
[   39.693505] googlevisionbonnet spi0.0: Inital bonnet boot failed: -113
[   39.754714] googlevisionbonnet: probe of spi0.0 failed with error -113
```
I also tried uncommenting lines in the boot.txt.
`dtoverlay=dwc2` was already uncommented and I left this as is for the next couple of tests-

When I uncommented just `dtoverlay=googlevisionbonnet-myriad` and rebooted, I get the same errors as above:

When I uncommented `dtoverlay=aiy-io` I get the following:
```
pi@raspberrypi:~ $ dmesg | grep 'google\|aiy\|i2c'
[    3.928849] i2c /dev entries driver
[   11.358450] i2c i2c-1: Failed to register i2c client aiy-io-i2c at 0x51 (-16)
[   11.358475] i2c i2c-1: of_i2c: Failure registering /soc/i2c@7e804000/aiy-io-i2c@51
[   11.358490] i2c i2c-1: Failed to create I2C device for /soc/i2c@7e804000/aiy-io-i2c@51
[   24.926923] aiy-io-i2c 1-0051: AIYIO Status:OK-V0.1
[   25.205617] googlevisionbonnet spi0.0: Initializing
[   25.324009] googlevisionbonnet spi0.0: Failed to bind reset GPIO
```

I also tried to comment out all the lines, including `dtoverlay=dwc2` but that gives the same `spi0.0` errors (but not the i2c one above). One of the aiy services must be changing this since `dtoverlay=dwc2`was uncommented when I went back to `boot/config.txt`.

dmitriykovalev, 2017-12-31:
Changing boot.txt probably will not help to fix "Myriad did not boot in a timely fashion". Can you flash another SD card and check the result? This is just to make sure Myriad firmware is not damaged (although it's very unlikely). 

What kind of power source do you use? Just in case, can you try with wall outlet USB adapter?

joezen777, 2017-12-31:
I'm also getting the same issues as chadwallaceh art. On Reddit one person mentioned that they got it working with a zero 1.1 but can't get it working with a 1.3 w. Anyone actually get this working on a 1.3 w zero? Very frustrating to get this over the weekend and find not the same standard of documentation as the audio AIY kit.

chadwallacehart, 2017-12-31:
@dmitriykovalev - I actually already reflashed the image before posting here, though that was on the same SD card. During my 1st install It froze and I did a power cycle. When I first saw the errors I thought maybe the initial boot install did not finish. 

When I originally had boot issues I switched to my best 2.5A wall power supply.

In any case I tried a fresh SD card using a freshly downloaded image from https://dl.google.com/dl/aiyprojects/vision/aiyprojects-2017-12-18.img.xz. I got the same result.

Could my VPU chip be bad somehow? Is there any kind of hardware check I can do?

chadwallacehart, 2018-01-01:
per @joezen777 comment on [this reddit forum](https://www.reddit.com/r/aiyprojects/comments/7n8g6n/anyone_get_their_vision_kit_working/)- I hooked up the Vision Bonnet to an original Pi Zero and it worked. 

```
pi@aiyvisionpi:~ $ journalctl | grep visionbonnet
Dec 31 19:21:10 aiyvisionpi kernel: googlevisionbonnet spi0.0: Initializing
Dec 31 19:21:10 aiyvisionpi kernel: googlevisionbonnet spi0.0: Resetting myriad on probe
Dec 31 19:21:11 aiyvisionpi kernel: googlevisionbonnet spi0.0: Resetting myriad
Dec 31 19:21:15 aiyvisionpi kernel: googlevisionbonnet spi0.0: Writing myriad firmware
Dec 31 19:21:21 aiyvisionpi kernel: googlevisionbonnet spi0.0: Myriad booting
Dec 31 19:21:21 aiyvisionpi kernel: googlevisionbonnet spi0.0: Myriad ready
```

It seems there is something about the Pi Zero W that prevents the Myriad firmware from loading.

PeterMalkin, 2018-01-01:
@chadwallacehart 

It is probable that your bonnet is bad, but highly unlikely, because we have tested all of the boards coming out of manufacturing line at the factory by attaching them to the Rpi0W.

Is there a chance you could use your good 2.5A wall power supply, disconnect all the camera cables, attach the bonnet to your pi 0 w and try to boot it up?

The hunch I have is about the soldering on the headers - but I have no way to know if its correct of not.

pbelevich, 2018-01-02:
> What does error -113 mean?

/home/pi/drivers-raspi/aiy/aiy/staging/myriad/googlevisionbonnet-myriad.c:495
```
  if (!visionbonnet_wait_slave_ready(bonnet)) {
    dev_err(&bonnet->spidev->dev, "Myriad did not boot in a timely fashion\n");
    ret = -EHOSTUNREACH;
    goto beach;
  }
```
EHOSTUNREACH = 113

chadwallacehart, 2018-01-02:
@pbelevich - I think your hunch may be right. I just did a quick [Wiring Pi pin test](http://wiringpi.com/the-gpio-utility/pin-test/) and had a failure on one pin. I will do some more investigation when I have more time later today.

dmitriykovalev, 2018-01-02:
For the reference attaching 'good' dmesg output from Pi Zero W with attached Vision Bonnet:
[dmesg_good.txt](https://github.com/google/aiyprojects-raspbian/files/1598326/dmesg_good.txt)

pbelevich, 2018-01-02:
@dmitriykovalev what version of rpi0w did you use? 1.1 or 1.3?

dmitriykovalev, 2018-01-02:
@pbelevich 

```
pi@raspberrypi:~ $ cat /proc/cpuinfo | grep Revision
Revision	: 9000c1
```

9000c1 means exactly Pi Zero W.

dmitriykovalev, 2018-01-02:
@pbelevich 

Available Pi Zero [revisions](https://elinux.org/RPi_HardwareHistory):

```
900092 -- Pi Zero without camera connector, no wireless
900093 -- Pi Zero with camera connector, no wireless
9000c1 -- Pi Zero W with camera connector + Wi-Fi/Bluetooth
```
Vision Bonnet should work with both 900093 and 9000c1.

chadwallacehart, 2018-01-02:
I was able to pickup a new Pi Zero W. Everything works fine with this new one. It appears my issue is with the original Pi Zero W I used.

I reinspected all the pins and I do not see anything visibly wrong in my header soldering, but who knows.

For reference, here is the full dmesg output from my _**bad**_ Pi Zero W. 
[bad_pi0w_dmesg.txt](https://github.com/google/aiyprojects-raspbian/files/1599047/bad_pi0w_dmesg.txt)

And here is the dmesg output from the _**new**_ one I just setup: 
[newpi0_no_cables_dmesg.txt](https://github.com/google/aiyprojects-raspbian/files/1599052/newpi0_no_cables_dmesg.txt)

Thanks for the guidance. I can close this unless there are additional tests you would like me to run that could help others.

dmitriykovalev, 2018-01-03:
@chadwallacehart 

Perfect! Can you please attach photographs of bad Pi Zero W, so we also can look at 40-pin header (ideally from both sides) ?

Thank you very much for staying with us and all the efforts you put into this. I am very happy it finally works.

pbelevich, 2018-01-03:
@dmitriykovalev it would be nice to have some kind of diagnostic tool to understand what's wrong, because I have the same problem as @chadwallacehart had and don't know how to attack the problem

aallan, 2018-01-03:
Have you tried running the [Pin Test](http://wiringpi.com/the-gpio-utility/pin-test/) utility included in Gordon Henderson's WiringPi library? Make sure you should [download](http://wiringpi.com/download-and-install/) the latest version of the library (remove the pre-installed version on the SD Card) and make sure you remove the Vision Bonnet before running the pin test or it will return incorrect results. It's not fool proof, but can give a good indication if there is a header problem.

chadwallacehart, 2018-01-03:
@dmitriykovalev - here are pictures of the Pi Zero W that _did not_ work for me. Admittedly this was not my best soldering job :( I started using some solder that was way too thick, leaving too much on some pins that I had to carefully cut off to remove since I lost my solder braid.

[image]
[image]
[image]

If you see something off please let me know and maybe I can salvage this unit for GPIO use in future projects.

aallan, 2018-01-03:
@chadwallacehart On the back view there seems to be some copper visible between the third and fourth pins from the right on the lower view indicating silk screen damage. As well as copper on the two rightmost pins on the top row. It's possible you overheated the board and damaged the ground plane? Have you tried running the [Pin Test utility](http://wiringpi.com/the-gpio-utility/pin-test/) included in Gordon Henderson's WiringPi library?

sheridat, 2018-01-03:
I had a similar issue earlier in the year when I couldn't get a justboom DAC to work on a Pi W and it was my first ever attempt at soldering headers. 
When I upended it with the pins facing downwards and then pushed the pins again my kitchen worktop/countertop a couple of the pins actually moved - I reflowed the solder a couple of times until they no longer moved.
The post by @aallan is a great idea.

chadwallacehart, 2018-01-04:
@aallan - I did run the [Wiring Pi pin test](http://wiringpi.com/the-gpio-utility/pin-test/) 2 days ago an it showed I had a bad PIN. Before I took the pictures above I did resolder one pin and tried to do a little clean-up, but it still gave me the same Vision Bonnet errors.

Now when I run the same Wiring Pi test I get no errors:
```
          The main 8 GPIO pins  0: 7:  OK
                The 5 SPI pins 10:14:  OK
               The serial pins 15:16:  OK
                  The I2C pins  8: 9:  OK
```

Obviously something is messed up, but I am still unsure what specifically. I definitely scrapped up the board improperly trying to remove solder, but I would think that would give a consistent error on one of the pins. It looks like Wiring Pi does not test all the pins and it only tests outputs, so maybe something untested is causing problems with the bonett. Given the replacements are now only $5 at Microcenter, I might save this one for a project that doesn't need a GPIO.

closed 2018-01-04 by chadwallacehart