google/aiyprojects-raspbian #223: AIY Vision Kit: cannot run joy_detection_demo due to missing LED device file

dchang0, 2018-01-02:
I'm encountering this error while trying to run the very first project, joy_detection_demo, that is supposed to run automatically on boot-up.

I assume the LED they are attempting to connect to is the one inside the arcade button on top and not the privacy LED on the front.

Any tips on why the LED would not be detected and its device files not created?

This might be caused by upgrading the kernel to 4.9.73+ using apt-get update;apt-get upgrade;apt-get dist-upgrade;rpi-update

I found the files in: /var/lib/dkms/leds-ktd202x
and they mention 4.9.59+ in the path. Is this driver dependent on the kernel version?

Thanks in advance!

Jan 2 02:15:19 raspberrypi python[888]: Traceback (most recent call last):
Jan 2 02:15:19 raspberrypi python[888]: File "/home/pi/AIY-projects-python/src/examples/vision/joy/joy_detection_demo.py", line 167, in <module>
Jan 2 02:15:19 raspberrypi python[888]: main()
Jan 2 02:15:19 raspberrypi python[888]: File "/home/pi/AIY-projects-python/src/examples/vision/joy/joy_detection_demo.py", line 161, in main
Jan 2 02:15:19 raspberrypi python[888]: detector = JoyDetector(args.num_frames, args.preview_alpha)
Jan 2 02:15:19 raspberrypi python[888]: File "/home/pi/AIY-projects-python/src/examples/vision/joy/joy_detection_demo.py", line 54, in __init__
Jan 2 02:15:19 raspberrypi python[888]: self._rgbled = RGBLED(debug=False)
Jan 2 02:15:19 raspberrypi python[888]: File "/home/pi/AIY-projects-python/src/aiy/_drivers/_rgbled.py", line 63, in __init__
Jan 2 02:15:19 raspberrypi python[888]: self.Reset()
Jan 2 02:15:19 raspberrypi python[888]: File "/home/pi/AIY-projects-python/src/aiy/_drivers/_rgbled.py", line 192, in Reset
Jan 2 02:15:19 raspberrypi python[888]: self._PWriteInt('red', 'device/reset', 1)
Jan 2 02:15:19 raspberrypi python[888]: File "/home/pi/AIY-projects-python/src/aiy/_drivers/_rgbled.py", line 95, in _PWriteInt
Jan 2 02:15:19 raspberrypi python[888]: with open(path, 'w') as output:
Jan 2 02:15:19 raspberrypi python[888]: FileNotFoundError: [Errno 2] No such file or directory: '/sys/class/leds/ktd202x:led1/device/reset'
Jan 2 02:15:19 raspberrypi python[888]: Exception ignored in: <bound method RGBLED.__del__ of <aiy._drivers._rgbled.RGBLED object at 0xb5e62a50>>
Jan 2 02:15:19 raspberrypi python[888]: Traceback (most recent call last):
Jan 2 02:15:19 raspberrypi python[888]: File "/home/pi/AIY-projects-python/src/aiy/_drivers/_rgbled.py", line 66, in __del__
Jan 2 02:15:19 raspberrypi python[888]: self.Reset()
Jan 2 02:15:19 raspberrypi python[888]: File "/home/pi/AIY-projects-python/src/aiy/_drivers/_rgbled.py", line 192, in Reset
Jan 2 02:15:19 raspberrypi python[888]: self._PWriteInt('red', 'device/reset', 1)
Jan 2 02:15:19 raspberrypi python[888]: File "/home/pi/AIY-projects-python/src/aiy/_drivers/_rgbled.py", line 95, in _PWriteInt
Jan 2 02:15:19 raspberrypi python[888]: with open(path, 'w') as output:
Jan 2 02:15:19 raspberrypi python[888]: FileNotFoundError: [Errno 2] No such file or directory: '/sys/class/leds/ktd202x:led1/device/reset'

dchang0, 2018-01-02:
Just confirmed: if I re-image the microSD card back to the official image, it now works. So, DO NOT upgrade the kernel.

Is this something that should be fixed? I think so--the driver should not break just because of a kernel upgrade. But it's not my call. Please close this issue if it is determined that newer kernels will not be supported. Thanks!

sheridat, 2018-01-02:
Perhaps the code in /home/pi/drivers-raspi has something to do with this?

jtgans, 2018-01-02:
Sounds like DKMS didn't rebuild the ktd202x driver when you upgraded the kernel, @dchang0. I'll see if I can reproduce locally.

jtgans, 2018-01-02:
As a workaround, if you go into /home/pi/drivers-raspi and run `sudo install.sh` that should force a rebuild and reinstall of the drivers for the newer kernel.

dchang0, 2018-01-02:
Thanks, sheridat and jtgans--I will give this a try and get back to you with the results.

dchang0, 2018-01-02:
No luck.

This is what I did:
reimaged microSD with aiyprojects-2017-12-18.img
sudo apt-get update
sudo apt-get upgrade
sudo apt-get dist-upgrade
sudo rpi-update
sudo shutdown -r now
[LED driver missing, joy_detection_demo fails]
cd /home/pi/drivers-raspi
sudo ./install.sh
[Watch as various Voice Hat-related drivers are removed; watch as LED driver is rebuilt.]
sudo shutdown -r now
[joy_detection_demo fails with same error as in original post.]

I'll leave it in this current broken state to help you debug. Let me know what else you would like me to try.

dchang0, 2018-01-02:
I do notice this error recurring throughout the install.sh output:

Error! echo
Your kernel headers for kernel 4.9.73+ cannot be found at
/lib/modules/4.9.73+/build or /lib/modules/4.9.73+/source.
Error! echo
Your kernel headers for kernel 4.9.73-v7+ cannot be found at
/lib/modules/4.9.73-v7+/build or /lib/modules/4.9.73-v7+/source.

Maybe the driver was not rebuilt after all. I will install the kernel headers and retry the install.sh.

jtgans, 2018-01-02:
I'm attempting to reproduce using the same image and after an `apt-get dist-upgrade -u` I'm still on the same kernel. Can you post the contents of your `/etc/apt/sources.list` and `/etc/apt/sources.list.d`?

jtgans, 2018-01-02:
Ah, I see. `rpi-update` must be updating things outside of apt. I'll give that a shot to see what's going on.

dchang0, 2018-01-02:
Yes, I confirmed that rpi-update is the one command causing the problem. Thanks for working on this!

jtgans, 2018-01-02:
Yeah, in general, don't use rpi-update unless you want bleeding edge versions of the drivers, firmware, etc. We're working off of the stable stretch version of raspbian, so you probably want to stick with `apt-get dist-upgrade` instead and keep tracking stable.

rpi-update manually clones git repositories and builds them locally, which goes outside of the apt/dpkg system -- and explains why the header files are missing. =o)

jtgans, 2018-01-02:
I'm closing this because rpi-update is a bit out of scope for us. The drivers should work fine as long as you track the apt repositories, though. Theoretically if you clone and build your own kernel locally, it should also work, provided you use the install.sh script in drivers-raspi to rebuild the drivers for the newer kernel as well.

closed 2018-01-02 by jtgans

dchang0, 2018-01-02:
I'm okay with not using rpi-update for my Vision Kit. It is somewhat standalone in typical use cases.  Thanks again!

jtgans, 2018-01-02:
Anytime -- that's why we have a public bugtracker. =o)