google/aiyprojects-raspbian #224: Vision Bonnet Camera not working

annoscholten, 2018-01-02:
Running on a Pi 0 W V1.1, but the camera won't connect.

The camera works when I connect it directly to the Pi 0 W, it works fine:

[   23.968383] googlevisionbonnet spi0.0: Initializing
[   24.424624] googlevisionbonnet spi0.0: Resetting myriad on probe
[   24.424641] googlevisionbonnet spi0.0: Resetting myriad
[   28.343088] googlevisionbonnet spi0.0: Writing myriad firmware
[   34.492645] googlevisionbonnet spi0.0: Myriad booting
[   34.712593] googlevisionbonnet spi0.0: Myriad ready

(env) pi@raspberrypi:~ $
~/AIY-projects-python/src/examples/vision/joy/joy_detection_demo.py
Starting JoyDetector...
ERROR:root:Camera is not running.
ERROR:root:CameraInference: Camera inference is not running.
ERROR:root:StopCameraInference: Camera inference is not running.
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/home/pi/AIY-projects-python/src/examples/vision/joy/joy_detection_demo.py",
line 127, in _run_detector
    faces = face_detection.get_faces(result)
  File "/home/pi/AIY-projects-python/src/aiy/vision/models/face_detection.py",
line 63, in get_faces
    assert len(result.tensors) == 3
AssertionError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.5/threading.py", line 914, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.5/threading.py", line 862, in run
    self._target(*self._args, **self._kwargs)
  File "/home/pi/AIY-projects-python/src/examples/vision/joy/joy_detection_demo.py",
line 138, in _run_detector
    break
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 758,
in __exit__
    self.close()
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 752, in close
    raise exc
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 1995,
in _control_callback
    "No data recevied from sensor. Check all connections, "
picamera.exc.PiCameraRuntimeError: No data recevied from sensor. Check
all connections, including the SUNNY chip on the camera board

Are there any diagnostics I can run?

dmitriykovalev, 2018-01-02:
Joy Demo service starts automatically when Pi boots. It should be already working in you case, could you check with `sudo systemctl status joy_detection_demo.service`?

In order to run other examples you probably need to stop and disable this service:
```
sudo systemctl stop joy_detection_demo.service
sudo systemctl disable joy_detection_demo.service
```

annoscholten, 2018-01-02:
Hi Dimitri,

Thanks for the followup.  The service is not running because the 'Camera is not running' per: sudo systemctl status joy_detection_demo.service and my error log above.

When I connect the camera directly into the Pi 0 W (without the Vision Bonnet), the camera works ok.

When I connect the camera to the Vision Bonnet, I get 'Camera is not running'...

dmitriykovalev, 2018-01-02:
According to your log, you are trying to manually run `~/AIY-projects-python/src/examples/vision/joy/joy_detection_demo.py` command. That wouldn't work because `joy_detection_demo.service` should be already running and using the camera (it cannot be shared between processes). 

Can you please reboot the Pi and right after that run `sudo systemctl status joy_detection_demo.service` and copy the output here?

In addition, it would be very helpful if you run `dmesg` and copy output as well.

annoscholten, 2018-01-02:
Hi Dimitri,

See below:
● joy_detection_demo.service - AIY Joy Detection Demo
   Loaded: loaded (/lib/systemd/system/joy_detection_demo.service;
enabled; vendor preset: enabled)
   Active: active (running) since Tue 2018-01-02 12:36:13 CST; 52min ago
Main PID: 253 (python)
   CGroup: /system.slice/joy_detection_demo.service
           └─253 /home/pi/AIY-projects-python/env/bin/python
/home/pi/AIY-projects-python/src/examples/vision/

Jan 02 12:36:13 raspberrypi systemd[1]: Started AIY Joy Detection Demo.
Jan 02 13:24:43 raspberrypi python[253]: ERROR:root:Camera is not running.
Jan 02 13:24:43 raspberrypi python[253]: ERROR:root:CameraInference:
Camera inference is not running.
Jan 02 13:24:43 raspberrypi python[253]:
ERROR:root:StopCameraInference: Camera inference is not running.

dmesg:

pi@raspberrypi:~ $ dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.9.59+ (dc4@dc4-XPS13-9333) (gcc version
4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #1047 Sun Oct
29 11:47:10 GMT 2017
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7
(ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT
nonaliasing instruction cache
[    0.000000] OF: fdt:Machine model: Raspberry Pi Zero W Rev 1.1
[    0.000000] cma: Reserved 8 MiB at 0x17400000
[    0.000000] Memory policy: Data cache writeback
[    0.000000] On node 0 totalpages: 98304
[    0.000000] free_area_init_node: node 0, pgdat c0914e10,
node_mem_map d7c8b900
[    0.000000]   Normal zone: 864 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 98304 pages, LIFO batch:31
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.
Total pages: 97440
[    0.000000] Kernel command line: 8250.nr_uarts=0
bcm2708_fb.fbwidth=1600 bcm2708_fb.fbheight=1200 bcm2708_fb.fbswap=1
smsc95xx.macaddr=B8:27:EB:1B:63:4A vc_mem.mem_base=0x1ec00000
vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=ttyS0,115200
console=tty1 root=PARTUUID=fb718213-02 rootfstype=ext4
elevator=deadline fsck.repair=yes rootwait quiet splash
plymouth.ignore-serial-consoles
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 370980K/393216K available (5950K kernel code,
491K rwdata, 1948K rodata, 396K init, 725K bss, 14044K reserved, 8192K
cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xd8800000 - 0xff800000   ( 624 MB)
                   lowmem  : 0xc0000000 - 0xd8000000   ( 384 MB)
                   modules : 0xbf000000 - 0xc0000000   (  16 MB)
                     .text : 0xc0008000 - 0xc05d7a48   (5951 kB)
                     .init : 0xc0841000 - 0xc08a4000   ( 396 kB)
                     .data : 0xc08a4000 - 0xc091ef48   ( 492 kB)
                      .bss : 0xc091ef48 - 0xc09d4648   ( 726 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000031] sched_clock: 32 bits at 1000kHz, resolution 1000ns,
wraps every 2147483647500ns
[... 220 lines cut ...]
[   65.265198] systemd[1]: apt-daily-upgrade.timer: Adding 56min
36.194911s random time.
[   65.266046] systemd[1]: apt-daily.timer: Adding 5h 2min 6.814011s
random time.
[   69.112722] googlevisionbonnet spi0.0: Myriad booting
[   69.304790] googlevisionbonnet spi0.0: Myriad ready
[   82.342686] fuse init (API version 7.26)
[  103.850121] Registered device pwm22
[  103.885585] Starting timer (pulse).
[  103.885704] Stopping timer.
[  103.885864] Starting timer (period).
[  104.053746] Starting timer (pulse).
[  104.054327] Starting timer (period).
[  104.222224] Starting timer (pulse).
[  104.224838] Starting timer (period).
[  104.394414] Stopping timer.
[  104.401353] Unregistered device pwm22
[  309.562456] Bluetooth: HIDP (Human Interface Emulation) ver 1.2
[  309.562515] Bluetooth: HIDP socket layer initialized
[  309.632569] hid-generic 0005:045E:0705.0002: unknown main item tag 0x0
[  309.634034] input: Microsoft Wireless Entertainment Keyboard 7000
as /devices/platform/soc/20201000.serial/tty/ttyAMA0/hci0/hci0:11/0005:045E:0705.0002/input/input1
[  309.647262] hid-generic 0005:045E:0705.0002: input,hidraw1:
BLUETOOTH HID v1.0c Keyboard [Microsoft Wireless Entertainment
Keyboard 7000] on b8:27:eb:b1:c9:e0

annoscholten, 2018-01-02:
Dimitri - that was right after a reboot and no other manual commands.

Anno

dmitriykovalev, 2018-01-02:
Perfect, thank you! Looks like you connected camera cables in a wrong way. Most likely this is the cable between Pi and Vision Bonnet. You should be able to see contact pads from both sides (if look from top and from the bottom to the whole assembly). Can you please attach a photograph?

Just in case, please also verify connection between Vision Bonnet and Camera. In this case you shouldn't see contact pads from the top (it's opposite to the short cable between Vision Bonnet and Pi).

annoscholten, 2018-01-02:
Hi Dimitri,

I believe I have them correct (per your notes above), but please see attached photos.
Anno
[image]
[image]
[image]

dmitriykovalev, 2018-01-02:
That looks correct. Can you also verify connector on the Camera side? (most likely it's correct, otherwise camera plugged in directly to the Pi wouldn't work).  

What's the Camera module version by the way? We only support V2 for now.

annoscholten, 2018-01-02:
Hi Dimitri,

Well, your last statement may be the issue.  See photo's below, but this may not be a V2 camera?

[image]
[image]

dmitriykovalev, 2018-01-02:
Yep, that seems to be a 5MP camera. Check this [image](https://images-na.ssl-images-amazon.com/images/I/41vGRDyVq6L._SX355_.jpg), it has the same label as yours on the flex + "Rev 1.3" label.  For now we have only support for official Camera V2: https://www.raspberrypi.org/products/camera-module-v2/

annoscholten, 2018-01-02:
Yes, that's it.  Totally missed that in the instructions.....

Thank you for the attention and help on this!  Off to go get a v2 camera now ...

dmitriykovalev, 2018-01-02:
Sure, no problem! We'll improve error reporting mechanism, so it would be easier to detect. Let us know how everything goes with V2.

annoscholten, 2018-01-02:
Do you know if there'll be an issue with the: https://www.raspberrypi.org/products/pi-noir-camera-v2/

(For low light applications)

dmitriykovalev, 2018-01-02:
Pi Camera NoIR V2 should _technically_ work because it's the same sensor as Pi V2 but our internal ISP pipeline is not optimized for it.

annoscholten, 2018-01-02:
ok, thank you, I'll stick with tried and true..

annoscholten, 2018-01-07:
Just an update... I installed the V2 camera and everything runs well - including all the demos.

dmitriykovalev, 2018-01-07:
Good to know, thanks for the update! Closing issue now.

closed 2018-01-07 by dmitriykovalev

sainttelant, 2018-01-22:
@dmitriykovalev hey, guy, will you intend to figure out a way to support V1.3 camera in the future? thanks

dmitriykovalev, 2018-01-23:
@sainttelant V2.1 and V1.3 have different sensors (Sony vs OmniVision). Our internal processing is heavily optimized for V2.1 sensor. We have experimental code which can work with V1.3 but inference quality is worse. For now there are no plans to make it official.

sainttelant, 2018-01-25:
@dmitriykovalev 
thanks by the way, it seemed we have to choose V2.1 camera this time,maybe someday v1.3  supports perfectly, let us know..

santhoshnp, 2018-08-28:
hi all i need help, in my vision bonnet kit camera is not enabling. i tried su
[image]
do raspi-config but no output

dmitriykovalev, 2018-08-29:
@santhoshnp  check that your camera is proper connected physically.

santhoshnp, 2018-08-29:
Thank for reply bro,

[image]
[image]
[image]
[image]
[image]

santhoshnp, 2018-08-29:
I also have one doubt in joy detector,
while running vision kit it should be in active or inactive position,

manoj7410, 2018-08-29:
@santhoshnp One thing you can try is to connect your camera directly with PI0 (Where short flex is connected currently). Then run the command "raspistill -o output.jpg" and check if camera is working. 

By default joy_detection is enabled and make sure you stop it by running the command "sudo systemctl stop joy_detection_demo"

santhoshnp, 2018-08-29:
@manoj bro camera is working well in pi0 but camera is not enabling in vision bonnet i don't why.

dmitriykovalev, 2018-08-29:
@santhoshnp check camera connector latches for short flex cable (Pi board and Vision Bonnet board), they must be locked (not loose).

santhoshnp, 2018-08-30:
@dmitri
raspistill -w 1640 -h 922 -o image.jpg
mmal: Cannot read camera info, keeping the defaults for OV5647
mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
mmal: Failed to create camera component
mmal: main: Failed to create camera component
mmal: Camera is not detected. Please check carefully the camera module is installed correctly

santhoshnp, 2018-08-30:
any one help me how to update raspistill camera app v1.3.11 why because i am using v2.1
please help me
raspistill -v
mmal: Cannot read camera info, keeping the defaults for OV5647

raspistill Camera App v1.3.11

Width 2592, Height 1944, quality 85, filename (null)
Time delay 5000, Raw no
Thumbnail enabled Yes, width 64, height 48, quality 35
Link to latest frame enabled  no
Full resolution preview No
Capture method : Single capture

Preview Yes, Full screen Yes
Preview window 0,0,1024,768
Opacity 255
Sharpness 0, Contrast 0, Brightness 50
Saturation 0, ISO 0, Video Stabilisation No, Exposure compensation 0
Exposure Mode 'auto', AWB Mode 'auto', Image Effect 'none'
Flicker Avoid Mode 'off'
Metering Mode 'average', Colour Effect Enabled No with U = 128, V = 128
Rotation 0, hflip No, vflip No
ROI x 0.000000, y 0.000000, w 1.000000 h 1.000000
mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
mmal: Failed to create camera component
mmal: main: Failed to create camera component
mmal: Camera is not detected. Please check carefully the camera module is installed correctly

dmitriykovalev, 2018-08-30:
@santhoshnp check flex connectors.

santhoshnp, 2018-08-31:
I think my camera version 2.1 but when I type raspistill -v (display the version as v1,3)
I think this is the problem.
can any one help me

[image]

trsbalaji, 2018-10-26:
Hi, I bought vision kit version 1.1. Normal Raspberry pi zero w boots fine. Once connected  MIPI flat flex cable to Raspberrypi zero w board is not booting.

https://aiyprojects.withgoogle.com/vision-v1/

Please help me.

magdesign, 2019-05-06:
Answer to [post](https://github.com/google/aiyprojects-raspbian/issues/224#issuecomment-354880794)

**Pi Camera NoIR V2  does not work.**

ssanzr, 2019-05-11:
@dmitriykovalev, have you tried the Pi Camera NoIR V2? According to @magdesign it does not work.

Has anyone tried it?

Thanks!