google/aiyprojects-raspbian #225: aiyprojects image - updating /home/pi/AIY-projects-python - git?

sheridat, 2018-01-02:
I've always used your supplied images for the software on my voicekit.

I've always updated the software by changing directory into the software folders (which have always had a hidden git directory) and then issuing a couple of git commands (and they are the only git commands I know). The commands
git checkout "name of the branch"
git pull origin "name of the branch"
That has always worked beautifully.

I am now running the voicekit from the aiyprojects image - but i've noticed that /home/pi/AIY-projects-python folder does not have a git folder - but it does have a makefile.

 Is it a fault that there is no git folder or should I now be doing something else?
Thank You

PeterMalkin, 2018-01-02:
Sheridat,

It is by default that we do not include the full revision history (.git folder) in the image. 
You would need to add the git remote and pull just as you were doing before.

But you bring up a good point. If you think that pulling from github directly to the pi is your preferred method of getting the updates to your sdcard images, instead of reimaging the card, then we can change the release process around to accomodate.

Or just let me know if you need help with adding a git remote and pulling, I can give you a step by step

sheridat, 2018-01-03:
Peter, Thanks for responding.
In the early days of the voicekit we (the raspberrypi.org AIYprojects forum)  needed help with how to update - see https://github.com/google/aiyprojects-raspbian/issues/72. The info was published here http://eduncan911.com/stem/embedded/aiy-google-assistant-upgrades-may-2017.html. Once the voicekit branch was introduced all we had to do was advise people to change the branch name.

Updating by git pulling is definitely my preference. Whenever a merge takes place for the voicekit repo I pull it within hours and if it adds new functionality I normally post details about it on the forum.

From a continuity perspective it would be great if we could carry on as before (the image has the .git folder).
If we went down the ( each individual) adding a git remote route the ideal solution from my perspective would be  that a section called something like "keeping the software up to date" was added to https://aiyprojects.withgoogle.com/voice/ 

A similar section in the https://aiyprojects.withgoogle.com/vision/ would be equally appreciated.

The documentation is crucial to people running and maintaining these wonderful devices.

sheridat, 2018-01-05:
@PeterMalkin @drigz 
Thank You for providing a new image containing the git folder - just updated for 233 and 234 - flawless.

closed 2018-01-05 by sheridat