google/aiyprojects-raspbian #229: So far a disaster for me. Apparently fried board or something cause google mislabelled the 'Pi' endof the cable, also switched out a new camera cable, - which way does that one go ', what image do I need and where cause nothing runs now.

jhewitt123, 2018-01-03:


PeterMalkin, 2018-01-03:
Very sorry to hear it.

There is a chance that the sticker on the cable was applied incorrectly. We have updated the instructions on the website to reflect this. Please follow the Assembly guide, as it is on https://aiyprojects.withgoogle.com/vision/#assembly-guide-1-get-the-vision-kit-sd-image

jhewitt123, 2018-01-03:
> withgoogle.com/vision/#assembly-guide-1-get-the-vision-kit-sd-image

Thanks for your attention Peter, the three issues I've identified are;
-does shorting the tiny cable fry the Google Vision HAT itself?
-the longer cable camera is completely new and different from the gosh darn
guide, how does it go? up or down?
-my image doesn't start upo any programs, the modules don;t load, I can;t
run them from command line, nor from Python 2 or 3 or the other python
program, I tried all. what image do I need, where do I get it, when will it
be available, and can I somehow re-burn it over my existing SD card image
or do I need a new card.

I went threw troubleshooting onthe older VoiceHAT when I wanted to do GPIO
like advertised and google wouldn;t provide the correct transistor output
circuit schematic to do so, I fried two boards.

Please help.

lleonid, 2018-01-03:
-Plugging tiny cable the wrong way shouldn't fry Pi or Vision Hat
-All flex cable should be plugged in with pins facing the boards (Pi, Bonnet, Camera).
- I'm not sure what image are you using. The one referenced at https://aiyprojects.withgoogle.com/vision/#assembly-guide-1-get-the-vision-kit-sd-image should be working. It also suggests to use etcher.io or similar to easily burn it to sd card.

jhewitt123, 2018-01-03:
Awesome I will try it. Appreciate the feedback. As with the VoiceHAT, do
you have details on using the VisionHAT GPIO? It was tricky to drive
outputs like relays previously, even with opto-isolated inputs.

jhewitt123, 2018-01-03:
And yes, I am using the 12-18  image already, I dowloaded a new 12-18 that
says last modified on 12-30 but didn;t burn it, my old image says 12-18
last modified 12-18, do I need the new 12-18?If so agiain, do I need a new
SD card on can etcher overwrite the old image.
THe problem is nothing runs, No camera action whenI look into it, can;t
start other programs when run. How are they supposed to RUN, from the
command line, frominside Pythin GUI, HOW?
They just give run and compile erros, can;t load modules.

??????

PeterMalkin, 2018-01-03:
jhewitt123,

here's a troubleshooting guide. Try to start there and narrow down the problem, we will try to help
https://aiyprojects.withgoogle.com/vision/#assembly-guide-8-troubleshooting

jhewitt123, 2018-01-04:
Thanks, actually I can't run any of the demos still, other than the default
joy detector,

no matter how I put in a command it just says 'not found',
 even when I follow instructions exactly and run in the environment, for
example, anfter running

source ~/AIY-projects-python/env/bin/activate

and then something like:

~/AIY-projects-python/src/examples/vision/face_detection_camera.py
--num_frames=10

drigz, 2018-01-04:
@jhewitt123 Could you post a screenshot of the terminal with the error message?

jhewitt123, 2018-01-04:
Sure, thanks,
 I did stop the joy demo although I did not run the 'disable' command cause
at least I have the joy demo, wouldn;t know how to restart or re-enable it
anyway you know.

...........................................................

(env) pi@raspberrypi:/ $
(env) pi@raspberrypi:/ $
(env) pi@raspberrypi:/ $
(env) pi@raspberrypi:/ $
(env) pi@raspberrypi:/ $
(env) pi@raspberrypi:/ $
(env) pi@raspberrypi:/ $

(env) pi@raspberrypi:/ $
(env) pi@raspberrypi:/ $
(env) pi@raspberrypi:/ $
(env) pi@raspberrypi:/ $ sudo
~/AIY-projects-python/src/examples/vision/face_detection_camera.py
--num_frames=10
sudo:
/home/pi/AIY-projects-python/src/examples/vision/face_detection_camera.py:
command not found
(env) pi@raspberrypi:/ $

drigz, 2018-01-04:
@jhewitt123 Does it work if you run without `sudo`? For safety, `sudo` ignores things like the virtualenv.

Alternatively, you could try:

```
cd ~/AIY-projects-python
sudo env/bin/python src/examples/vision/face_detection_camera.py
```

(although I haven't tried that myself).

jhewitt123, 2018-01-04:
You are a genius Rodrigo! Not sure why that works but it does, the
instructions in the guide didn't work for me.
Also, would really liketo run headless like in the guide for doing SSH
through the usb cable but really, the guide falls short, I mean what do you
actually dotoconnect through windows or linux or something and type in the
IP and actually connect I wonder?
You mention DHCP etc but that ain;t enough to actually do it.

drigz, 2018-01-04:
@jhewitt123 Can you clarify what worked for you? If you need the second command I posted using `sudo`, can you explain what happens if you run without `sudo`? We'd like to understand why the intructions in the guide aren't working.

Re: SSH, I'd recommend the Pi docs: https://www.raspberrypi.org/documentation/remote-access/ssh/

Let us know if those help and we can amend the instructions on the website.

jhewitt123, 2018-01-04:
> documentation/remote-access/ssh/

Sure, will look into a little later, needto shovel out of this fun
northeast climate blizzard.
Things run well using VNC, however the video itself doesn;t tunnel through
the session(if that is how to say it) likeother Pi video overlays, although
I think there are fixes around that already for server programs I have used
like PiCam

drigz, 2018-01-04:
OK, thanks for bearing with us on this and good luck in the blizzard. Here in Munich, people are complaining that there's not enough snow for good skiing - maybe you could send some our way?

jhewitt123, 2018-01-04:
Sure happy to help.

when I remove the'sudo' this is what I get, althoughIshould mention that
when I first halt the previous running face_dection_....the video overlay
isstill present on the screen, (maybe haveto kill that process somehow
first and then try again?)

...............................................................
  1123  0.0  1.1   6128  4276 pts/0    Ss   Jan03   0:03 bash
pi        1131  0.0  1.0   6120  4136 pts/1    Ss+  Jan03   0:00 bash
root     20587  0.3  0.0      0     0 ?        S    13:40   0:10
[kworker/u2:2]
root     20601  0.3  0.0      0     0 ?        S    13:46   0:08
[kworker/u2:1]
root     20606  0.0  0.0      0     0 ?        S    13:47   0:00
[kworker/0:2]
root     20623  0.0  0.8   7208  3368 pts/0    T    13:53   0:00 sudo
env/bin/python src/examples/vision/face_detection_camera.py
root     20627 18.8  7.5 101932 28532 pts/0    Tl   13:53   6:17
env/bin/python src/examples/vision/face_detection_camera.py
root     20703  0.0  0.0      0     0 ?        S    14:17   0:00
[kworker/0:0]
root     20720  0.6  0.0      0     0 ?        S    14:24   0:01
[kworker/u2:0]
pi       20733  0.0  0.7   7724  2944 pts/0    R+   14:27   0:00 ps -aux
(env) pi@raspberrypi:~/AIY-projects-python $ env/bin/python
src/examples/vision/face_detection_camera.py
mmal: mmal_vc_port_enable: failed to enable port vc.null_sink:in:0(OPQV):
ENOSPC
mmal: mmal_port_enable: failed to enable connected port
(vc.null_sink:in:0(OPQV))0x11563a0 (ENOSPC)
mmal: mmal_connection_enable: output port couldn't be enabled
Traceback (most recent call last):
  File "src/examples/vision/face_detection_camera.py", line 88, in <module>
    main()
  File "src/examples/vision/face_detection_camera.py", line 43, in main
    with PiCamera() as camera:
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 433, in
__init__
    self._init_preview()
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 513, in
_init_preview
    self, self._camera.outputs[self.CAMERA_PREVIEW_PORT])
  File "/usr/lib/python3/dist-packages/picamera/renderers.py", line 558, in
__init__
    self.renderer.inputs[0].connect(source).enable()
  File "/usr/lib/python3/dist-packages/picamera/mmalobj.py", line 2212, in
enable
    prefix="Failed to enable connection")
  File "/usr/lib/python3/dist-packages/picamera/exc.py", line 184, in
mmal_check
    raise PiCameraMMALError(status, prefix)
picamera.exc.PiCameraMMALError: Failed to enable connection: Out of
resources
(env) pi@raspberrypi:~/AIY-projects-python $

drigz, 2018-01-04:
@dmitriykovalev, any idea why @jhewitt123 needs to run with `sudo`?

jhewitt123, 2018-01-04:
I think first I have to figure out how to stop the camera video overlay
once I try to quit the program. Maybe useful to know more about the camera
program used in this image?

jhewitt123, 2018-01-04:
Okay just killed the whole terminal and opened a new one and i CAN run it
WITHOUT sudo so that works,

dmitriykovalev, 2018-01-04:
@jhewitt123, you can also try to run the following:

1. Setup our environment:

`~/bin/AIY-projects-shell.sh`

2. Run face detection camera demo:

`python3 src/examples/vision/face_detection_camera.py`

Press `Ctrl-C` to stop it.

dmitriykovalev, 2018-01-22:
Please reopen if problem still exists.

closed 2018-01-22 by dmitriykovalev

jhewitt123, 2018-02-25:
Hi folks, I was wondering if the new code spoken of previously is available
for the AIY vision board.
Mainly can we do simple image classification in real time using the camera
instead of using a file?
It seems this guy in the video has clearly done it, see the link in the
video, but I don;t know who he is, is he google?
More impoartantly how do we get the vision kit to work with the voice kit
(or a Home Mini for that matter), so the thing can see what objects it is
seeing. I figured that was pretty much the point of all this?
Thanks again for your follow up,
John Hewitt

PeterMalkin, 2018-02-25:
Hi John!

The code is ready, we will be pushing it in the next day or two. I tried to push it on Friday, but hit an unexpected snag with GitHub permissions. trying to figure it out now.

jhewitt123, 2018-02-25:
Awesome!
That video I had linked to showedd how todo it using the V1 camera so I am
hesitant to try it with the normal V2 camera, so I'll wait for your code.
How do project developers and hobbyists like me know when or where new code
is available in general?