google/aiyprojects-raspbian #237: Update aiyprojects please

amerinoo, 2018-01-04:
Hello,
You put in the README " The newest code is in the aiyprojects branch, which supports all AIY kits. The voicekit branch is deprecated, and users should switch to the aiyprojects" so, could you update the code with the new improvements please?
This line is deprecated "with Assistant(credentials, model_id) as assistant:" and it is in aiyprojects branch and the examples don't work.

Thank you.

drigz, 2018-01-24:
Sorry about the delay! I've got changes ready for this 
(and for the update to v1alpha2 of the Assistant gRPC API, which has my favourite feature - changing languages between requests with `aiy.i18n.set_language_code`) and they're on their way through testing & release.

The new process for testing code before we push it to GitHub is slower, but it should avoid some of the breakages we've had before.

janvda, 2018-02-05:
Any idea when the aiyprojects branch will be updated ?
Currently my aiyprojects installation is no longer working as I have upgraded to Assistant SDK v0.1.0.