google/aiyprojects-raspbian #238: ImportError: No module named 'aiy'

amerinoo, 2018-01-04:
As you can see in the image there is some problem with the imports. I'm using **aiyprojects** branch.

[image]

sheridat, 2018-01-05:
Hi, If i do what you have done it get the same error 
try
1. cd /home/pi/AIY-projects-python
2. source env/bin/activate
3. src/examples/voice/assistant_library_demo.py

amerinoo, 2018-01-05:
I have already activated the virtualenv. You can se in the prompt (env) pi@raspberrypi: indicating the virtualenv is working.

sheridat, 2018-01-05:
Just suggesting you drop your ssh connection then log back in and do the sequence above.
I'm on the aiyprojects branch and doing it that way works for me.

amerinoo, 2018-01-05:
Ok thanks, I will try this and I will give you the feedback.  Thanks sheridat!

amerinoo, 2018-01-05:
Done. I try it with your same commands and I got the same result...T_T

drigz, 2018-01-05:
@amerinoo, can you try this command?

```
echo "/home/pi/AIY-projects-python/src" > \
  /home/pi/AIY-projects-python/env/lib/python3.5/site-packages/aiy.pth
```

If that doesn't work, try recreating the virtualenv:

```
cd /home/pi/AIY-projects-python
mv env env.bak
./scripts/install-deps.sh
source env/bin/activate
src/examples/voice/assistant_library_demo.py
```

You can also try with the latest [2018-01-03](https://dl.google.com/dl/aiyprojects/vision/aiyprojects-2018-01-03.img.xz) image.

If one of these works, let us know so others can try the same thing.

amerinoo, 2018-01-05:
Hello @drigz I tried eh echo command and this is the output.
[image]
After see this output I recreated the env as you said but with the same wrong result. 
My next step will be upgrade the image with the latest and see if I can resolve the problem. However, I think should be great to know the cause of my problem because maybe another user have the same problem.
I would also like to take this opportunity to say you should update the  [Voice Kit SD image](https://www.raspberrypi.org/forums/viewtopic.php?t=193379) with the latest [2018-01-03](https://dl.google.com/dl/aiyprojects/vision/aiyprojects-2018-01-03.img.xz)

drigz, 2018-01-05:
I could be that this is a problem that occurs when you switch 2017-09-11 image to use the aiyprojects branch instead of the voicekit branch, due to some changes in path.

If anyone else runs in to it they could try:

```
echo "/home/pi/AIY-voice-kit-python/src" > \
  /home/pi/AIY-voice-kit-python/env/lib/python3.5/site-packages/aiy.pth
```

but updating the image is very likely to fix the issue.

kc9xg, 2018-01-05:
I have the same issue using aiyprojects-2018-01-03.img.xz on s Rpi0 W v 1.3

When I disable joy_detection_demo.service and try to run joy_detection_demo.py I get import error, no module named 'aiy'.  No idea what I am doing incorrectly, or if the source file contains errors.

maginator, 2018-01-07:
Same on me. Try it with the aiyprojects-2018-01-03.img. Check_audio.py say no modul aiy.audio.

OscarVanL, 2018-01-07:
I've got the same problem. I'm not using an image but trying to install on my own existing Raspbian installation using the instructions here: https://github.com/google/aiyprojects-raspbian/blob/voicekit/HACKING.md

drigz, 2018-01-08:
@kc9xg, @maginator: could you tell me exactly how you're running the scripts? They should be run from within virtualenv / dev terminal. Does "Check Audio" work from the Desktop?

@OscarVanL If you're having trouble using another installation, you can try asking for support on the [MagPi forums](https://www.raspberrypi.org/forums/viewforum.php?f=114).

OscarVanL, 2018-01-08:
I just went through the installation process once again very carefully. I think I've found the issue.

When executing scripts/install-deps.sh , at the end of the execution I'm given this error:

> scripts/install-deps.sh: line 35: /home/pi/AIY-projects-python/env/lib/python3.5/site-packages/aiy.pth: No such file or directory

It appears this error is due to the install-deps.sh program using the folder **AIY-projects-python**, despite the instructions [here](https://github.com/google/aiyprojects-raspbian/blob/voicekit/HACKING.md) saying to name the folder **voice-recognizer-raspi**

If we modify install-deps.sh and change any reference to AIY-projects-python to voice-recognizer-raspi, the installation completes successfully and the issue in this thread is resolved.

It seems this fix is also mentioned in https://github.com/google/aiyprojects-raspbian/pull/212 however is yet to be merged.

PaulFinch, 2018-01-09:
I compared the check audio script with the one in voicekit branch and I added this line just before importing aiy module like [Here](https://github.com/google/aiyprojects-raspbian/blob/voicekit/checkpoints/check_audio.py) :

`sys.path.append(os.path.realpath(os.path.join(__file__, '..', '..')) + '/src/')`

I also used `sudo` to launch the script as it seems to write inthe /boot/config.txt file.
The script ran "normally" after that ...

OscarVanL, 2018-01-09:
Thanks Paul, your solution worked for me.

PaulFinch, 2018-01-09:
Unfortunately, i'm stuck with the device_module_id issue [Here](https://github.com/google/aiyprojects-raspbian/issues/202)

Looks like the Google API changed, and the code in this repo has not been updated accordingly yet

amerinoo, 2018-01-12:
With the latest [2018-01-03](https://dl.google.com/dl/aiyprojects/vision/aiyprojects-2018-01-03.img.xz) image all works fine on my Raspberry Pi 3. I'm using just the voice examples as I have the Voice Kit.

closed 2018-01-12 by amerinoo

PaulFinch, 2018-01-17:
Are you sure your check_audio.py works with the last version from this repo ?
Maybe this works with `source env/bin/activate` but the HACKING.md file does not precise it :

`cd ~/AIY-projects-python
sudo scripts/install-alsa-config.sh
python3 checkpoints/check_audio.py
sudo reboot`

drigz, 2018-01-17:
Thanks, for pointing that out @PaulFinch! The issue is in HACKING.md.

vladlee098, 2018-01-20:
Worked for me after "source env/bin/activate" command, thanks for the clue

Suncoast-Cyclist, 2018-01-23:
I've just started trying to get the AIY Vision kit working and found the same error about not being able to find the aiy import when I tried to run the python scripts from the vision examples.  

I tried running the check_audio.py from @PaulFinch and the import does find aiy but then python complains later when it tries to use the audio hardware which is not present on my vision device.

I still can't get any of the vision examples to run.

drigz, 2018-01-23:
@Suncoast-Cyclist Are you using the "dev terminal" from the desktop shortcut?

Suncoast-Cyclist, 2018-01-23:
@drigz 

I hadn't been using the dev terminal, but I did use the command to manually establish the environment that was specified in the vision kit instructions.

source ~/AIY-projects-python/env/bin/activate

I just tried the dev terminal icon on the desktop and it still didn't work for the vision examples.

It sounds like the people using the voice kit examples are helped by running with the dev terminal.

Should there be two different virtual environments, one for the voice kit and something different for the vision kit?

drigz, 2018-01-23:
@Suncoast-Cyclist, I don't know a lot about the vision kit's internals, unfortunately. Are you using the official image?

Suncoast-Cyclist, 2018-01-23:
@drigz, Yes I think I have the latest aiyprojects-2018-01-03.img.  All the voice support is there along with the new vision support.  

I have been thinking about updating my voice kit to also use this new image, but since it is working right now, I don't want to risk breaking the voice kit unless there are some advantages to installing the update.

drigz, 2018-01-23:
@Suncoast-Cyclist Could you post the error you get from running the vision examples inside the dev terminal? Is it the `No module named "aiy"` error?

Suncoast-Cyclist, 2018-01-23:
@drigz Yes, it sure looks like the exact same error.

`  File "src/examples/vision/buzzer/buzzer_demo.py", line 21, in <module>
    import aiy.toneplayer
ImportError: No module named 'aiy'
`

Suncoast-Cyclist, 2018-01-23:
@drigz I'm making some progress.  Several days ago @PaulFinch suggested putting the following line immediately before the import `sys.path.append(os.path.realpath(os.path.join(__file__, '..', '..')) + '/src/')`

I tried that and it didn't help me because my source code wasn't at the same relative level in the directory structure.  I decided to use brute force and used `sys.path.append('/home/pi/AIY-projects-python/src')`  and now my python script runs.

As a retired professional programmer I wouldn't advise using an approach line this in general.  However, it does work as a quick fix to get on with trying out the rest of the example code.

drigz, 2018-01-24:
@Suncoast-Cyclist It seems the issue is that you ran the script with `sudo`, which ignores the virtualenv. An alternative would be to run `sudo env/bin/python src/examples/...` - this sets up the virtualenv inside `sudo`.

Why are you using `sudo` for the scripts? Do they work without it?

Suncoast-Cyclist, 2018-01-24:
@drigz  My background is mainly in Windows and C/C++.  I don't know a lot about Linux since I just got my first Raspberry Pi recently.  I'm picking it up gradually by reading and following advice I find on various forums.

I don't understand all the things that go on with sudo.  The basic reason that I use sudo is to get root authority when it is required to perform some operation.  The script I was trying to run failed because the import couldn't find the aiy module.  I was originally trying to run the script from a standard terminal session.  I soon found the instruction in the Vision kit instructions to use `source ~/AIY-projects-python/env/bin/activate` but that didn't seen to get the import to work.

Then I found this GitHub issue and the tip @PaulFinch gave about sys.path.append.  That didn't work either but when I realized what it was trying to do I changed the value to the literal path to where I knew the aiy module was located.  I'm fairly certain that the sys.path.append function requires root authority and so I started using sudo.  That got a very simple example from the Vision project to run.   When I tried one of the more complex scripts I ran into another import failure for a google module and I wasn't able to use the same technique to resolve that.

There is something fairly fundamental that I am missing here.  I agree that using sys.path.append should not be necessary.  I need to take a step back and try to use the virtual environment. 

If I understand the import operation correctly, the current working directory is automatically part of the path that is searched for imports so it may make a difference where the script is run.

drigz, 2018-01-24:
@Suncoast-Cyclist Could you try running the unmodified scripts from within the dev terminal, without `sudo`? You shouldn't need `sudo` for this. (It's not required for `sys.path` - that's just telling Python where to look for modules and doesn't require `sudo`)

If you want to try without your local modifications, you can run `git stash`, and then use `git stash pop` to get them back.

Suncoast-Cyclist, 2018-01-24:
@drigz I tried running my script from the dev terminal command line with the sys.path.append still in it.  It failed with a permission denied message.  Then I commented out the two lines I had added and tried it again and got the same permission denied error.  When I tried running it in Thonny I got the error about not being able to find the aiy module.  Finally I just ran the python3 command from the src directory and it ran correctly.  As an additional test I did a cd .. and ran the python3 command after adjusting the path to the script by adding src/ before it and it also ran correctly.  It looks like you can't run one of these scripts as a command.  I thought that the first line of the script was supposed to ensure that it was run by the correct program. `#!/usr/bin/env python`  Is this possibly what is sending things off into the ditch?

drigz, 2018-01-25:
@Suncoast-Cyclist When I try running in the dev terminal with `python`, `python3` and running the scripts directly, it works all three ways. Which script exactly were you having trouble with?

The "permission denied" errors are probably an error in the script permissions - they're not executable, so sudo won't help, it'll just said an ImportError. Thanks for pointing me to this - I'll get it fixed!

Suncoast-Cyclist, 2018-01-25:
@drigz  The python script where I found my problems with was in: ~/AIY-projects-python/src/examples/vision/buzzer.  When I checked the permissions I found that neither of the python scripts has any 'x' permissions.  I had forgotten about execute permissions since I have been so brain-washed by my years in the Windows environment where there are certain file types that are designated as executable.  FWIW, I don't think that any of the python scripts in the src/examples/vision area are executable. Since those programs are meant as samples to be played around with, it probably isn't a big deal.  If there is a way to get them to compile properly with Thonny, it would make it a lot easier to experiment with them.

Maybe it is just the fact that I am a Linus novice and was just confused when everyone else just knows that you need to invoke the python program to run a python script and wouldn't have run into this and thought it was a problem.  Shell scripts are normally considered directly executable and the first line indicates which shell it was written for.  I saw a similar line at the beginning of the python scripts and assumed that they could be directly executed but I never thought to check for execute permissions.  I just answered my own question by adding execute permission to buzzer_demo.py and when I entered the command ./buzzer_demo.py it worked this time.  Oh, no! After being out of school for over 50 years, I'm back in classes again. ;-)

drigz, 2018-01-26:
@Suncoast-Cyclist We've just made the scripts executable in our internal repo, so that will be published in the next image soon. We also have a plan for installing the `aiy` module globally, which would let Thonny work out-of-the-box. In the short term, you can try setting "Tools -> Options -> Interpreter" to `~/AIY-projects-python/env/bin/python3`. (I haven't had a chance to test this, so let me know if it works)

Lots of our users are Linux and/or Python novices so this sort of feedback is super valuable - thank you!

Suncoast-Cyclist, 2018-01-27:
@drigz Yes, the Tools->options->Interpreter override does work.

Thanks for your help