google/aiyprojects-raspbian #240: threads can only be started once

ZiYiBear, 2018-01-05:
`#!/usr/bin/env python3

import logging

import aiy.audio as audioControl

logging.basicConfig(
    level=logging.INFO,
    format="[%(asctime)s] %(levelname)s:%(name)s:%(message)s"
)   

def main():
    while True:
        print('recording')
        audioControl.record_to_wave('test.wav', 10)
        print('recording finish')
        aiy.audio.play_wave('test.wav')
        
    
if __name__ == '__main__':
	main()`

I tried to let it keep recoding and play the recorded file, but somehow it just can record an play the wav sound on first time the code run.

if remove the while loop, it works fine, but i need it to keep recording and play the recorded sound, how can i fix it? thanks

d-amend, 2018-01-16:
it's the same here.

Sarahli19, 2018-01-16:
Also for me - there is a problem when calling record_to_wave for the second time in in _recorder.py with the self.start()

ocassano, 2018-02-11:
Hi. I'm facing the same issue.

Any solution / work around on this ?

Thanks.

dfresh613, 2018-10-28:
Hello! I too ran into this problem. I Dug around the code a little bit and found a workaround.

You must set `aiy.audio._voicehat_recorder = None` anytime you close the recorder or stop the thread:

```
recorder = aiy.audio.get_recorder()
recorder.start()
...
recorder.stop()
recorder.stop()
aiy.audio._voicehat_recorder = None
```

The problem is that the _drivers._recorder is a subclass of threading.Thread, and it's not being treated properly by aiy.auido.

After running get_recorder, you can call recorder.stop(), and  _drivers._recorder will clean itself up properly, and is ready to be thrown away, but is never disposed of by aiy.audio:

```
def get_recorder():
    """Returns a driver to control the VoiceHat microphones.

    The aiy modules automatically use this recorder. So usually you do not need to
    use this.
    """
    global _voicehat_recorder
    
    if not _voicehat_recorder:
        _voicehat_recorder = aiy._drivers._recorder.Recorder()
        
    return _voicehat_recorder
```

As seen above, _get_recorder will always return the same thread instance of _voicehat_recorder, even if that thread has been stopped and cleaned up. 

Suggested fix would be to have a close_recorder() method in aiy.audio. This method would stop the recorder thread, and set _voicehat_recorder to None.
The

dfresh613, 2018-10-28:
An example workaround more specific for the OPs problem:

```
import aiy
while True:
        print('recording')
        aiy.audio.record_to_wave('test.wav', 10)
        print('recording finish')
        aiy.audio._voicehat_recorder.stop()
        aiy.audio._voicehat_recorder = None
        aiy.audio.play_wave('test.wav')
```

ocassano, 2018-10-29:
It's working by adding 
        aiy.audio._voicehat_recorder.stop()
        aiy.audio._voicehat_recorder = None

Thank you dfresh613 for the workaround.

dmitriykovalev, 2018-11-17:
There a new recorder API in [voice/audio.py](url) now:

https://github.com/google/aiyprojects-raspbian/blob/5c74ce5154fe9ecdab5901e01b567d47fbdc068f/src/aiy/voice/audio.py#L132

closed 2018-11-17 by dmitriykovalev