google/aiyprojects-raspbian #241: Failed to install aiyproject scripts

jjfretired, 2018-01-05:
Could not install scripts when command issued with sudo scripts/...

When logged is as su ( sudo su) and ran the script it executed without errors.
It seems that $EUID is not being evaluated properly and causes the script to fail.
[error.txt](https://github.com/google/aiyprojects-raspbian/files/1607840/error.txt)

jtgans, 2018-01-05:
Looks like running that script under sudo isn't running it under bash:

```
+ set -o errexit
+ [[ -ne 0 ]]
scripts/install-services.sh: 22: scripts/install-services.sh: [[: not found
scripts/install-services.sh: 27: scripts/install-services.sh: Bad substitution
```

The affected lines are:

```
set -o errexit

if [[ $EUID -ne 0 ]]; then
   echo "This script must be run as root (use sudo)" 1>&2
   exit 1
fi
```
The [[ ]] are bash-isms. My guess is that this is actually being run through something like dash or ash. Oddly, we do have a shebang at the top of the script to /bin/bash, so that shouldn't be the case.

divx118, 2018-01-06:
IMO it should be better to make all shell scripts POSIX compliant, to avoid such problems.

jjfretired, 2018-01-06:
My problem.  I placed the set -x as line 1 which negated the shebang.  Admin UNIX many years age.

closed 2018-01-06 by jjfretired