google/aiyprojects-raspbian #243: Shutdown/crash logging

Kattihatt, 2018-01-07:
So sometimes when I use Google assistant it breaks of mid response or doesn't respond at all. After that it becomes completely unresponsive and the RPi becomes unreachable and shuts down. I don't know if the shutdown issue is directly connected to the use of Google assistant but it's the only connection I can see. I would like to see what causes this but I can't find the log files or any other indication of what could be wrong. I know how to use journalctl -f for looking for issues but this hardly helps when the system shuts down and I need to see the logfiles. How would I go about this to find the issues?

divx118, 2018-01-08:
You could check `/var/log/daemon.log` the journalctl logs should also be in there. Older logs when the file gets to big are gun zipped for example `/var/log/daemon.log.1.gz`

drigz, 2018-01-08:
If the RPi is shutting down, it probably means that your power supply isn't supplying enough current. What kind of power supply do you have? We recommend a 2.5A power supply.

Kattihatt, 2018-01-09:
@drigz Feeding it through a LM2596 fed directly to the GPIO-pins, so I've never had any problems with supplying power. The buck converter can do 3 A continuously and above for short periods. It dies in connection to me using Google assistant and this last day, I've had to restart the service three times after it becoming unresponsive. The RPi didn't shut down though.

drigz, 2018-01-09:
Could be the same as #188.

Kattihatt, 2018-01-09:
@drigz No, but I've experienced that as well. After losing connection I have to manually restart the service for it to work.

" 26 Jan  7 06:25:38 Jesus13 python3[16902]: During handling of the above exception, another exception occurred:
    27 Jan  7 06:25:38 Jesus13 python3[16902]: Traceback (most recent call last):
    28 Jan  7 06:25:38 Jesus13 python3[16902]:   File "/home/pi/voice-recognizer-raspi/env/lib/python3.5/site-packages/requests/adapters.py", line 438, in send
    29 Jan  7 06:25:38 Jesus13 python3[16902]:     timeout=timeout
    30 Jan  7 06:25:38 Jesus13 python3[16902]:   File "/home/pi/voice-recognizer-raspi/env/lib/python3.5/site-packages/requests/packages/urllib3/connectionpool.py", line 649, in urlopen
    31 Jan  7 06:25:38 Jesus13 python3[16902]:     _stacktrace=sys.exc_info()[2])
    32 Jan  7 06:25:38 Jesus13 python3[16902]:   File "/home/pi/voice-recognizer-raspi/env/lib/python3.5/site-packages/requests/packages/urllib3/util/retry.py", line 388, in increment
    33 Jan  7 06:25:38 Jesus13 python3[16902]:     raise MaxRetryError(_pool, url, error or ResponseError(cause))
    34 Jan  7 06:25:38 Jesus13 python3[16902]: requests.packages.urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='accounts.google.com', port=443): Max retries exceeded with url: /o/oauth2/token (Caused by NewConnectionError('<       requests.packages.urllib3.connection.VerifiedHTTPSConnection object at 0x75c35490>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution',))"

drigz, 2018-01-09:
OK, thanks for the clarifications. By the way, if you wrap multiline code snippets with three backticks, it renders properly. eg:

    ```
    line one
    line two
    ```

manoj7410, 2019-11-14:
@Kattihatt Not sure if you are still working on this project but is this also reproducible with latest AIY image https://github.com/google/aiyprojects-raspbian/releases/tag/v20191113 ?