google/aiyprojects-raspbian #246: How I get the Vision-kit?

alephman, 2018-01-08:
The vision-kit seems to be very cool, but it's not available out of US. In mean while, just check the kit in the micro-center website and it is also not available. So is there easy and quick way to buy it?
 
thanks!

drigz, 2018-01-08:
We're working on this - if you [sign up for the waitlist](https://services.google.com/fb/forms/aiycommunicationpreferences), you'll hear when the kits are available for purchase.

closed 2018-01-08 by drigz

larstbone, 2018-03-10:
Where in the US is the Vision Kit?  I signed up at MicroCenter and by the time I got to the store, they had baled on the Vision Kit.  Only place I can find it is from scalpers on eBay wanting over $200.

PeterMalkin, 2018-03-11:
Please be patient with us. We are working with our manufacturing partner to produce more inventory. In the VERY near future you will be able to purchase more kits.

alephman, 2018-04-03:
Just asking when it is ready?