google/aiyprojects-raspbian #247: V2 raspberry pi camera not working with Vision Bonnet

MishaLaskin, 2018-01-08:
I went through the assembly guide and got the image running correctly

`dmesg | grep googlevisionbonnet` returns the desired output

However, it seems that the Vision Bonnet doesn't detect a connection to the V2 camera. I've stopped and disabled the demo joy detector for now. When I run 

` python ~/AIY-projects-python/src/examples/vision/joy/joy_detection_demo.py --num_frames=100`

I receive the following output:

Starting JoyDetector...
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/home/pi/AIY-projects-python/src/examples/vision/joy/joy_detection_demo.py", line 125, in _run_detector
    with CameraInference(face_detection.model()) as inference:
  File "/home/pi/AIY-projects-python/src/aiy/vision/inference.py", line 42, in __init__
    self._engine.start_camera_inference(self._key, params)
  File "/home/pi/AIY-projects-python/src/aiy/vision/inference.py", line 228, in start_camera_inference
    self._communicate(request)
  File "/home/pi/AIY-projects-python/src/aiy/vision/inference.py", line 162, in _communicate
    raise InferenceException(response.status.message)
aiy.vision.inference.InferenceException: Camera is not running.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.5/threading.py", line 914, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.5/threading.py", line 862, in run
    self._target(*self._args, **self._kwargs)
  File "/home/pi/AIY-projects-python/src/examples/vision/joy/joy_detection_demo.py", line 140, in _run_detector
    break
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 758, in __exit__
    self.close()
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 752, in close
    raise exc
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 1995, in _control_callback
    "No data recevied from sensor. Check all connections, "
picamera.exc.PiCameraRuntimeError: No data recevied from sensor. Check all connections, including the SUNNY chip on the camera board

Traceback (most recent call last):
  File "/home/pi/AIY-projects-python/src/examples/vision/joy/joy_detection_demo.py", line 174, in <module>
    main()
  File "/home/pi/AIY-projects-python/src/examples/vision/joy/joy_detection_demo.py", line 170, in main
    detector.join()
  File "/home/pi/AIY-projects-python/src/examples/vision/joy/joy_detection_demo.py", line 87, in join
    self._animator.join()
  File "/usr/lib/python3.5/threading.py", line 1049, in join
    raise RuntimeError("cannot join thread before it is started")
RuntimeError: cannot join thread before it is started
Exception ignored in: <bound method RGBLED.__del__ of <aiy._drivers._rgbled.RGBLED object at 0xb5dc43d0>>
Traceback (most recent call last):
  File "/home/pi/AIY-projects-python/src/aiy/_drivers/_rgbled.py", line 66, in __del__
  File "/home/pi/AIY-projects-python/src/aiy/_drivers/_rgbled.py", line 192, in Reset
  File "/home/pi/AIY-projects-python/src/aiy/_drivers/_rgbled.py", line 95, in _PWriteInt
NameError: name 'open' is not defined

Judging from "No data recevied from sensor" - it's clearly not getting data from the camera. Here is a photo of my set up, I double checked that it's connected properly.

I suspect the culprit is the connection to the Vision Bonnet board (close up image). The Vision Bonnet receiver (white piece) arrived flimsy. The flat flex cable fits loosely in it, whereas all the other flex cable connections are sturdy. 

Could you please help identify / solve the problem???

thank you!

[image]
[image]

dmitriykovalev, 2018-01-08:
@MishaLaskin  
It looks like you forgot to do this step: "Secure the cable with the release lever." from our website. You should push and slide this black thing in the connector. Compare your setup to the [image](https://user-images.githubusercontent.com/6282596/34498215-40972af4-efc6-11e7-92af-31f78c462909.png)
from https://github.com/google/aiyprojects-raspbian/issues/224. Please let me know if that helps!

MishaLaskin, 2018-01-08:
ah, I didn't realize that the black thing could slide. Rookie mistake :)

simple fix! thank you!

MishaLaskin, 2018-01-08:
I'll close the issue then

closed 2018-01-08 by MishaLaskin

larboyer, 2018-06-07:
Ah, I see. Slide or it's also kind-of like a pivot. Thanks so much.

santhoshnp, 2018-08-28:
![screenshot 44](https://user-images.githubusercontent.com/39831264/44718450-bd95b700-aadd-11e8-8e62-6f7f71cd9a65.png