google/aiyprojects-raspbian #249: newest version throws an error about too many arguments

Thatkookooguy, 2018-01-08:
I updated my local branch, and when I tried to run any of the demo scripts, I got the following error:
```
Traceback (most recent call last):
  File "./src/assistant_library_demo.py", line 71, in <module>
    main()
  File "./src/assistant_library_demo.py", line 65, in main
    with Assistant(credentials, model_id) as assistant:
TypeError: __init__() takes 2 positional arguments but 3 were given
```

I fixed it temporarily with changing line 65:
```python
with Assistant(credentials, model_id) as assistant:
```
with
```python
with Assistant(credentials) as assistant:
```
looks like that fixed the problem.

I'm running this on a raspberry pi model 3 if that helps

drigz, 2018-01-08:
@Thatkookooguy Can you tell me if the following command makes the unedited script work? If so, I'll add an instruction to do that:

```
env/bin/pip install google-assistant-library==0.1.0
```

Thatkookooguy, 2018-01-08:
@drigz Yes, seems like running `env/bin/pip install google-assistant-library==0.1.0` fixed the problem
(without my change)

drigz, 2018-01-08:
Thank you! Eventually the old library will stop working, so users need an easy way to update to the new version. I'll see what I can come up with.

closed 2018-05-10 by dmitriykovalev