google/aiyprojects-raspbian #250: Conflicting Dependencies

ameer1234567890, 2018-01-08:
After running pip check and trying to fix dependencies I have noticed that there are conflicting dependencies. Is there a way to overcome this?
```
google-assistant-library 0.0.3 has requirement requests==2.14.2
google-api-core 0.1.3 has requirement requests<3.0.0dev,>=2.18.0
```

drigz, 2018-01-09:
Are you using the `aiyprojects` branch or the `voicekit` branch? Can you post the output of `env/bin/pip freeze`?

If you're on the `voicekit` branch, if you update to the latest code (to get #207) then recreate your virtualenv (`rm -r env && ./scripts/install-deps.sh`) you should be OK. We're upgrading `aiyprojects` but it'll take a little while - in the short term you could try cherry-picking #207 from voicekit.

ameer1234567890, 2018-01-09:
I am using the `aiyprojects` branch. If I cherry pick #207 then it fixes the issue. So, this needs to be fixed in `aiyprojects` branch.

dmitriykovalev, 2018-08-16:
This should be already fixed in the latest SD card image.

closed 2018-08-16 by dmitriykovalev