google/aiyprojects-raspbian #252: AIY Vision Bonnet

mphillips53, 2018-01-11:
Although dmesg reports that the bonnet booted correctly, none of the samples work for me. The joy_detection_demo starts with no errors but does not open a preview window overlay. The face_dection_demo starts with no errors and a preview window overlay opens but the app reports no faces were detected regardless of the number of frames chosen. All of the gpiozero demos launch without errors but none of the expected outputs occur(i.e., no led light on or off, no button pressed events on or off, etc.).
Using gpio, I can read and write outputs to each pin.  Is there a test routine that I can run directly on the bonnet to see if it works?

dmitriykovalev, 2018-01-15:
Please check [Troubleshooting](https://aiyprojects.withgoogle.com/vision/#assembly-guide-8-troubleshooting) section on our website. If that doesn't help, can you copy-paste your dmesg output?

dmitriykovalev, 2018-01-22:
@mphillips53, please reopen if you still have issues.

closed 2018-01-22 by dmitriykovalev