google/aiyprojects-raspbian #253: voice - other languages - please advise when available.

sheridat, 2018-01-12:
Hi, answering questions regarding how to use other languages by saying your working on it https://www.raspberrypi.org/forums/posting.php?mode=reply&f=114&t=200520

Without hassling yourselves - any idea how long it might be.

PaulFinch, 2018-01-17:
Other languages are already available (i use it), but curiously on the voicekit branch (which use the google assistant 0.1.0 version). I think the aiyprojects branch will be updated soon.

sheridat, 2018-01-17:
@PaulFinch Hi it is the aiyprojects branch that I am interested in (I've also got it working on voicekit). I've changed the title of this issue to hopefully make it plain that the issue is about the aiyprojects branch.

blandiant, 2018-01-21:
@PaulFinch Hello, How did you manage to use other languages (on voicekit) ? (i  can't do it...)
just to know, what the difference between the voicekit branch and the aiyprojects branch?
Thank you

PaulFinch, 2018-01-21:
The main diff is the google assistant library version, aiyprojects is 0.0.3 , the voicekit branch is 0.1.0

The 0.1.0 can handle other languages, as you can change the language in the assistant phone app.
That said, it didnt work out of box for me, i had to re-register the device with the google assistant sdk.

tei-chino, 2018-02-08:
Dear Sir,
I downloaded the voice branch zip file, but I found that the files remained as old version. For example, the requirements.txt is still being "google-assistant-grpc==0.0.2", rather than "google-assistant-grpc==0.1.0" .  would you like to check the download file to see if there is any problem.

drigz, 2018-02-08:
@tei-chino Make sure you're downloading the latest version of the repository. You may want to use `git clone` instead of downloading a zip.

tei-chino, 2018-02-08:
Thanks for your advice.
However, I tried to use  
git clone https://github.com/google/aiyprojects-raspbian.git AIY-projects-python,
but the files still are old ones ( "google-assistant-grpc==0.0.2").

drigz, 2018-02-08:
Only the `voicekit` branch has been updated so far - the `aiyprojects` branch will be updated soon.

tei-chino, 2018-02-08:
Thanks for your information. 
Would you like tell me when the the image file in the voice kie site
https://aiyprojects.withgoogle.com/voice
will be updated?

drigz, 2018-02-08:
We're working on it, but we don't have a date yet. We'll update this issue once it has been released.

tei-chino, 2018-02-08:
Thanks for your rapid response. We are looking forword to the updated image.

tei-chino, 2018-02-09:
I think that the new update allow us to interact using languages other than english, not just TTS.
I checked the source file device_helpers.py, but I could not find any language settings for my device.  Would you like to tell me how we can switch the laungauge for Assistant dialogue.

drigz, 2018-02-09:
@tei-chino You have to change the language on your phone. See this issue for more details: https://github.com/googlesamples/assistant-sdk-python/issues/150

tei-chino, 2018-02-09:
Thanks for your information.
I found the following information.
https://stackoverflow.com/questions/48017507/how-do-i-change-the-language-locale-in-the-google-assistant-sdk
I understood that becuase the AIY use the Google Assistant Library , we have to change the language in the Google Assistant settings on the phone. It is not possible to change language programmatically

Tetsuo30, 2018-03-13:
Hi , I understand that for now the only solution is to change the language in the APP on smartphone ...
I own a galaxy A3 2016 (this smartphone has the requierment for the assistant APP) but the app isn't avalaible for it.  Is there any way to run the assistant app on this phone, or in emulator ? or no news about change the language programmatically ?

drigz, 2018-03-13:
@Tetsuo30 If you're using assistant_grpc_demo.py or cloudspeech_demo.py you can change the language programmatically, although you won't have hotwording. https://github.com/googlesamples/assistant-sdk-python/issues/150 tracks programmatically changing the language with the Assistant Library.

tei-chino, 2018-03-13:
@Tetsuo30 If you upgrade your smartphone's OS to Android 6.0, maybe you can install the Assistant App to your smartphone.

blandiant, 2018-03-13:
Hi, before opening a new issue, 
I want to know if you have a solution: 
I change the language to french and now, to activate the Google assistant, i have to speak at 10cm from the mic or to yell...

drigz, 2018-03-13:
@blandiant That sounds like an issue in the Assistant Library. If it hasn't already been reported here: https://github.com/googlesamples/assistant-sdk-python/issues then that is the place to report it.

Tetsuo30, 2018-03-13:
Sorry but when I try to change language with the assistant_grpc_demo.py, it doesn't work . 
I try this in the terminal : assistant_grpc_demo.py -- lang fr-FR but no effect ... maybe I'm wrong , but can't find how to do this ...

drigz, 2018-03-13:
@Tetsuo30 You need to add `aiy.i18n.set_language_code('fr-FR')` to the demo file.

Tetsuo30, 2018-03-13:
Yes thx, I tried this code in the assistant_library_demo.py but forgot to try it in the grpc_demo.py.  It works,. But now, when asking something ...the sound stop in the middle of the response ... after 3 or 4 seconds ...

lhvo, 2018-03-24:
I have the same pb as Tetsuo30.
Putting the locale code 'fr-FR' is working for recognizing but for the response it's truncated.
Removing the regional setting back to EN then everything works fine. 
Dumping the audio data, at the end thre was a lot of "0x00". Some limitation from google ?

Tetsuo30, 2018-03-25:
Hi Ihvo, try the code from [Protovist](https://github.com/google/aiyprojects-raspbian/issues/297) in this post...it works for me ... The only issue for me now, is that I can't adjust the volume with the GRPC demo ... Impossible with the GUI or with the voice... when asking what's the volume the answer is always 50%, and can't lower it