google/aiyprojects-raspbian #255: Voice Hat Failure With check_audio.py

CelloVerb, 2018-01-13:
Hi there,

Using the 2018-01-03 image and the voice hat, I get the following error with the check_audio.py script:
`Traceback (most recent call last):
  File "/home/pi/AIY-projects-python/checkpoints/check_audio.py", line 159, in <module>
    main()
  File "/home/pi/AIY-projects-python/checkpoints/check_audio.py", line 154, in main
    do_checks()
  File "/home/pi/AIY-projects-python/checkpoints/check_audio.py", line 111, in do_checks
    if not check_voicehat_present():
  File "/home/pi/AIY-projects-python/checkpoints/check_audio.py", line 70, in check_voicehat_present
    card_id = CARDS_ID[get_aiy_device_name()]
KeyError: None`

Digging into the calls, the problem is that `get_aiy_device_name()` returns None, which is because there's no /proc/device-tree/hat device running.  I've run the configure-driver.sh script, and the right things seem to be enabled in /boot/config.txt.   Also, I can play audio through the speaker using the alsa driver and aplay, so the voice hat is working it seems.

Any suggestions?

Many thanks,
Cello

sheridat, 2018-01-13:
hi, the contents of my /proc/asound/cards file is as follows
0 [sndrpigooglevoi]: snd_rpi_googlev - snd_rpi_googlevoicehat_soundcar
                      snd_rpi_googlevoicehat_soundcard

What does your's contain?

CelloVerb, 2018-01-14:
Hi there,

Thanks for the reply.  Yes, /proc/asound/cards is the same as what you post:
 0 [sndrpigooglevoi]: snd_rpi_googlev - snd_rpi_googlevoicehat_soundcar
                      snd_rpi_googlevoicehat_soundcard

Similarly, `aplay -l` shows the same thing.

As I say, the sound device seems to work; I can play the sample audio using aplay.

The script is unhappy in `get_aiy_device_name()` due to looking for the `/proc/device-tree/hat` device.  What's that device do?  Which driver creates that device?  Wondering if there's something in the boot log that might give a clue

Any suggestions appreciated!
Cello

sheridat, 2018-01-14:
Hi, In my /proc/device-tree/hat there are 6 files
-r--r--r-- 1 root root  4 Jan 14 10:48 name
-r--r--r-- 1 root root 29 Jan 14 00:04 product
-r--r--r-- 1 root root  7 Jan 14 00:04 product_id
-r--r--r-- 1 root root  7 Jan 14 10:48 product_ver
-r--r--r-- 1 root root 37 Jan 14 10:48 uuid
-r--r--r-- 1 root root 13 Jan 14 10:48 vendor

Contents are  (i've surrounded the contents in ' ')
Name contains 'hat'
Product contains 'Google AIY audioHAT Board v1'
Product ID contains '0x0001'
Product Ver contains '0x0001'
UUID contains '0ddfb752-31be-4291-92da-73360695695e'
Vendor contains 'Google, Inc.'

looked in /var/log/boot.log - nothing related 

and had a look at dmesg
dmesg | grep voice
[    5.933008] snd-googlevoicehat-soundcard soc:sound: ASoC: CODEC DAI voicehat-hifi not registered - will retry
[    5.933030] snd-googlevoicehat-soundcard soc:sound: snd_soc_register_card() failed: -517
[    5.948181] voicehat-codec voicehat-codec: property 'voicehat_sdmode_delay' not found default 5 mS
[    5.951812] snd-googlevoicehat-soundcard soc:sound: voicehat-hifi <-> 3f203000.i2s mapping ok
[    5.963299] Error: Driver 'voicehat-codec' is already registered, aborting...
[  610.686692] voicehat-codec voicehat-codec: Enabling audio amp...
[  612.137215] voicehat-codec voicehat-codec: Disabling audio amp...
[  625.921847] voicehat-codec voicehat-codec: Enabling audio amp...
[  628.951088] voicehat-codec voicehat-codec: Disabling audio amp...
[  900.812179] voicehat-codec voicehat-codec: Enabling audio amp...
[  902.262875] voicehat-codec voicehat-codec: Disabling audio amp...
[  911.194873] voicehat-codec voicehat-codec: Enabling audio amp...
[  914.224097] voicehat-codec voicehat-codec: Disabling audio amp...

Have you tried just saving the credential files and then taking it apart/putting back together and reimaging?

drigz, 2018-01-15:
@CelloVerb I've seen this before, but only on a pre-production prototype Voice HAT, so I thought it wouldn't be seen with retail Voice Kits. The good news is that it doesn't seem to affect anything except check_audio.py, so you should hopefully be able to use your kit as normal.

That said, we'd like to figure this out as it's probably affecting other users, and it's a very unclear error. Thanks for reporting it!

If you have access to another Voice HAT, could you try swapping them out and letting us know if you still see the issue?

drigz, 2018-01-15:
PS. If you want to paste in multiline code samples, you can use three backticks like: <code>```</code> to wrap the code without putting it all on one line. I edited your description to do this.

pmiller624, 2018-01-15:
I ran into this issue yesterday. I ended up modifying check_audio.py to just run check_speaker_works() and check_mic_works(). Both returned successfully and I haven't noticed any other issues.

CelloVerb, 2018-01-15:
Ah, sure enough, after setting up credentials, etc., every does seem to work fine.  

I don't have another HAT unfortunately, but I did have another Raspberry PI, which didn't have any different results, even with a fresh image.

Nonetheless, everything else is working, so we're happy. Thanks for the tips.

Cello

P.S., Not sure if I should close the issue or not - sounds like others have seen this, and it might cause them to stumble, given that it's central in the setup instructions.

drigz, 2018-01-16:
Nope, let's leave it open - it may help others to see that this is a known problem (even if it's not a showstopper).

neorazz, 2018-01-23:
I wanted to chime in and say I'm having the same problem . I've edited the config.txt that activated the sound drivers.
Still getting the same error   how do I go about skipping this ?

drigz, 2018-01-23:
You should just be able to ignore the error in Check Audio and continue with the instructions.

neorazz, 2018-01-24:
Spent several hours trying to get this to work .
Found an aiy image dated 9-11-17. Works flawlessly no issues .

drigz, 2018-01-24:
@neorazz Glad you got your kit going. Could you clarify if the problem was just in Check Audio, or were the demo scripts also having problems?

neorazz, 2018-01-24:
Nothing worked  everything that accessed the aiy would crash with error Google didn't even have matches for .  The main difference I see is on boot this image doesn't start with a clk error . I will try redownloaded the newest image on another sd card tonight and see if something was screwed up there . I've noticed if my TV isnt at the no signal before I power the rasberry I don't get any video 

That may have glitched the initial setup. Well see

PicoPi, 2018-01-24:
Just installed the 20180103 verison and its screwed up. Nice going Google.
The 20170911 version worked perfectly.
WTH is wrong with your people Google? Didn't even bother testing it before uploading the new version? 
Imagine how many hours of people's time you wasted? Specially people who are beginners and don't know if its something they did, or a bug!

AlanJACarter, 2018-01-24:
I have installed 20180103 image to a 16G SD card from https://aiyprojects.withgoogle.com/voice/ and having the same issue reported on #255. I am a newbie so will try the fixes describe in #255 & #261. 
However a few other issues I have noticed with this image that may or may not be related: 
1. On reboot I get the message 'Failed to start and load kernel see systemct1 status see system-module-load.service'. Not sure if this related to changes made in /etc/modules.load.d/ according to https://www.raspberrypi.org/forums/viewtopic.php?t=147333.
2. I had to load the 20180103 image using Etcher.io on a Windows8 machine. Running the verify on a Windows10 machine caused a verify error. The verify error didn't occur on a NOOBS build flashed on the same SD card on same Windows10 machine.

drigz, 2018-01-24:
I'm sorry you're all having so much trouble with the 2018-01-03 image! We have tested it (indeed, I tested it with the Voice Kit myself) but the issues here seem to be related to different hardware versions, or to only show up occasionally, which is why I didn't find them before release.

@neorazz If you can't find references to the error messages, please copy-paste them into a GitHub issue or post a screenshot, so that we can have a look and see if we can do anything about them. The 2018-01-03 image has quite a lot of changes (including a newer version of Raspbian and support for the Vision Kit) so it's possible that one of them is causing problems.

@AlanJACarter Thanks for all the information. Regarding each issue:

1. The path correction shouldn't be necessary in theory (but we all know how useless theory can be sometimes). You should just be able to run the examples from the "dev terminal" on the Desktop. Make sure not to use `sudo`, as that ignores the dev terminal settings.

1. I've never seen the `Failed to start Load Kernel Modules` error before. Did you try running `systemct1 status systemd-modules-load.service` to see if it gives you any more information?

1. Sometimes the flashing doesn't work perfectly - that's why the verification runs. If you reflash the SD card with the 2018-01-03 image, do you get the same problem again? If it didn't flash correctly, that could be causing the other issues.

rok, 2018-01-24:
@drigz - I had the same issue (hat not recognized by raspbian) with the new AIY image. Kernel modules failed to start on boot.

The old AIY voice image works fine like @neorazz wrote. The old image here for people who want to get going now: https://dl.google.com/dl/aiyprojects/voice/aiyprojects-2017-09-11.img.xz

corwashere, 2018-01-25:
Just wanted to say that I am having the exact same problem as everyone else with the 2018-01-03 image. One thing I did not see posted on this thread is that the contents of the /proc/asound/cards file shows "no soundcards". Hopefully this can help someone differentiate if they have a config error or a problem with the image. I'm going to try the old image in the hopes that it works for me.

AlanJACarter, 2018-01-25:
**@rok**  - thanks for the link and I will drop back to the older image later today.

**@drigz** - Some further feedback:
1. I have not run under sudo and tried various ways of running the check audio script and they all returned the same problem as reported in  #238 and #255

2. systemctl status systemd-modules-load.service didn't seem to return any useful information,. Result is as follows:
systemd-modules-load.service - Load Kernel Modules
   Loaded: loaded (/lib/systemd/system/systemd-modules-load.service; static; ven
   Active: failed (Result: exit-code) since Thu 2016-11-03 17:16:42 UTC; 1 years
     Docs: man:systemd-modules-load.service(8)
           man:modules-load.d(5)
  Process: 114 ExecStart=/lib/systemd/systemd-modules-load (code=exited, status=
 Main PID: 114 (code=exited, status=1/FAILURE)
Warning: Journal has been rotated since unit was started. Log output is incomple

3. I am almost 95% sure that my SD card flashed correctly as Etcher.io returned a validation successful on the Windows 8 machine but failed earlier on Windows 10 machine. The Etcher.io version
I was using on Windows 10 machine was still beta.

I did some further investigation which I will summarise below. Being a newbie I don't want to spam this forum so please let me know if there is too much detail:

Both dwc2 and libcomposite are referenced in googadget.conf and seem to be missing.

**From <syslog> in /var/log...**
**Jan 25 08:06:29 raspberrypi systemd-modules-load[114]: Failed to find module 'dwc2'
Jan 25 08:06:29 raspberrypi systemd-modules-load[114]: Failed to find module 'libcomposite'**
Jan 25 08:06:29 raspberrypi fake-hwclock[109]: Thu 25 Jan 08:06:25 UTC 2018
Jan 25 08:06:29 raspberrypi systemd-modules-load[114]: Inserted module 'i2c_dev'
Jan 25 08:06:29 raspberrypi systemd-modules-load[114]: Inserted module 'pwm_soft'
Jan 25 08:06:29 raspberrypi systemd-fsck[115]: e2fsck 1.43.4 (31-Jan-2017)
Jan 25 08:06:29 raspberrypi systemd[1]: Started Create Static Device Nodes in /dev.
Jan 25 08:06:29 raspberrypi systemd[1]: Starting udev Kernel Device Manager...
Jan 25 08:06:29 raspberrypi systemd-fsck[115]: /dev/mmcblk0p2: clean, 149738/955808 files, 1021040/3877760 blocks
Jan 25 08:06:29 raspberrypi systemd[1]: Started File System Check on Root Device.
Jan 25 08:06:29 raspberrypi systemd[1]: Starting Remount Root and Kernel File Systems...
Jan 25 08:06:29 raspberrypi systemd[1]: Started Remount Root and Kernel File Systems.
Jan 25 08:06:29 raspberrypi systemd[1]: Starting Load/Save Random Seed...
Jan 25 08:06:29 raspberrypi kernel: [    0.000000] Booting Linux on physical CPU 0x0
Jan 25 08:06:29 raspberrypi kernel: [    0.000000] Linux version 4.9.59-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #1047 SMP Sun Oct 29 12:19:23 GMT 2017
Jan 25 08:06:29 raspberrypi kernel: [    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
Jan 25 08:06:29 raspberrypi kernel: [    0.000000] CPU: div instructions available: patching division code
Jan 25 08:06:29 raspberrypi kernel: [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
Jan 25 08:06:29 raspberrypi kernel: [    0.000000] OF: fdt:Machine model: Raspberry Pi 3 Model B Rev 1.2

**From <user.log>** Not sure if this is related to dwc2 and libcomposite
Jan 25 08:06:30 raspberrypi lightdm[465]: Error getting user list from org.freedesktop.Accounts: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.Accounts was not provided by any .service files
Jan 25 08:06:33 raspberrypi lightdm[601]: Error getting user list from org.freedesktop.Accounts: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.Accounts was not provided by any .service files
**Jan 25 08:06:34 raspberrypi /home/pi/bt-prov-server/bt_prov_server.py[367]: Reading kit id failed: [Errno 2] No such file or directory: '/proc/device-tree/hat/product_id'
Jan 25 08:06:34 raspberrypi /home/pi/bt-prov-server/bt_prov_server.py[367]: Setting device name: 'Kit-5762' ('AIY-0-Kit-5762')**
 
In terms of hardware I have purchased the Raspberry Pi 3 a AIY audio board from UK distributors in the last two weeks. I collected the following hardware/kernel/bootloader information for my Pi while running the 2018-01-03 image:

Raspberry Pi 3 Model B Rev 1.2 from file model under /proc/device-tree

cat /proc/cpuinfo
cat /sys/firmware/devicetree/base/model
Obtained:
Hardware	: BCM2835
Revision	: a02082
Serial: 00000000d976f26a

cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

uname -a 
Linux raspberrypi 4.9.59-v7+ #1047 SMP Sun Oct 29 12:19:23 GMT 2017 armv7l GNU/Linux

vcgencmd version
Bootloader version:
Aug  8 2017 12:05:45 
Copyright (c) 2012 Broadcom
version 196063438d7bb0769861c86f948ae653727adb7e (clean) (release)

No version number that I can see on the HAT AIY audio board.

LondonSi72, 2018-01-25:
For what it's worth, I get the same message about those two same modules failing to load. It doesn't seem to be impacting the operation of the Pi, though. I'm on image 01_03 running on a Pi3 and a kit I bought on early ship late last year. Is there a serial number on the voice hat boards we can lookup or some other way of identifying if this is an EEPROM problem with particular boards?

AlanJACarter, 2018-01-25:
**@LondonSi72** and others thanks for the help/advice.

Flashed the 9-11-17 image and have run Check Audio script no problem - nice to hear my own voice :). Given that I would suspect the 2018-01-03 image rather than HAT hardware/EEPROM but it wouldn't be the first time I was wrong.

corwashere, 2018-01-25:
Just as a followup, flashing to an earlier image worked. So clearly there's am issue with the 2018-01-03 image.

TomjrHunt, 2018-01-28:
Just thought I would add to this, 
Bought a new AIY voice kit, used on two PI3 and had kernel and 'no hat present' error messages when following and installing the new 2018-01-03 image. Couldn't get the 'test audio' to work, so was unable to proceed following the installation. FTR, total noob here also.  

Installed the 9-11-17 image and everything worked fine.

dbuggz, 2018-02-06:
Where did you find the older (9-11-17) image?

dbuggz, 2018-02-06:
?? is this because the image referenced in the VOICE docs 
"https://aiyprojects.withgoogle.com/voice#assembly-guide-1-get-the-voice-kit-sd-image"
is the VISION image?
 
https://dl.google.com/dl/aiyprojects/vision/aiyprojects-2018-01-03.img.xz

drigz, 2018-02-06:
@dbuggz, even though the URL says "vision", that image should work with the Voice Kit if you apply the workaround I added to the [issue description](https://github.com/google/aiyprojects-raspbian/issues/255#issue-288300943). If you prefer to use the older image, you can get it here: https://dl.google.com/dl/aiyprojects/voice/aiyprojects-2017-09-11.img.xz

dbuggz, 2018-02-06:
@drigz, I'll do the older image first, just to prove the Hardware. 
Thanks

TomjrHunt, 2018-02-06:
sorry @dbuggz, I can't remember now... But I did google it and i think i found it on the Raspberian Pi forums.

hyzhak, 2018-02-07:
@drigz thank you for your *workaround*. It really helps, I have just run one example and it worked fine.

But I still have issue with `systemd-modules-load.service` :(

```bash
pi@raspberrypi:~ $ systemctl status systemd-modules-load.service
● systemd-modules-load.service - Load Kernel Modules
   Loaded: loaded (/lib/systemd/system/systemd-modules-load.service; static; vendor preset: enabled)
   Active: failed (Result: exit-code) since Wed 2018-02-07 00:30:29 UTC; 13min ago
     Docs: man:systemd-modules-load.service(8)
           man:modules-load.d(5)
  Process: 346 ExecStart=/lib/systemd/systemd-modules-load (code=exited, status=1/FAILURE)
 Main PID: 346 (code=exited, status=1/FAILURE)

Feb 07 00:30:29 raspberrypi systemd[1]: Starting Load Kernel Modules...
Feb 07 00:30:29 raspberrypi systemd[1]: systemd-modules-load.service: Main process exited, code=exited, status=1/FAILURE
Feb 07 00:30:29 raspberrypi systemd[1]: Failed to start Load Kernel Modules.
Feb 07 00:30:29 raspberrypi systemd[1]: systemd-modules-load.service: Unit entered failed state.
Feb 07 00:30:29 raspberrypi systemd[1]: systemd-modules-load.service: Failed with result 'exit-code'.
```

I'm using:

- 2018-01-03 image
- Rpi3
- and sure Voice Kits. _I don't how to check hardware version of HAT_

If you need more information ping me in comments, I could debug something there.

drigz, 2018-02-07:
@hyzhak Are you having any problems as a result of the "Load Kernel Modules" service, or are you just concerned about the error? The Voice Kit should work fine: the error comes up because it's trying to load a module that is only available on the Pi Zero, but isn't needed by the Voice Kit.

hyzhak, 2018-02-07:
@drigz ok, I see. If `The Voice Kit` should find it ok to me too. I just wonder whether that workaround completely fix the issue of the absence of sound card or it is a just temporal patch.
Once I will find any related issues I will drop it here.

commit 51a09101 (2018-02-27): Handle old Voice HATs in get_aiy_device_name()

With this change, check_audio.py enables the audio driver as it should.

Fixes #255, b/71575605.

Change-Id: I35355d9462e9a8cfae196c905645a6cdb2e03061
closed 2018-02-27 by mbrooksx