google/aiyprojects-raspbian #256: GPIO not working

TimoCrafter, 2018-01-13:
Hello,
I am struggling with the GPIO pins since several months.
I tried all, 4, 17, 27 and 22, but none work. I am writing "1" to /sys/class/gpio/gpio17/value for example, and when I don't have the AIY voice board attached, it works and my LED goes on. But when I attach it and restart my raspberry, it just doesn't work. There is "1" in the value file, but the pin isn't powered. Do you have any idea?

Thanks!
Kind regards,
Timo

sheridat, 2018-01-13:
Hi,  Think your problem is that for lighting a led you should avoid the "driver" pins

See chapter 7 of this guide https://www.raspberrypi.org/magpi-issues/Essentials_AIY_Projects_Voice_v1.pdf 

It shows how to control a led and how to wire it up.   The other chapters

drigz, 2018-01-15:
This isn't a software problem with the kit - you can try asking on the [AIY Forum](https://www.raspberrypi.org/forums/viewforum.php?f=114).

closed 2018-01-15 by drigz